FreeCAD/FreeCAD #17113: PartDesign: Helix: crash - endless recursion when creating helix from sketch around cylinder

AIRCAP, 2024-10-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The following file consists of a cylinder and a simple triangular sketch on it's rim. its perfect to make a helix in PartDesign

if you select the sketch and click the helix tool, Freecad will enter an endless-recursion which ends up in a segfault

I could not get a useful backtrace from that since the recursion at the time of crash is tens of thousands of levels deep

[helixBig.zip](https://github.com/user-attachments/files/17298640/helixBig.zip)

### Full version info

```shell
OS: Ubuntu 20.04.6 LTS (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38925 (Git)
Build type: Debug
Branch: main
Hash: 12c6fc2920d083c7407419f8b9ae3651798c005a
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

PartDesign

### Anything else?

**Contributing factor:**

The part in question has a diameter of **250km** (pad) and a height of **10km**.  

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-10-08:
Confirmed on  
```
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38923 (Git) AppImage
Build type: Release
Branch: main
Hash: d20cb9e6ee198beb2bfd7e72d3dec0a575e3f28c
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

FlachyJoe, 2024-10-09:
It seems the precision is too high. It is hardcoded to `Precision::Confusion` equals 1e-7 everywhere, but it's an absolute value, should be relative to the requested shape dimension, yes?  :raised_eyebrow:

AIRCAP, 2024-10-09:
In theory FreeCAD has a accuracy feature:

```
double TopoShape::getAccuracy() const
{
    double deviation = 0.2;
    Base::BoundBox3d bbox = getBoundBox();
    if (bbox.IsValid())
        return ((bbox.LengthX() + bbox.LengthY() + bbox.LengthZ())/300.0 * deviation);
    return ComplexGeoData::getAccuracy();
}

defaults to - 1/1500 of the shape size (l+w+h)

but its not used anywhere and especially not passed on to OCCT anywhere

```

AIRCAP, 2024-10-10:
here is a backtrace ... after that it just keeps going deeper and deeper into 

`
GeomFill_Frenet::SingularD0(double, int, gp_Vec&, gp_Vec&, gp_Vec&, double&)
GeomFill_Frenet::D0(double, gp_Vec&, gp_Vec&, gp_Vec&)
GeomFill_Frenet::DoSingular(double, int, gp_Vec&, gp_Vec&, int&, int&, int&, int&, double&)
`
while the parameter values no longer change after a few dozen iterations of those three

[backtrace.txt](https://github.com/user-attachments/files/17320027/backtrace.txt)

AIRCAP, 2024-10-10:
Analysis:
`PartDesign::Helix::execute()`
calls
`BRepOffsetAPI_MakePipe::BRepOffsetAPI_MakePipe`
with a really long helix wire
which then calls
`BRepFill_Sweep::Build()`
which has an endless recursion inside OCCT

not much we can do but possibly change precision values OR (if that doesn't work) scale all shapes to a normalized size before handing them to OCCT  (let's say all operations OCCT ever gets to perform are on parts exactly 1m in size)

PR #17356 (2024-11-11): increase tolerance on helical shapes to avoid crash / broken geometry 
fix #17113
fix #9377
fix #11083

Helical shapes in PartDesign (and to a lesser degree helical paths) suffer from a precision flaw (see #9337) -- cylinders, cones, spheres, boxes, etc... are all "perfect" shapes in OCCT - ruled by exact mathematical formulas where their edges and surfaces are.

But OCCT has no primitive shape for helical shapes - be it 2d and 3d - so helical paths are approximated with bsplines and helicap solids are pipes along these bsplines.

now the issue is this approximation makes errors far in excess  [... cut]
commit 7680872e (2024-11-11): increase tolerance on helical shapes to avoid crash / broken geometry  (#17356)

* increase tolerance on helical shapes to avoid crash

fix #17113
fix #9377
fix #11083

* added test cases for additive and subtractive helix

* fixed whitespace issue, reduced turncount for speed

* relaxed large scale testcases for OCCT 7.3 and below, work around windows CI error #17394

* Further reduce max size in test for buggy Ubuntu OCCT
commit 6d68d222 (2024-11-11): increase tolerance on helical shapes to avoid crash / broken geometry  (#17356)

* increase tolerance on helical shapes to avoid crash

fix #17113
fix #9377
fix #11083

* added test cases for additive and subtractive helix

* fixed whitespace issue, reduced turncount for speed

* relaxed large scale testcases for OCCT 7.3 and below, work around windows CI error #17394

* Further reduce max size in test for buggy Ubuntu OCCT
closed 2024-11-11 by yorikvanhavre