FreeCAD/FreeCAD #17116: TechDraw: Changing the color doesn't work when creating a centerline

FEA-eng, 2024-10-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When creating centerlines (using any of the 3 centerline creation tools), all the properties can be changed normally but changing the color has no effect, the line stays black. However, after that, one can double-click on that line and change the color - then it works.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38923 (Git)
Build type: Release
Branch: main
Hash: d20cb9e6ee198beb2bfd7e72d3dec0a575e3f28c
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-10-08:
CC @WandererFan

Syres916, 2024-10-11:
When the user double-clicks the newly created centre line it invokes `TaskLineDecor` but when the centre line is created by `TaskCenterLine` it has it's own color and style settings, is it possible for `TaskCenterLine` to use `TaskLineDecor` as a sub dialog so there's a reduction in duplication and at the same time fix this issue or am I thinking too much like a user?

luzpaz, 2025-01-31:
@FEA-eng 👆

FEA-eng, 2025-02-02:
TechDraw has quite a lot of redundancy so it would be good to unify the tools as much as possible. There should be a single tool to set the appearance of lines. The centerline creation task panel has some additional options related to the length and orientation of the line but it would indeed be nice to handle the appearance with _TaskLineDecor_ somehow if it's not too hard to implement.

WandererFan, 2025-08-19:
Having trouble figuring out what the 3 centerline tools are.  These
[image]
seem to be the ones that work properly.

[image]

The first of these doesn't present a dialog so there is no opportunity to specify a colour.
[image]

Is there another tool I am forgetting?

FEA-eng, 2025-10-23:
Still cannot reproduce so closing this

closed 2025-10-23 by FEA-eng