FreeCAD/FreeCAD #17129: Assembly: Cannot add joints to grounded part

maxwxyz, 2024-10-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Simple demo file: 
[asm-no-joint-to-grounded.zip](https://github.com/user-attachments/files/17303155/asm-no-joint-to-grounded.zip)

The assembly has two parts of the same link in it, one is grounded, the other not. I cannot add joints between them.

[image]

### Full version info

```shell
OS: Windows 11 build 26100
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38871 (Git)
Build type: Release
Branch: main
Hash: 813ee36cb80e0d29c2c9b991a7b2fee2c61f5cf6
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CfdOF 1.27.10
  * Curves 0.6.48
  * dodo 1.0.1
  * fasteners 0.5.29
  * fasteners.backup1721936159.7542229 0.5.24 (Disabled)
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.24
  * Rocket 3.3.0
  * sheetmetal 0.4.26
  * Silk 0.1.5
  * Silk.backup1727166062.7227929 0.1.5 (Disabled)
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-09:
@PaddleStroke FYI

maxwxyz, 2024-10-09:
cube outside of the assembly was displayed on top, sorry!

closed 2024-10-09 by maxwxyz

rdgarlo, 2025-09-03:
I just came here to report this "bug." I made the same mistake. Oops! My bad. Thanks for the help!

hughrawlinson, 2025-09-19:
Fwiw, I do think this is a usability issue that should be addressed. Maybe the new operation guide box in 1.1 could be a good place to tell the user what's going wrong?