FreeCAD/FreeCAD #17133: Cancelling variable set creation crashes FreeCAD

blank-supportgis, 2024-10-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When starting the creation of a new variable set and then closing the creation window with escape, FreeCAD reacts with a crash-to-desktop.
Note that this does not occur when you close the creation window by clicking `cancel` or the window's close button.

1. Create a new project
2. Start creating a new variable set by clicking on `Create a variable set`.
3. Enter "width1" as name and leave the textfield by clicking somewhere else.
4. Close the window by pressing escape. FreeCAD should crash.

### Full version info

```shell
OS: Ubuntu 20.04.6 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * btl 0.9.9 (Disabled)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-09:
@pieterhijma FYI
I cannot reproduce
```
OS: Windows 11 build 26100
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38923 (Git)
Build type: Release
Branch: main
Hash: d20cb9e6ee198beb2bfd7e72d3dec0a575e3f28c
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CfdOF 1.27.10
  * Curves 0.6.48
  * dodo 1.0.1
  * fasteners 0.5.29
  * fasteners.backup1721936159.7542229 0.5.24 (Disabled)
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.24
  * Rocket 3.3.0
  * sheetmetal 0.4.26
  * Silk 0.1.5
  * Silk.backup1727166062.7227929 0.1.5 (Disabled)

```

byteofsoren, 2024-10-09:
I can not reproduce ether.

```
OS: Arch Linux (awesome)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38898 (Git) AppImage
Build type: Release
Branch: main
Hash: 32e09d9554db26414697ed19e0189ccab4ea9029
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
Installed mods: 
  * kicadStepUpMod 11.3.2
```

hyarion, 2024-10-09:
After a few tries I managed to reproduce the issue on RC2 and a recent dev build on macOS:

1. Start FreeCAD
2. Std_New
3. Std_VarSet
4. enter `test` as name
5. press tab to go to group
6. press escape

Crash

https://github.com/user-attachments/assets/d9b709ee-3d0f-4f8c-b09c-4f9c257259c2

<details><summary>Stack trace RC2 (without debug symbols)</summary>
<p>

```
Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libFreeCADApp.dylib           	       0x10573e870 App::Property::getGroup() const + 4
1   libFreeCADGui.dylib           	       0x106aff9e4 Gui::Dialog::DlgAddPropertyVarSet::changePropertyToAdd() + 580
2   libQt5Core.5.15.13.dylib      	       0x1064ade84 0x106368000 + 1334916
3   libQt5Core.5.15.13.dylib      	       0x106491094 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 176
4   libQt5Core.5.15.13.dylib      	       0x106491094 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 176
5   libQt5Core.5.15.13.dylib      	       0x106491094 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 176
6   AppKit                        	       0x18de38118 -[NSWindow(NSEventRouting) sendEvent:] + 284
7   libqcocoa.dylib               	       0x10a9b69fc 0x10a98c000 + 174588
8   libqcocoa.dylib               	       0x10a9bae78 0x10a98c000 + 192120
9   AppKit                        	       0x18dcff0c0 -[NSApplication run] + 512
10  libqcocoa.dylib               	       0x10a9b84d0 0x10a98c000 + 181456
11  freecad                       	       0x10401b248 main + 3464
12  dyld                          	       0x18a0460e0 start + 2360
```

</p>
</details> 

<details><summary>Stack trace dev build (with debug symbols)</summary>
<p>

```
Console is in 'commands' mode, prefix expressions with '?'.
Launching: /Users/nauck/projects/freecad/build-debug/bin/FreeCAD --log-file test.log
Launched process 45689
Stop reason: EXC_BAD_ACCESS (code=1, address=0x10)
bt
* thread #1, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0x10)
  * frame #0: 0x0000000102b4cd08 libFreeCADApp.dylib`App::Property::getGroup(this=0x0000000000000000) const at Property.cpp:145:12
    frame #1: 0x0000000105628f98 libFreeCADGui.dylib`Gui::Dialog::DlgAddPropertyVarSet::changePropertyToAdd(this=0x0000000318c28640) at DlgAddPropertyVarSet.cpp:307:15
    frame #2: 0x000000010562731c libFreeCADGui.dylib`Gui::Dialog::DlgAddPropertyVarSet::onEditFinished(this=0x0000000318c28640) at DlgAddPropertyVarSet.cpp:429:9
    frame #3: 0x000000010562d854 libFreeCADGui.dylib`QtPrivate::FunctorCall<QtPrivate::IndexesList<>, QtPrivate::List<>, void, void (Gui::Dialog::DlgAddPropertyVarSet::*)()>::call(f=(libFreeCADGui.dylib`Gui::Dialog::DlgAddPropertyVarSet::onEditFinished() at DlgAddPropertyVarSet.cpp:399), o=0x0000000318c28640, arg=0x000000016fdfbe80) at qobjectdefs_impl.h:152:13
    frame #4: 0x000000010562d798 libFreeCADGui.dylib`void QtPrivate::FunctionPointer<void (Gui::Dialog::DlgAddPropertyVarSet::*)()>::call<QtPrivate::List<>, void>(f=(libFreeCADGui.dylib`Gui::Dialog::DlgAddPropertyVarSet::onEditFinished() at DlgAddPropertyVarSet.cpp:399), o=0x0000000318c28640, arg=0x000000016fdfbe80) at qobjectdefs_impl.h:185:13
    frame #5: 0x000000010562d6a0 libFreeCADGui.dylib`QtPrivate::QSlotObject<void (Gui::Dialog::DlgAddPropertyVarSet::*)(), QtPrivate::List<>, void>::impl(which=1, this_=0x0000000318c49130, r=0x0000000318c28640, a=0x000000016fdfbe80, ret=0x0000000000000000) at qobjectdefs_impl.h:418:17
    frame #6: 0x0000000109c0ddfc libQt5Core.5.15.8.dylib`___lldb_unnamed_symbol16968 + 820
    frame #7: 0x0000000105627218 libFreeCADGui.dylib`Gui::Dialog::EditFinishedComboBox::editFinished(this=0x0000000318c286a8) at moc_DlgAddPropertyVarSet.cpp:132:5
    frame #8: 0x000000010562a7d4 libFreeCADGui.dylib`Gui::Dialog::EditFinishedComboBox::focusOutEvent(this=0x0000000318c286a8, event=0x000000016fdfc538) at DlgAddPropertyVarSet.h:53:16
    frame #9: 0x0000000104311f4c libQt5Widgets.5.15.8.dylib`QWidget::event(QEvent*) + 104
    frame #10: 0x000000010438af98 libQt5Widgets.5.15.8.dylib`QComboBox::event(QEvent*) + 104
    frame #11: 0x00000001042ec974 libQt5Widgets.5.15.8.dylib`QApplicationPrivate::notify_helper(QObject*, QEvent*) + 232
    frame #12: 0x00000001042ed7ac libQt5Widgets.5.15.8.dylib`QApplication::notify(QObject*, QEvent*) + 472
    frame #13: 0x00000001053de74c libFreeCADGui.dylib`Gui::GUIApplication::notify(this=0x000000016fdfe2b8, receiver=0x0000000318c286a8, event=0x000000016fdfc538) at GuiApplication.cpp:93:34
    frame #14: 0x0000000109bf0c74 libQt5Core.5.15.8.dylib`QCoreApplication::notifyInternal2(QObject*, QEvent*) + 176
    frame #15: 0x00000001042eb3a8 libQt5Widgets.5.15.8.dylib`QApplicationPrivate::setFocusWidget(QWidget*, Qt::FocusReason) + 204
    frame #16: 0x00000001042ebeb0 libQt5Widgets.5.15.8.dylib`QApplication::setActiveWindow(QWidget*) + 644
    frame #17: 0x0000000104309fd4 libQt5Widgets.5.15.8.dylib`QWidgetPrivate::deactivateWidgetCleanup() + 32
    frame #18: 0x0000000104310564 libQt5Widgets.5.15.8.dylib`QWidgetPrivate::hide_sys() + 140
    frame #19: 0x00000001043118e8 libQt5Widgets.5.15.8.dylib`QWidgetPrivate::hide_helper() + 100
    frame #20: 0x0000000104311c84 libQt5Widgets.5.15.8.dylib`QWidgetPrivate::setVisible(bool) + 284
    frame #21: 0x0000000104423c80 libQt5Widgets.5.15.8.dylib`QDialog::setVisible(bool) + 468
    frame #22: 0x0000000104422f50 libQt5Widgets.5.15.8.dylib`QDialogPrivate::hide(int) + 28
[... cut ...]
<p>

```
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38923 +2 (Git)
Build type: Release
Branch: fix-start-load-recent
Hash: dbee679388db2edebba5dd6ab5e3114e0c3d0ace
Python 3.11.8, Qt 5.15.8, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CfdOF 1.27.10
  * A2plus 0.4.68
  * lattice2 1.0.0
  * CurvedShapes 1.0.10
  * sheetmetal 0.4.26
  * Curves 0.6.6

```

</p>
</details>

kriswilk, 2024-10-09:
Following @hyarion's steps, the issue exists in Windows, too. However, it does not completely crash. In 1.0RC2 it generates an error in the Report View:

```
11:36:27  Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: Access violation
```

[image]

OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/unset/Qt default

pieterhijma, 2024-10-09:
Got it, can reproduce. I will fix it.

pieterhijma, 2024-10-12:
Thank you all for reporting and attempting to reproduce! It was a tricky one, a case that I didn't anticipate, see the PR for more details if you're interested.

PR #17215 (2024-10-18): Core: Fix VarSet add property dialog Esc segfault
Closes #17133.

The signal handlers act when the editing has finished. They assume that a new property is created and act on that assumption. When the user presses cancel or escape, the transaction is aborted which removes the newly created property. This of itself is not a problem but it turns out that after cancelling, signals can still arise and activate the signal handler that acts under the assumption that a new property has been created. This results in the new property not to be found and the segfault.

This PR disconnects the signal han [... cut]
closed 2024-10-18 by chennes