FreeCAD/FreeCAD #17134: Group name of first variable in a variable set is not honored

blank-supportgis, 2024-10-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When creating a new variable set, one must enter the first variable of that set right away. If you also assign a group (other than "Base"), I'd expect this variable to be referencable as `VarSet.<groupname>_<varname>`. However, autocompletion only offers this variable as `VarSet.<varname>`, omitting the group. All further variables appended to this set seem to be fine.

**Steps to reproduce**
1. Create a new file
2. Create a new variable set with `Create new variable set`.
3. Enter "width1" as the name and "foo" as the group.
4. Add another variable "width2" with group "foo" to the existing variable set
5. Create a new sketch and draw a single line in it.
6. Put a length constraint on that line and enter the constraint's expression editor.
7. The editor should offer `VarSet.foo_width1` as expression, but it only offers `VarSet.width1`. The expression `VarSet.foo_width2` is available as expected.

### Full version info

```shell
OS: Ubuntu 20.04.6 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * btl 0.9.9 (Disabled)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-09:
@pieterhijma FYI

furgo16, 2024-10-09:
This is because you're seeing two different dialogs. See the screenshot in the related issue: https://github.com/FreeCAD/FreeCAD/issues/14633. This seems like a duplicate.

blank-supportgis, 2024-10-09:
The issue #14633 isn't exactly the same, but it's solution would likely also cover this issue. 
So, I'd say, this issue can be closed in favor of the other.

closed 2024-10-09 by blank-supportgis