FreeCAD/FreeCAD #17137: Fillet shows as chamfer

naich58, 2024-10-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Filleting an internal angle in Part Design. Operation doesn't crash, but the 3d view shows a fillet on one side the the object and a chamfer on the other.

[image]

To repeat the problem, delete "Fillet01", select the corner where the chamfer is shown here, click on fillet and select the 3 other corners using the "select" button in "Fillet Parameters". If I select all corners and then choose Fillet, it doesn't happen.

[image]

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Assembly4 0.12.2
  * Manipulator 1.4.9
```

### Subproject(s) affected?

None

### Anything else?

[Pinas SDD layer.zip](https://github.com/user-attachments/files/17307151/Pinas.SDD.layer.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-09:
I cannot reproduce this.
Is it the same behavior in the stable 0.21.2 on your system? Can you try this with the latest weekly?
[image]

```
OS: Windows 11 build 26100
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38923 (Git)
Build type: Release
Branch: main
Hash: d20cb9e6ee198beb2bfd7e72d3dec0a575e3f28c
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CfdOF 1.27.10
  * Curves 0.6.48
  * dodo 1.0.1
  * fasteners 0.5.29
  * fasteners.backup1721936159.7542229 0.5.24 (Disabled)
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.24
  * Rocket 3.3.0
  * sheetmetal 0.4.26
  * Silk 0.1.5
  * Silk.backup1727166062.7227929 0.1.5 (Disabled)
```

naich58, 2024-10-09:
It doesn't do it on the nightly build. Close the issue, I guess. Sorry.

luzpaz, 2024-10-09:
@naich58 all good. Could you please add the full [About](https:///wiki.freecad.org/About) info for the weekly you tested on ?

Appreciate the time you took to fill out a report.

naich58, 2024-10-09:
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38923 (Git)
Build type: Release
Branch: main
Hash: d20cb9e6ee198beb2bfd7e72d3dec0a575e3f28c
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Assembly4 0.12.2
  * Manipulator 1.4.9