FreeCAD/FreeCAD #17138: Access Violation

13061951, 2024-10-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Whilst Converting step files to Solid. The above message came up. I could not delete what I was converting. I had to shut the programme down and restart it. 

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38871 (Git)
Build type: Release
Branch: main
Hash: 813ee36cb80e0d29c2c9b991a7b2fee2c61f5cf6
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * A2plus 0.4.68
  * AirPlaneDesign 0.4.1
  * Assembly4 0.50.15
  * CurvedShapes 1.0.10
  * Curves 0.6.48
  * ExplodedAssembly
  * fasteners 0.5.29
  * freecad.gears 1.3.0
  * Gridfinity 0.6.0
  * lattice2 1.0.0
  * MeshRemodel 1.10.25
  * POV-Ray-Rendering
  * Render 2024.9.25
  * sheetmetal 0.4.26
  * woodworking 0.21.2.33771
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

hyarion, 2024-10-09:
Can you provide steps to reproduce the issue? (and/or a file where it happens)

luzpaz, 2024-10-14:
@13061951 :point_up_2:

13061951, 2024-10-16:
> DuckDuckGo removed one tracker. More<https://duckduckgo.com/-6vzhQNYSzjUcltnuenxGKh58WEMKG-QMVO6ecCFhBrbk2lgqCgz1bm7W72y2qAbKXFCE-2koNhpWYmi9qFbq8guLWSz7M8AJCArPcs_UU4G-J_T16URZrEAOy7tDhBDfoCA_9ODDFpFruTBmTl5OZlfzhKXzHf6p-OTLN2DATqvncbjsJIN3ihetiiqAGfn01dv6S_ehlHTQ8BY4CWO6GGvYkMteA5_RqB3v6A0RG-PSx01Yk_aL1fuNSb1n6W-vmcq9IAgLokGQjPJ_yALGQQhxsjOXalg22hHCp13IMSIA7oyJNQ8kQAJ_CnH-OKG8pzRZ3ONoaQqZewlLWpNFRVJhdhqU4xJeRvgSuIxP-Ti3Z18xmSModCBu_TR1_PjhZlR23nh87Xm9OX4pqGfT9Ly4J-5C9IJ2ycbUexxIalWTaAoh_4Gp0lHGk3mSUpZ4LOsRHV3ruQwLbiLCD9-1eQpLAs3xF-zMFk1dDiqxJXYQgpjzXoAA>
> Deactivate <https://duckduckgo.com/>

Good Morning sir, I have tried several times, with no avail. I am now using the latest version and will capture as much data as I can if this happens again. Thank you for your Patients.

Kindest Regards.

Lynn Harris.

closed 2024-10-16 by luzpaz