FreeCAD/FreeCAD #17141: Techdraw drawing does not update.

imoldovavan, 2024-10-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

ThechDraw does not update when the part changes. 

### Full version info

```shell
freecad 1 rc2
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-09:
Please check out how to [report issues](https://github.com/FreeCAD/FreeCAD?tab=readme-ov-file#reporting-issues) which could help us diagnose and fix it.

- add full [About](https://wiki.freecad.org/About) info to the issue (`Help > About FreeCAD > Copy to clipboard`).
- share an example file with the issue present. You can upload a zipped file to the original issue.
- add steps to reproduce the issue. Ideally, provide a macro of the issue: `Macro > Macro recording...` and repeat all steps until the issue occurs.

imoldovavan, 2024-10-09:
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * A2plus 0.4.60k
  * Assembly3 0.11.4
  * Assembly4 0.12.7
  * fasteners 0.4.56
  * Manipulator 1.5.0
  * sheetmetal 0.2.63

[riser.zip](https://github.com/user-attachments/files/17316793/riser.zip)

Steps to reproduce.
1. Open the file
2. Go to drawing page
3. Try to update drawing. The part has an extra square pocket that was added after the views were inserted into the drawing. I could not update the views to match the part.

maxwxyz, 2024-10-10:
Confirmed. The source seems to stay at the previous feature. Manually changing it does also not solve the issue. Unfortunately the file does not work in the 0.21.2 version, the view is outside the page.
@WandererFan any idea why the view does not update?

furgo16, 2024-10-10:
Confirmed. I'm not sure how the OP runs the `3. Try to update drawing.` step, but marking the page to recompute and hitting <kbd>Ctrl</kbd> + <kbd>T</kbd> does not update the view, as per the issue description.

A workaround is to create a new page and add a new view. Then the square pocket is in the drawing.

I had a similar issue recently whereby after dragging an object in an assembly to a specific position (via a revolute joint), the drawing view would not update to show the new position. I'm not sure the issues are related, it could be that dragging components to a new position in an assembly is not supported in TechDraw.

```
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38926 (Git) Snap 1161
Build type: Release
Branch: main
Hash: a9bff78974f301100134a13c636e23aa91be5bcf
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * OpenTheme 2024.9.1
  * ezydxf
  * Ondsel-Lens 2024.9.13.02
  * Assembly4 0.50.15
```

WandererFan, 2024-10-10:
A couple of obvious things wrong here:

a) you are using a feature (chamfer) within Body001 that is not the tip.  Any features that occur after chamfer will not be included in the view - so Pocket002 will never be shown. (red box)

b) the view is from the bottom.  since Pocket002 does not cut through the entire thickness, you will not see it in a view from the bottom. (blue box)

[image]

imoldovavan, 2024-10-10:
The view is not from the bottom. Looking at the View Data it clearly shows that this view is the Front view. Also if I change the pocket to Thru All and the problem persists.

imoldovavan, 2024-10-10:
Did more testing. Changing the chamfer size updates the view. So it seems that the view is indeed from the bottom. Changing the pocket to thru all will not update the view.

imoldovavan, 2024-10-10:
Not sure if this is related, so let me know if I should start a new issue. ActiveView does not update when changes are being made to the Body or new features are added. I don't see a way to select Source for ActiveView.

imoldovavan, 2024-10-10:
I might have. If this is a feature it should be removed or force the user to select the body only. I can't think of a reason in my 30 years engineering career where this would be helpful.

WandererFan, 2024-10-11:
ActiveView never updates. It is just a wrapper around Tools > Save Image.

WandererFan, 2024-10-17:
There is no test in place to find a higher level object for the source item in the selection.   If the source is an App::Part, we draw its shape.  If the source is a  Part::Feature, PartDesign::Body, etc, we draw its shape.

As soon as we change this to substitute a feature's body (where it exists) as the source, somebody is going to demand the ability to draw steps in the body's development.   I think this might have come up before.

imoldovavan, 2024-10-17:
Fair enough. Been playing with the tech draw and some of the problems that I found have been fixed in the latest weekly build, including the Projection Angle problem that contributed to this issue being generated. 
We could close this, since this is not really an issue, more like a pitfall for new users.

closed 2024-10-17 by imoldovavan