FreeCAD/FreeCAD #17142: GUI: Zoom Adjust to selection does not work

gianlucafiore, 2024-10-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

[image]
when you select an object in BIM bank and perform the V,S command or with the drop-down menu, it does not perform any action

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Spanish/Argentina (es_AR)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-09:
What is BIM bank? Could you share a demo file?
@yorikvanhavre FYI

Syres916, 2024-10-09:
Having carried out a little testing, when an edge/face/vertex of a _BIM object_ is selected the `Std_ViewFitSelection` appears to do nothing but if the user selects the _BIM object_ in the tree (therefore no sub-objects are shown in the selection info) the camera does function on triggering the command. This is notably different to a 'standard' object (_Part__Feature_) where is doesn't matter whether the whole object or sub-object is selected the command still functions (as expected IMHO).

Roy-043, 2025-06-28:
This is not a BIM but a Core issue. It is related to the shape type of the object. It occurs for all compounds.

Test:

1. Create a Part_Box.
2. Create a Part_Cylinder.
3. Combine them in a Part_Compound.
4. Select an edge of the compound.
5. Invoke Std_ViewFitSelection.

Rexbas, 2025-11-08:
No

maxwxyz, 2025-11-08:
Confirmed, no change.