FreeCAD/FreeCAD #17143: Freecad 1.0 RC2 on MacOS crashes opening a file created with 0.21.2

pboe, 2024-10-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Freecad 1.0 RC2 crashes on o
[Workbench.FCStd.zip](https://github.com/user-attachments/files/17310700/Workbench.FCStd.zip)
pening an old project created with

[code]
OS: macOS 14.7
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
[/code]

### Full version info

```shell
OS: macOS 14.7
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-09:
@pboe how do you open the file? Could it be the same issue as https://github.com/FreeCAD/FreeCAD/issues/16828
@JohnOCFII sorry for tagging you again, does the file crash on your side?

pboe, 2024-10-09:
Same issue I think. I didn't find that one when i searched cor macos crashes. Sorry for the inconvenience.
I'm pretty shure I had a crash opening the document from an open Freecad instance, but can't reproduce it at the moment.
Opening the document from the finder with an double click while Freecad is not startet yet, crashes FC every time.

maxwxyz, 2024-10-09:
closing in favor of #16828

closed 2024-10-09 by maxwxyz

JohnOCFII, 2024-10-10:
> @JohnOCFII sorry for tagging you again, does the file crash on your side?

It did not crash for me (M1/arm MacBook Pro, but I did receive the two messages in the Report window:
```
22:10:15  Migrating Start Workbench to Start command... done.
22:11:05  Dimension: Autocorrect failed to fix broken references
```

(I'm out of town on different Mac -- so this was a completely fresh FreeCAD installation on a machine that has never had FreeCAD installed).

```
OS: macOS 14.6.1
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38923 (Git)
Build type: Release
Branch: main
Hash: d20cb9e6ee198beb2bfd7e72d3dec0a575e3f28c
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```