FreeCAD/FreeCAD #17144: Assembly crash

leventelist, 2024-10-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When creating a distance joint, I get segfault.

Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x3c050) [0x7f7bbc85b050]
#1  0x7f7b91ac9d10 in Standard_Transient::IncrementRefCounter() const from /lib/x86_64-linux-gnu/libTKernel.so.7+0
#2  0x7f7b983b727d in opencascade::handle<TopoDS_TShape>::BeginScope() from /usr/local/lib/Measure.so+0x27
#3  0x7f7b983c5536 in opencascade::handle<TopoDS_TShape>::Assign(Standard_Transient*) from /usr/local/lib/Measure.so+0x40
#4  0x7f7b983c544c in opencascade::handle<TopoDS_TShape>::operator=(opencascade::handle<TopoDS_TShape> const&) from /usr/local/lib/Measure.so+0x26
#5  0x7f7b983c53f7 in TopoDS_Shape::operator=(TopoDS_Shape const&) from /usr/local/lib/Measure.so+0x23
#6  0x7f7b93389ed7 in Part::TopoShape::setShape(TopoDS_Shape const&, bool) from /usr/local/lib/Part.so+0xe3
#7  0x7f7b9338a440 in Part::TopoShape::operator=(Part::TopoShape const&) from /usr/local/lib/Part.so+0x3e
#8  0x7f7b93358398 in Part::TopoShape::TopoShape(Part::TopoShape const&) from /usr/local/lib/Part.so+0xa2
#9  0x7f7b35f7134a in Assembly::AssemblyObject::isFaceType(App::DocumentObject*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, GeomAbs_SurfaceType) from /usr/local/lib/AssemblyApp.so+0x52
#10  0x7f7b35f71b5f in Assembly::AssemblyObject::getDistanceType(App::DocumentObject*) from /usr/local/lib/AssemblyApp.so+0x3b9
#11  0x7f7b35f6b874 in Assembly::AssemblyObject::makeMbdJointDistance(App::DocumentObject*) from /usr/local/lib/AssemblyApp.so+0x30
#12  0x7f7b35f6b37d in Assembly::AssemblyObject::makeMbdJointOfType(App::DocumentObject*, Assembly::JointType) from /usr/local/lib/AssemblyApp.so+0x1b9
#13  0x7f7b35f6c85b in Assembly::AssemblyObject::makeMbdJoint(App::DocumentObject*) from /usr/local/lib/AssemblyApp.so+0x5b
#14  0x7f7b35f6b0bc in Assembly::AssemblyObject::jointParts(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> >) from /usr/local/lib/AssemblyApp.so+0x82
#15  0x7f7b35f66d19 in Assembly::AssemblyObject::solve(bool, bool) from /usr/local/lib/AssemblyApp.so+0x151
#16  0x7f7b35fbedfe in Assembly::AssemblyObjectPy::solve(_object*) from /usr/local/lib/AssemblyApp.so+0xa6
#17  0x7f7b35fbd363 in Assembly::AssemblyObjectPy::staticCallback_solve(_object*, _object*) from /usr/local/lib/AssemblyApp.so+0xf3
#18  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(+0x1c9bc1) [0x7f7bbe1c9bc1]
#19  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x7d) [0x7f7bbe17be8d]
#20  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x4a63) [0x7f7bbe109c63]
#21  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(+0x26ab6a) [0x7f7bbe26ab6a]
#22  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(_PyObject_FastCallDictTstate+0x56) [0x7f7bbe17ed76]
#23  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(_PyObject_Call_Prepend+0xe0) [0x7f7bbe17efa0]
#24  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(+0x1f07a6) [0x7f7bbe1f07a6]
#25  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(+0x1e078e) [0x7f7bbe1e078e]
#26  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x7d) [0x7f7bbe17be8d]
#27  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x4a63) [0x7f7bbe109c63]
#28  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(+0x26ab6a) [0x7f7bbe26ab6a]
#29  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(+0x17f6ac) [0x7f7bbe17f6ac]
#30  /usr/local/lib/libFreeCADBase.so(PP_Run_Method+0x16b) [0x7f7bbebe4925]
#31  0x7f7bbeafd101 in Base::InterpreterSingleton::runMethodVoid(_object*, char const*) from /usr/local/lib/libFreeCADBase.so+0x49
#32  0x7f7bc163614c in Gui::PythonCommand::activated(int) from /usr/local/lib/libFreeCADGui.so+0xae
#33  0x7f7bc1630bcc in Gui::Command::_invoke(int, bool) from /usr/local/lib/libFreeCADGui.so+0x272
#34  0x7f7bc163092c in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /usr/local/lib/libFreeCADGui.so+0x12a
#35  0x7f7bc161b64a in Gui::Action::onActivated() from /usr/local/lib/libFreeCADGui.so+0x2a
#36  0x7f7bc1629f8d in QtPrivate::FunctorCall<QtPrivate::IndexesList<>, QtPrivate::List<>, void, void (Gui::Action::*)()>::call(void (Gui::Action::*)(), Gui::Action*, void**) from /usr/local/lib/libFreeCADGui.so+0x6b
#37  0x7f7bc16294f9 in void QtPrivate::FunctionPointer<void (Gui::Action::*)()>::call<QtPrivate::List<>, void>(void (Gui::Action::*)(), Gui::Action*, void**) from /usr/local/lib/libFreeCADGui.so+0x42
#38  0x7f7bc16284cf in QtPrivate::QSlotObject<void (Gui::Action::*)(), QtPrivate::List<>, void>::impl(int, QtPrivate::QSlotObjectBase*, QObject*, void**, bool*) from /usr/local/lib/libFreeCADGui.so+0x80
#39  /lib/x86_64-linux-gnu/libQt5Core.so.5(+0x2e8f8f) [0x7f7bbcce8f8f]
#40  0x7f7bbd95c782 in QAction::triggered(bool) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x42
#41  0x7f7bbd95f3ab in QAction::activate(QAction::ActionEvent) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xab
#42  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x256e8a) [0x7f7bbda56e8a]
#43  0x7f7bbda56fd7 in QAbstractButton::mouseReleaseEvent(QMouseEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xf7
#44  0x7f7bbdb4faaa in QToolButton::mouseReleaseEvent(QMouseEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xa
#45  0x7f7bbd9a4db8 in QWidget::event(QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1a8
#46  0x7f7bbd962fae in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x7e
#47  0x7f7bbd96b552 in QApplication::notify(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x13c2
#48  0x7f7bc15d26b4 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/local/lib/libFreeCADGui.so+0x134
#49  0x7f7bbccb1738 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x118
#50  0x7f7bbd96965e in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1be
#51  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1bdbd8) [0x7f7bbd9bdbd8]
#52  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1c0f60) [0x7f7bbd9c0f60]
#53  0x7f7bbd962fae in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x7e
#54  0x7f7bc15d26b4 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/local/lib/libFreeCADGui.so+0x134
#55  0x7f7bbccb1738 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x118
#56  0x7f7bbd13d42d in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0x80d
#57  0x7f7bbd111cec in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0xac
#58  /lib/x86_64-linux-gnu/libQt5XcbQpa.so.5(+0x6deca) [0x7f7bb73fbeca]
#59  /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x299) [0x7f7bbaf1e7a9]
#60  /lib/x86_64-linux-gnu/libglib-2.0.so.0(+0x54a38) [0x7f7bbaf1ea38]
#61  /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_iteration+0x2c) [0x7f7bbaf1eacc]
#62  0x7f7bbcd09876 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x66
#63  0x7f7bbccb01bb in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x12b
#64  0x7f7bbccb8316 in QCoreApplication::exec() from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x96
#65  /usr/local/lib/libFreeCADGui.so(+0x128efce) [0x7f7bc148efce]
#66  /usr/local/lib/libFreeCADGui.so(+0x128f283) [0x7f7bc148f283]
#67  0x7f7bc148f560 in Gui::Application::runApplication() from /usr/local/lib/libFreeCADGui.so+0x1e2
#68  FreeCAD(+0x2d683) [0x5555948fa683]
#69  /lib/x86_64-linux-gnu/libc.so.6(+0x2724a) [0x7f7bbc84624a]
#70  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x85) [0x7f7bbc846305]
#71  FreeCAD(+0x2c801) [0x5555948f9801]

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (XFCE/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.30115
Build type: Unknown
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
Installed mods: 
  * Manipulator 1.5.7
  * A2plus 0.4.68
  * Assembly4 0.50.15
  * QuickMeasure 2022.10.28
  * dxf-library
  * pcb 6.2023.1
  * kicadStepUpMod 11.3.9

git describe gives 1.0rc1-250-gc02ad52bfd
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

leventelist, 2024-10-09:
It seems that it is not Assembly specific. It crashes all the time.

maxwxyz, 2024-10-09:
Please check out how to [report issues](https://github.com/FreeCAD/FreeCAD?tab=readme-ov-file#reporting-issues) which could help us diagnose and fix it.

- share an example file with the issue present. You can upload a zipped file to the original issue.
- add steps to reproduce the issue. Ideally, provide a macro of the issue: `Macro > Macro recording...` and repeat all steps until the issue occurs.

leventelist, 2024-10-09:
Further analyzing the issue, I discovered it is only present when there is a grounded object in the assembly.

leventelist, 2024-10-09:
[crash.zip](https://github.com/user-attachments/files/17317289/crash.zip)
Attached is the test case for the crash. I started to build everything from scratch, and it crashed after a while. To reproduce the issue, start adding more joints to the pushbuttons.

luzpaz, 2024-10-09:
@leventelist can you reproduce on latest dev ? (https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds)

leventelist, 2024-10-09:
Here is some gdb trace...

assembly_new#Distance006 is unconnected to a grounded part so it is ignored.

Thread 1 "FreeCAD" received signal SIGSEGV, Segmentation fault.
0x00007ffff5936966 in std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_add_ref_copy() () from /usr/local/lib/libFreeCADGui.so
(gdb) where
#0  0x00007ffff5936966 in std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_add_ref_copy() () at /usr/local/lib/libFreeCADGui.so
#1  0x00007ffff5b29f07 in std::__shared_count<(__gnu_cxx::_Lock_policy)2>::operator=(std::__shared_count<(__gnu_cxx::_Lock_policy)2> const&) ()
    at /usr/local/lib/libFreeCADGui.so
#2  0x00007fff76d881ef in std::__shared_ptr<Part::TopoShapeCache, (__gnu_cxx::_Lock_policy)2>::operator=(std::__shared_ptr<Part::TopoShapeCache, (__gnu_cxx::_Lock_policy)2> const&) () at /usr/local/lib/Part.so
#3  0x00007fff76d88219 in std::shared_ptr<Part::TopoShapeCache>::operator=(std::shared_ptr<Part::TopoShapeCache> const&) () at /usr/local/lib/Part.so
#4  0x00007fff76d8a4a1 in Part::TopoShape::operator=(Part::TopoShape const&) () at /usr/local/lib/Part.so
#5  0x00007fff76d58398 in Part::TopoShape::TopoShape(Part::TopoShape const&) () at /usr/local/lib/Part.so
#6  0x00007fff0da5e34a in Assembly::AssemblyObject::isFaceType(App::DocumentObject*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, GeomAbs_SurfaceType) () at /usr/local/lib/AssemblyApp.so
#7  0x00007fff0da5eb5f in Assembly::AssemblyObject::getDistanceType(App::DocumentObject*) () at /usr/local/lib/AssemblyApp.so
#8  0x00007fff0da58874 in Assembly::AssemblyObject::makeMbdJointDistance(App::DocumentObject*) () at /usr/local/lib/AssemblyApp.so
#9  0x00007fff0da5837d in Assembly::AssemblyObject::makeMbdJointOfType(App::DocumentObject*, Assembly::JointType) () at /usr/local/lib/AssemblyApp.so
#10 0x00007fff0da5985b in Assembly::AssemblyObject::makeMbdJoint(App::DocumentObject*) () at /usr/local/lib/AssemblyApp.so
#11 0x00007fff0da580bc in Assembly::AssemblyObject::jointParts(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> >) ()
    at /usr/local/lib/AssemblyApp.so
#12 0x00007fff0da53d19 in Assembly::AssemblyObject::solve(bool, bool) () at /usr/local/lib/AssemblyApp.so
#13 0x00007fff0daabdfe in Assembly::AssemblyObjectPy::solve(_object*) () at /usr/local/lib/AssemblyApp.so
#14 0x00007fff0daaa363 in Assembly::AssemblyObjectPy::staticCallback_solve(_object*, _object*) () at /usr/local/lib/AssemblyApp.so
#15 0x00007ffff25c9bc1 in  () at /lib/x86_64-linux-gnu/libpython3.11.so.1.0
#16 0x00007ffff257be8d in _PyObject_MakeTpCall () at /lib/x86_64-linux-gnu/libpython3.11.so.1.0
#17 0x00007ffff2509c63 in _PyEval_EvalFrameDefault () at /lib/x86_64-linux-gnu/libpython3.11.so.1.0
#18 0x00007ffff266ab6a in  () at /lib/x86_64-linux-gnu/libpython3.11.so.1.0
#19 0x00007ffff257ed76 in _PyObject_FastCallDictTstate () at /lib/x86_64-linux-gnu/libpython3.11.so.1.0
--Type <RET> for more, q to quit, c to continue without paging--c
#20 0x00007ffff257efa0 in _PyObject_Call_Prepend () at /lib/x86_64-linux-gnu/libpython3.11.so.1.0
#21 0x00007ffff25f07a6 in  () at /lib/x86_64-linux-gnu/libpython3.11.so.1.0
#22 0x00007ffff25e078e in  () at /lib/x86_64-linux-gnu/libpython3.11.so.1.0
#23 0x00007ffff257be8d in _PyObject_MakeTpCall () at /lib/x86_64-linux-gnu/libpython3.11.so.1.0
#24 0x00007ffff2509c63 in _PyEval_EvalFrameDefault () at /lib/x86_64-linux-gnu/libpython3.11.so.1.0
#25 0x00007ffff266ab6a in  () at /lib/x86_64-linux-gnu/libpython3.11.so.1.0
#26 0x00007ffff257f6ac in  () at /lib/x86_64-linux-gnu/libpython3.11.so.1.0
#27 0x00007ffff2fe4925 in PP_Run_Method () at /usr/local/lib/libFreeCADBase.so
#28 0x00007ffff2efd101 in Base::InterpreterSingleton::runMethodVoid(_object*, char const*) () at /usr/local/lib/libFreeCADBase.so
#29 0x00007ffff5a347e8 in Gui::PythonCommand::activated(int) () at /usr/local/lib/libFreeCADGui.so
#30 0x00007ffff5a2f268 in Gui::Command::_invoke(int, bool) () at /usr/local/lib/libFreeCADGui.so
#31 0x00007ffff5a2efc8 in Gui::Command::invoke(int, Gui::Command::TriggerSource) () at /usr/local/lib/libFreeCADGui.so
#32 0x00007ffff5a19ce6 in Gui::Action::onActivated() () at /usr/local/lib/libFreeCADGui.so
#33 0x00007ffff5a28629 in QtPrivate::FunctorCall<QtPrivate::IndexesList<>, QtPrivate::List<>, void, void (Gui::Action::*)()>::call(void (Gui::Action::*)(), Gui::Action*, void**) () at /usr/local/lib/libFreeCADGui.so
#34 0x00007ffff5a27b95 in void QtPrivate::FunctionPointer<void (Gui::Action::*)()>::call<QtPrivate::List<>, void>(void (Gui::Action::*)(), Gui::Action*, void**) () at /usr/local/lib/libFreeCADGui.so
#35 0x00007ffff5a26b6b in QtPrivate::QSlotObject<void (Gui::Action::*)(), QtPrivate::List<>, void>::impl(int, QtPrivate::QSlotObjectBase*, QObject*, void**, bool*) () at /usr/local/lib/libFreeCADGui.so
#36 0x00007ffff10e8f8f in  () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#37 0x00007ffff1d5c782 in QAction::triggered(bool) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#38 0x00007ffff1d5f3ab in QAction::activate(QAction::ActionEvent) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#39 0x00007ffff1e56e8a in  () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#40 0x00007ffff1e56fd7 in QAbstractButton::mouseReleaseEvent(QMouseEvent*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#41 0x00007ffff1f4faaa in QToolButton::mouseReleaseEvent(QMouseEvent*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
[... cut ...]
#60 0x00007ffff10b01bb in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#61 0x00007ffff10b8316 in QCoreApplication::exec() () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#62 0x00007ffff586e3da in (anonymous namespace)::tryRunEventLoop(Gui::GUISingleApplication&) () at /usr/local/lib/libFreeCADGui.so
#63 0x00007ffff586e68f in (anonymous namespace)::runEventLoop(Gui::GUISingleApplication&) () at /usr/local/lib/libFreeCADGui.so
#64 0x00007ffff586e96c in Gui::Application::runApplication() () at /usr/local/lib/libFreeCADGui.so
#65 0x0000555555581683 in main ()

leventelist, 2024-10-09:
@luzpaz I try it. Compiling the latest main.

leventelist, 2024-10-09:
@luzpaz I can confirm, that the same issue happens with latest main. 1.0rc1-285-ga9bff78974

OS: Debian GNU/Linux 12 (bookworm) (XFCE/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.30133
Build type: Unknown
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
Installed mods: 
  * kicadStepUpMod 11.3.9
  * A2plus 0.4.68

maxwxyz, 2024-10-10:
I cannot get it to crash. Can you record a macro or screencast of what you are doing?
@PaddleStroke FYI

leventelist, 2024-10-10:
https://github.com/user-attachments/assets/b787eafb-7f27-425d-9264-41d87299f6ed

I falsely enable A2P, but when I select the assembly, the right WB is selected.

This is how I configure FreeCAD:

cmake ../ -DFREECAD_QT_VERSION:=5 -DBUILD_DESIGNER_PLUGIN:=ON -DPYTHON_EXECUTABLE=/usr/bin/python3

maxwxyz, 2024-10-10:
could you try it with the weekly builds: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

leventelist, 2024-10-10:
I tested it yesterday on this commit: 1.0rc1-285-ga9bff78974, and it still had that bug.

luzpaz, 2024-10-11:
An aside, @leventelist can you put your backtraces in tripple backtick [code blocks](https://docs.github.com/en/get-started/writing-on-github/working-with-advanced-formatting/creating-and-highlighting-code-blocks). This way you aren't triggering links to non-relevant tickets when you paste backtraces. TIA

PaddleStroke, 2024-10-12:
I cannot reproduce the bug. However looking at the trace the problem seems to come from` AssemblyObject::isFaceType`
I think perhaps the problem came from the use of `static_cast to Part::Feature` .

Which is coincidentally fixed in the PR https://github.com/FreeCAD/FreeCAD/pull/17205
@leventelist can you try this PR and feedback if it still crashes? Alternatively if you can't build this PR can you try after this PR merges?

PaddleStroke, 2024-10-12:
As a side note, when filling a bug report, could you please post a minimal example and clear reproduction steps? ie clear the document to remove as much as possible to keep only the minimum objects that trigger the problem. This way it is easier to reproduce and fix.

PaddleStroke, 2024-10-12:
I have just tested your file again after building out of https://github.com/FreeCAD/FreeCAD/pull/17205, and  opening the file and calling a solve of your assembly does crash freecad.

So it looks like this bug is fixed by https://github.com/FreeCAD/FreeCAD/pull/17205.

closed 2024-10-14 by chennes