FreeCAD/FreeCAD #17145: BIM: switching to strict IFC mode breaks the Std_undo tool results

semhustej, 2024-10-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When user switches to strict IFC mode and tries to use the Std_undo tool, the result is that the switch to strict IFC mode is not undone and if Std_undo is invoked again, copy of the last created element is created.

Steps to reproduce:

- Open sample file 

[IFC-undo.zip](https://github.com/user-attachments/files/17311433/IFC-undo.zip)

- Create extrusion with BIM_extrude
- Switch to strict IFC mode
- Invoke Std_undo
- Nothing is undone
- Invoke Std_undo again
- Copy of the extrusion is created

See behavior in video:

https://github.com/user-attachments/assets/6989e4f0-948d-4369-bd8b-4045b53cfd58

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38923 (Git)
Build type: Release
Branch: main
Hash: d20cb9e6ee198beb2bfd7e72d3dec0a575e3f28c
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: English/United States (en_US) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-09:
@yorikvanhavre FYI

yorikvanhavre, 2024-10-10:
Undo/redo support is partially done already, but indeed it needs to be finished.

Roy-043, 2026-09-02:
There is actually a general problem with undo in strict IFC:
1. Create a strict IFC document without a default structure using the button in the Status Bar.
2. Create two "No baseline" walls.
3. Press the Undo button.
4. Result: the 3D View shows a single wall, but the Tree View shows 2.
5. Press Undo again.
6. Result: another extra wall item in the Tree View.

When undoing the document conversion there are 3 problems:
1. Custom properties attached to documents cannot be undone (#32354). This means that using the custom "IfcFilePath" document property as a main switch to control the strict IFC behavior does not work reliably after undo.
2. While performing the undo step old objects are re-added to the document. This triggers the `ifc_observer` via the `slotCreatedObject` function to immediately convert them to strict IFC again.
3. A section of the `lock_document` function is not included in the transaction. That is easily fixable of course.

semhustej, 2026-09-02:
> 1. Create a strict IFC document without a default structure using the button in the Status Bar.
>
>     2. Create two "No baseline" walls.
>
>     3. Press the Undo button.
>
>     4. Result: the 3D View [...]

I can confirm this behavior. I think I was encountering it since FreeCAD 1.0.

```
OS: Debian GNU/Linux 13 (trixie) (GNOME/gnome-xorg/xcb)
Architecture: x86_64
Version: 26.3.0dev.20260902 (Git shallow) AppImage
Build date: 2026/09/02 02:30:22
Build type: Release
Branch: (HEAD detached at c68c45227)
Hash: c68c452275ec188f35c02f6b30aba880b3783cf1
Python 3.11.14, Qt 6.8.3, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.0, OCC 7.8.1
Locale: English/Ireland (en_IE)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: xcb/via Qt/via Qt
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/158.196/1
OpenGL version/vendor/renderer: 4.6 (Compatibility Profile) Mesa 25.0.7-2+deb13u1/Intel/Mesa Intel(R) HD Graphics 4600 (HSW GT2)

```