FreeCAD/FreeCAD #17147: BIM: Add warning when switching to and from strict IFC mode

semhustej, 2024-10-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The button for switching to strict IFC mode is on the status panel, next to background color switch and nudge movement setup. Both of these are non-destructive changes. It is quite easy for user to click on the IFC button by accident. It is possible to accidentally click on the button all the time user is working in the BIM workbench. Conversion to IFC can take multiple minutes, the changes to the project are also quite extensive. If invoked by accident it is quite timely mistake.

Therefore, it would be beneficial, in my opinion, to make sure the user does not start this process unintentionally. As user will not be switching between strict and hybrid modes, making it a bit more difficult to switch between the modes will not slow down user workflow that much.

[image]

### Proposed options:
**A**
Add warning window with text something akin to: 
_Turning strict IFC mode on will transform all elements in the project to IfcElements. Would you like to proceed?  OK/Cancel_
_Turning strict IFC mode off will enable to create non-IfcElements in the project. Would you like to proceed?  OK/Cancel_

**B**
When user clicks on the IFC button, submenu with options shows up (similar to navigation style button)

- Strict IFC Mode
- Hybrid IFC Mode

or 

- Strict IFC Mode On
- Strict IFC Mode Off

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38923 (Git)
Build type: Release
Branch: main
Hash: d20cb9e6ee198beb2bfd7e72d3dec0a575e3f28c
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: English/United States (en_US) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-10:
@yorikvanhavre FYI

yorikvanhavre, 2024-10-10:
I think, even better than that, it needs to be fully supported by the undo/redo system. Then it would be non-destructive and it does not need any warning.

This is basically a duplicate of https://github.com/FreeCAD/FreeCAD/issues/17145 so I'll close this one, okay? Feel free to reopen if needed.

yorikvanhavre, 2024-10-10:
Duplicate of #17145

semhustej, 2024-10-10:
> Duplicate of #17145

It is closely related to that issue, but I would consider it a different issue.

- if Undo works 100% reliably, that solves the issue from a large part.
- even if the above point is true, my opinion is that it would be good practice to warn user that such a drastic change to the document will be processed only by 1 click in the interface. But you are the boss here :), so let's leave it as it is. I guess it is also not worth spending time at this stage of development.

yorikvanhavre, 2024-10-10:
If you are able to revert that change, then I think the warning is not necessary anymore, and can even be annoying... But of course we can always put a "Don't show me this dialog anymore" checkbox...

I'll reopen the issue then

semhustej, 2024-10-10:
Thanks for listening to the feedback, Yorik. I would put this issue to a backlog, if something like that exists. It is definitely not a pressing issue. Let's see if in the meantime other people will voice their opinion whether this would be useful or not.