FreeCAD/FreeCAD #17150: 'Mirroring' breaks both the original and the mirror of a specific part design.

gdecarpentier, 2024-10-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I tried to mirror a reasonably complex part design (that was largely built using v0.21 and then I upgraded to RC2) and that created edges in wrong new directions in RC2. I've attached a ZIP file containing the project with the part and the video showing the repro. Hope this helps!

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Subproject(s) affected?

PartDesign

### Anything else?

[mirroring_bug.zip](https://github.com/user-attachments/files/17317717/mirroring_bug.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-10:
Please provide an example file.
@NomAnor FYI

gdecarpentier, 2024-10-10:
File and repro are already in the zip

gdecarpentier, 2024-10-15:
@maxwxyz Can you tell me why you say it needs an example file and repro steps when both are included?

AIRCAP, 2024-10-18:
can be fixed by #17119

AIRCAP, 2024-10-18:
The demo part only breaks if the mirror operation is refined. if its not refined it works and passes BOP checks. I assume at the time of reporting the bug 0.22 had Refinement turned on by default. it has since been turned off again unless set otherwise in preferences

PR #17119 (2024-10-21): Fix boolean operations
Fix #17159
Fix #17150
Fix #17085
Fix #16081 
Fix #15825
Fix #15615
can alleviate #11083 ( fixes with nonstandard setting for boolean fuzzy > 1000 - but that's too high to cut out very delicate parts from very large blocks )
Fix #9666
Allows workaround for #17338 

OCCT has issues with boolean operations failing when parts are touching but not intersecting. This includes but is not limited to the PartDesign helix , which is notorious for producing results that do not pass BOP tests or even fail completely.

The suspected problem is too narrow precision. Part precision sho [... cut]
closed 2024-10-21 by yorikvanhavre