FreeCAD/FreeCAD #17153: SIGSEGV on startup, only conda builds

benj5378, 2024-10-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Both RC1 and RC2 crash upon startup. Splashscreen is the only GUI that gets to load before crash.

```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x45320) [0x7f59afe45320]
#1  /lib/x86_64-linux-gnu/libc.so.6(+0x19d85c) [0x7f59aff9d85c]
#2  /tmp/.mount_FreeCADQePVw/usr/bin/../lib/libpython3.11.so.1.0(PyType_FromModuleAndSpec+0x8c) [0x7f59b28b753c]
#3  /tmp/.mount_FreeCADQePVw/usr/bin/../lib/./libshiboken2.cpython-311-x86_64-linux-gnu.so.5.15(SbkType_FromSpecWithBases+0x72) [0x7f59b1d30b62]
#4  0x7f59b1d446c1 in Shiboken::Enum::newTypeWithName(char const*, char const*, _typeobject*) from /tmp/.mount_FreeCADQePVw/usr/bin/../lib/./libshiboken2.cpython-311-x86_64-linux-gnu.so.5.15+0x531
#5  0x7f59b1d44892 in Shiboken::Enum::createScopedEnum(SbkObjectType*, char const*, char const*, char const*, _typeobject*) from /tmp/.mount_FreeCADQePVw/usr/bin/../lib/./libshiboken2.cpython-311-x86_64-linux-gnu.so.5.15+0x22
#6  /home/bensay/.local/lib/python3.11/site-packages/PySide2/QtCore.abi3.so(+0x251078) [0x7f597cc51078]
#7  /home/bensay/.local/lib/python3.11/site-packages/PySide2/QtCore.abi3.so(PyInit_QtCore+0x4f) [0x7f597cccbe3f]
#8  /tmp/.mount_FreeCADQePVw/usr/bin/../lib/libpython3.11.so.1.0(+0x2d8dec) [0x7f59b28d8dec]
#9  /tmp/.mount_FreeCADQePVw/usr/bin/../lib/libpython3.11.so.1.0(+0x20cdcb) [0x7f59b280cdcb]
[...]
#108  /tmp/.mount_FreeCADQePVw/usr/bin/../lib/libpython3.11.so.1.0(+0x2c669a) [0x7f59b28c669a]
#109  /tmp/.mount_FreeCADQePVw/usr/bin/../lib/libpython3.11.so.1.0(PyRun_StringFlags+0x69) [0x7f59b28ba9c9]
#110  0x7f59b1b4641c in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /tmp/.mount_FreeCADQePVw/usr/bin/../lib/libFreeCADBase.so+0x6c
#111  0x7f59b30a5733 in Gui::Application::runInitGuiScript() from /tmp/.mount_FreeCADQePVw/usr/bin/../lib/libFreeCADGui.so+0x43
#112  0x7f59b319869e in Gui::StartupPostProcess::showMainWindow() from /tmp/.mount_FreeCADQePVw/usr/bin/../lib/libFreeCADGui.so+0xce
#113  0x7f59b319a6e6 in Gui::StartupPostProcess::execute() from /tmp/.mount_FreeCADQePVw/usr/bin/../lib/libFreeCADGui.so+0x76
#114  0x7f59b30c7f9f in Gui::Application::runApplication() from /tmp/.mount_FreeCADQePVw/usr/bin/../lib/libFreeCADGui.so+0x4ff
```

### Full version info

```shell
RC2
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

benj5378, 2024-10-09:
I am considering whether this could be related to Python version. I have previously changed some packages because for being able to run FreeCAD with Python3.12 installed. However, isn't all Python packages bundled with conda build? Or is it dependent on locally installed PySide2?

Building myself is fine. Snap edge is also fine.

xtemp09, 2024-10-10:
@benj5378, it looks like your local Python influences where it is not supposed to. What happens if you temporarily rename `~/.local/` folder?

I think changing variables like `PYTHONPATH` or `PYTHONHOME` may solve the issue.

benj5378, 2024-11-09:
> What happens if you temporarily rename ~/.local/ folder?

@xtemp09, this fixed it. Is an AppImage supposed to be affected by local environment?

adrianinsaval, 2024-11-10:
> > What happens if you temporarily rename ~/.local/ folder?
> 
> @xtemp09, this fixed it. Is an AppImage supposed to be affected by local environment?

to some extent yes, you installed a pip package to your user site-packages directory that is conflicting, doing this is really not recommended, there's a reason why pip nowadays doesn't let you do this by default.

I don't know how desirable it is to isolate freecad from the user site-packages directory since in some cases that's used to install 3rdparty software required by addons. FreeCAD has it's own directory where the addon manager can install dependencies for addons, but not all addons declare all their pip dependencies and sometimes you just want to install something else yourself.

I'm inclined to close this issue since there's no action to take on our part here.

closed 2024-12-20 by adrianinsaval