FreeCAD/FreeCAD #17157: Problem in TechDraw

sozzadf, 2024-10-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Back lines apparent in assembly perspective view.

### Full version info

```shell
OS: Pop!_OS 22.04 LTS (pop:GNOME/pop)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Portuguese/Brazil (pt_BR)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * fasteners 0.5.29
```

### Subproject(s) affected?

TechDraw

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2024-10-10:
sample file?

sozzadf, 2024-10-11:
[Teste_Montagem_2.zip](https://github.com/user-attachments/files/17336317/Teste_Montagem_2.zip)

luzpaz, 2024-10-14:
@sozzadf what do you mean by 'back lines' ?

luzpaz, 2024-11-23:
@WandererFan can you reproduce with attached file ? (https://github.com/FreeCAD/FreeCAD/issues/17157#issuecomment-2406352712)

WandererFan, 2024-12-22:
If Corpo and Corpo001 are Fused/Unioned then the drawing looks right.  Don't know why group or compound don't behave the same yet.

[image]

luzpaz, 2025-09-09:
> If Corpo and Corpo001 are Fused/Unioned then the drawing looks right. Don't know why group or compound don't behave the same yet.

@WandererFan any ideas since ?

FEA-eng, 2026-03-19:
Still relevant

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.2.0dev.20260318 (Git shallow)
Build date: 2026/03/18 23:42:42
Build type: Release
Branch: grafted,grafted
Hash: 7886ebae105b99f176cddbfc62a106ab06a6d021
Python 3.11.15, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```

FEA-eng, 2026-10-16:
Still relevant

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 27.1.0dev.20261013 (Git shallow)
Build date: 2026/10/13 19:55:30
Build type: Release
Branch: (HEAD detached at 50daee777)
Hash: 50daee777321b5085accd1f57b57f1a7c1781cae
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 2.0.1, IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 474.82/NVIDIA Corporation/Quadro K4000/PCIe/SSE2
```