FreeCAD/FreeCAD #17162: Assembly Workbench crashes while constraining features of separate sub-assemblies

tomjtoth, 2024-10-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:
- double click "70x100cm module" in tree
- select those 2 plains

[image]

- select `distance joint` command
- crash

command line output:

```
[ttj@L450 ~]$ freecad
FreeCAD 1.0.0, Libs: 1.0.0RC2R38806 (Git)
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

(qt.qpa.plugin) Could not find the Qt platform plugin "wayland" in ""
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/libc.so.6(+0x3d1d0) [0x70af5a64c1d0]
#1  0x70aee113a675 in MbD::PartFrame::removeRedundantConstraints(std::shared_ptr<std::vector<unsigned long, std::allocator<unsigned long> > >) from /usr/lib/freecad/lib/libOndselSolver.so.1+0x25
#2  0x70aee1130dff in MbD::Part::removeRedundantConstraints(std::shared_ptr<std::vector<unsigned long, std::allocator<unsigned long> > >) from /usr/lib/freecad/lib/libOndselSolver.so.1+0x4f
#3  /usr/lib/freecad/lib/libOndselSolver.so.1(+0x34fb19) [0x70aee114fb19]
#4  0x70aee1192e9a in std::function<void (std::shared_ptr<MbD::Item>)> std::for_each<__gnu_cxx::__normal_iterator<std::shared_ptr<MbD::Part>*, std::vector<std::shared_ptr<MbD::Part>, std::allocator<std::shared_ptr<MbD::Part> > > >, std::function<void (std::shared_ptr<MbD::Item>)> >(__gnu_cxx::__normal_iterator<std::shared_ptr<MbD::Part>*, std::vector<std::shared_ptr<MbD::Part>, std::allocator<std::shared_ptr<MbD::Part> > > >, __gnu_cxx::__normal_iterator<std::shared_ptr<MbD::Part>*, std::vector<std::shared_ptr<MbD::Part>, std::allocator<std::shared_ptr<MbD::Part> > > >, std::function<void (std::shared_ptr<MbD::Item>)>) from /usr/lib/freecad/lib/libOndselSolver.so.1+0x5a
#5  0x70aee1190a2e in MbD::System::partsJointsMotionsLimitsDo(std::function<void (std::shared_ptr<MbD::Item>)> const&) from /usr/lib/freecad/lib/libOndselSolver.so.1+0x7e
#6  /usr/lib/freecad/lib/libOndselSolver.so.1(+0x17aefa) [0x70aee0f7aefa]
#7  0x70aee11917e8 in MbD::System::runPreDrag(std::shared_ptr<MbD::System>) from /usr/lib/freecad/lib/libOndselSolver.so.1+0x128
#8  0x70aee0fbff78 in MbD::ASMTAssembly::runPreDrag() from /usr/lib/freecad/lib/libOndselSolver.so.1+0x1e8
#9  0x70af1056863d in Assembly::AssemblyObject::solve(bool, bool) from /usr/lib/freecad/lib/AssemblyApp.so+0x2ad
#10  0x70af1058573c in Assembly::AssemblyObjectPy::solve(_object*) from /usr/lib/freecad/lib/AssemblyApp.so+0x7c
#11  0x70af105857cd in Assembly::AssemblyObjectPy::staticCallback_solve(_object*, _object*) from /usr/lib/freecad/lib/AssemblyApp.so+0x2d
#12  /usr/lib/libpython3.12.so.1.0(+0x1afe1c) [0x70af5c5afe1c]
#13  /usr/lib/libpython3.12.so.1.0(_PyObject_MakeTpCall+0x2fb) [0x70af5c580c7b]
#14  /usr/lib/libpython3.12.so.1.0(_PyEval_EvalFrameDefault+0xb69) [0x70af5c589829]
#15  /usr/lib/libpython3.12.so.1.0(_PyObject_FastCallDictTstate+0x1e6) [0x70af5c583a46]
#16  /usr/lib/libpython3.12.so.1.0(+0x1c0877) [0x70af5c5c0877]
#17  /usr/lib/libpython3.12.so.1.0(_PyObject_MakeTpCall+0x2a8) [0x70af5c580c28]
#18  /usr/lib/libpython3.12.so.1.0(_PyEval_EvalFrameDefault+0xb69) [0x70af5c589829]
#19  /usr/lib/libpython3.12.so.1.0(+0x1e2995) [0x70af5c5e2995]
#20  /usr/lib/libpython3.12.so.1.0(+0x1e2498) [0x70af5c5e2498]
#21  /usr/lib/freecad/lib/libFreeCADBase.so(PP_Run_Method+0xed) [0x70af5ce3112d]
#22  0x70af5cd59168 in Base::InterpreterSingleton::runMethodVoid(_object*, char const*) from /usr/lib/freecad/lib/libFreeCADBase.so+0x38
#23  0x70af5df168f8 in Gui::PythonCommand::activated(int) from /usr/lib/freecad/lib/libFreeCADGui.so+0x58
#24  0x70af5df1707e in Gui::Command::_invoke(int, bool) from /usr/lib/freecad/lib/libFreeCADGui.so+0x26e
#25  0x70af5df175a7 in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /usr/lib/freecad/lib/libFreeCADGui.so+0x147
#26  /usr/lib/libQt6Core.so.6(+0x1a3457) [0x70af5ada3457]
#27  0x70af5b7259a4 in QAction::activate(QAction::ActionEvent) from /usr/lib/libQt6Gui.so.6+0x154
#28  /usr/lib/libQt6Widgets.so.6(+0x212275) [0x70af5be12275]
#29  0x70af5be1ed8e in QAbstractButton::mouseReleaseEvent(QMouseEvent*) from /usr/lib/libQt6Widgets.so.6+0xde
#30  0x70af5bf4924d in QToolButton::mouseReleaseEvent(QMouseEvent*) from /usr/lib/libQt6Widgets.so.6+0x2d
#31  0x70af5bd4e816 in QWidget::event(QEvent*) from /usr/lib/libQt6Widgets.so.6+0xc36
#32  0x70af5bcfd9cc in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/libQt6Widgets.so.6+0x9c
#33  0x70af5bd02f59 in QApplication::notify(QObject*, QEvent*) from /usr/lib/libQt6Widgets.so.6+0x10f9
#34  0x70af5dea7809 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x99
#35  0x70af5ad45be8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/libQt6Core.so.6+0x158
#36  0x70af5bcf507c in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /usr/lib/libQt6Widgets.so.6+0x23c
#37  /usr/lib/libQt6Widgets.so.6(+0x164b02) [0x70af5bd64b02]
#38  /usr/lib/libQt6Widgets.so.6(+0x166160) [0x70af5bd66160]
#39  0x70af5bcfd9cc in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/libQt6Widgets.so.6+0x9c
#40  0x70af5dea7809 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x99
#41  0x70af5ad45be8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/libQt6Core.so.6+0x158
#42  0x70af5b378695 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /usr/lib/libQt6Gui.so.6+0x7e5
#43  0x70af5b3e7d24 in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt6Gui.so.6+0x304
#44  /usr/lib/qt6/plugins/platforms/../../../libQt6XcbQpa.so.6(+0x51ac7) [0x70af54de0ac7]
#45  /usr/lib/libglib-2.0.so.0(+0x5d559) [0x70af595cc559]
#46  /usr/lib/libglib-2.0.so.0(+0xc0157) [0x70af5962f157]
#47  /usr/lib/libglib-2.0.so.0(g_main_context_iteration+0x35) [0x70af595cba55]
#48  0x70af5afa985d in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt6Core.so.6+0x8d
#49  0x70af5ad50106 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt6Core.so.6+0x1a6
#50  0x70af5ad4a27d in QCoreApplication::exec() from /usr/lib/libQt6Core.so.6+0x8d
#51  0x70af5ddaf835 in Gui::Application::runApplication() from /usr/lib/freecad/lib/libFreeCADGui.so+0xcf5
#52  freecad(+0x6d0c) [0x5c25d0f70d0c]
#53  /usr/lib/libc.so.6(+0x25e08) [0x70af5a634e08]
#54  /usr/lib/libc.so.6(__libc_start_main+0x8c) [0x70af5a634ecc]
#55  freecad(+0x7105) [0x5c25d0f71105]
```

### Full version info

```shell
OS: Arch Linux (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: makepkg
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.12.6, Qt 6.7.2, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
```

### Subproject(s) affected?

Assembly

### Anything else?

[140x200cm bed frame.zip](https://github.com/user-attachments/files/17323020/140x200cm.bed.frame.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-10:
Just opening the file I get errors:
[image]

tomjtoth, 2024-10-10:
here's a simpler attempt with actual `Body` entities in use, I thought the lack of those contributed, but got the same result, I have also installed the `fasteners 0.5.29` add-on

[image]
[leg.zip](https://github.com/user-attachments/files/17327348/leg.zip)

I tried deleting the ground joint from the sub-assembly, but makes no difference, the zipped file already contains this change

command line output:

```
[ttj@L450 ~]$ freecad
FreeCAD 1.0.0, Libs: 1.0.0RC2R38806 (Git)
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

(qt.qpa.plugin) Could not find the Qt platform plugin "wayland" in ""
Ignore recovery file  "/home/ttj/.cache/FreeCAD/Cache/FreeCAD_Doc_5941bb09-4500-45ca-8342-52949a2a7104_900fa7_967873/fc_recovery_file.fcstd"  because it is older than the project file "/home/ttj/Projects/s\xC3\xA4ngynrunko/140x200cm bed frame.FCStd" 

leg#Distance003 is unconnected to a grounded part so it is ignored.
leg#Distance004 is unconnected to a grounded part so it is ignored.
leg#Cylindrical is unconnected to a grounded part so it is ignored.
leg#Cylindrical001 is unconnected to a grounded part so it is ignored.
Assembly : Ignoring bad solve, a grounded object moved.
leg#Distance003 is unconnected to a grounded part so it is ignored.
leg#Distance004 is unconnected to a grounded part so it is ignored.
leg#Cylindrical is unconnected to a grounded part so it is ignored.
leg#Cylindrical001 is unconnected to a grounded part so it is ignored.
leg#Distance003 is unconnected to a grounded part so it is ignored.
leg#Distance004 is unconnected to a grounded part so it is ignored.
leg#Cylindrical is unconnected to a grounded part so it is ignored.
leg#Cylindrical001 is unconnected to a grounded part so it is ignored.
3.6e-08 <App> Document.cpp(2900): leg#Distance003 still touched after recompute
3.5663e-05 <App> Document.cpp(2900): leg#Distance004 still touched after recompute
6.213e-05 <App> Document.cpp(2900): leg#Cylindrical still touched after recompute
7.1194e-05 <App> Document.cpp(2900): leg#Cylindrical001 still touched after recompute
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/libc.so.6(+0x3d1d0) [0x7c675224c1d0]
#1  0x7c66d953a675 in MbD::PartFrame::removeRedundantConstraints(std::shared_ptr<std::vector<unsigned long, std::allocator<unsigned long> > >) from /usr/lib/freecad/lib/libOndselSolver.so.1+0x25
#2  0x7c66d9530dff in MbD::Part::removeRedundantConstraints(std::shared_ptr<std::vector<unsigned long, std::allocator<unsigned long> > >) from /usr/lib/freecad/lib/libOndselSolver.so.1+0x4f
#3  /usr/lib/freecad/lib/libOndselSolver.so.1(+0x34fb19) [0x7c66d954fb19]
#4  0x7c66d9592e9a in std::function<void (std::shared_ptr<MbD::Item>)> std::for_each<__gnu_cxx::__normal_iterator<std::shared_ptr<MbD::Part>*, std::vector<std::shared_ptr<MbD::Part>, std::allocator<std::shared_ptr<MbD::Part> > > >, std::function<void (std::shared_ptr<MbD::Item>)> >(__gnu_cxx::__normal_iterator<std::shared_ptr<MbD::Part>*, std::vector<std::shared_ptr<MbD::Part>, std::allocator<std::shared_ptr<MbD::Part> > > >, __gnu_cxx::__normal_iterator<std::shared_ptr<MbD::Part>*, std::vector<std::shared_ptr<MbD::Part>, std::allocator<std::shared_ptr<MbD::Part> > > >, std::function<void (std::shared_ptr<MbD::Item>)>) from /usr/lib/freecad/lib/libOndselSolver.so.1+0x5a
#5  0x7c66d9590a2e in MbD::System::partsJointsMotionsLimitsDo(std::function<void (std::shared_ptr<MbD::Item>)> const&) from /usr/lib/freecad/lib/libOndselSolver.so.1+0x7e
#6  /usr/lib/freecad/lib/libOndselSolver.so.1(+0x17aefa) [0x7c66d937aefa]
#7  0x7c66d95917e8 in MbD::System::runPreDrag(std::shared_ptr<MbD::System>) from /usr/lib/freecad/lib/libOndselSolver.so.1+0x128
#8  0x7c66d93bff78 in MbD::ASMTAssembly::runPreDrag() from /usr/lib/freecad/lib/libOndselSolver.so.1+0x1e8
#9  0x7c6709d6863d in Assembly::AssemblyObject::solve(bool, bool) from /usr/lib/freecad/lib/AssemblyApp.so+0x2ad
#10  0x7c6709d8573c in Assembly::AssemblyObjectPy::solve(_object*) from /usr/lib/freecad/lib/AssemblyApp.so+0x7c
#11  0x7c6709d857cd in Assembly::AssemblyObjectPy::staticCallback_solve(_object*, _object*) from /usr/lib/freecad/lib/AssemblyApp.so+0x2d
#12  /usr/lib/libpython3.12.so.1.0(+0x1afe1c) [0x7c67541afe1c]
#13  /usr/lib/libpython3.12.so.1.0(_PyObject_MakeTpCall+0x2fb) [0x7c6754180c7b]
#14  /usr/lib/libpython3.12.so.1.0(_PyEval_EvalFrameDefault+0xb69) [0x7c6754189829]
#15  /usr/lib/libpython3.12.so.1.0(_PyObject_FastCallDictTstate+0x1e6) [0x7c6754183a46]
#16  /usr/lib/libpython3.12.so.1.0(+0x1c0877) [0x7c67541c0877]
#17  /usr/lib/libpython3.12.so.1.0(_PyObject_MakeTpCall+0x2a8) [0x7c6754180c28]
#18  /usr/lib/libpython3.12.so.1.0(_PyEval_EvalFrameDefault+0xb69) [0x7c6754189829]
#19  /usr/lib/libpython3.12.so.1.0(+0x1e2995) [0x7c67541e2995]
#20  /usr/lib/libpython3.12.so.1.0(+0x1e2498) [0x7c67541e2498]
#21  /usr/lib/freecad/lib/libFreeCADBase.so(PP_Run_Method+0xed) [0x7c6754a3112d]
#22  0x7c6754959168 in Base::InterpreterSingleton::runMethodVoid(_object*, char const*) from /usr/lib/freecad/lib/libFreeCADBase.so+0x38
#23  0x7c6755b168f8 in Gui::PythonCommand::activated(int) from /usr/lib/freecad/lib/libFreeCADGui.so+0x58
#24  0x7c6755b1707e in Gui::Command::_invoke(int, bool) from /usr/lib/freecad/lib/libFreeCADGui.so+0x26e
#25  0x7c6755b175a7 in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /usr/lib/freecad/lib/libFreeCADGui.so+0x147
#26  /usr/lib/libQt6Core.so.6(+0x1a3457) [0x7c67529a3457]
[... cut ...]
#49  0x7c6752950106 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt6Core.so.6+0x1a6
#50  0x7c675294a27d in QCoreApplication::exec() from /usr/lib/libQt6Core.so.6+0x8d
#51  0x7c67559af835 in Gui::Application::runApplication() from /usr/lib/freecad/lib/libFreeCADGui.so+0xcf5
#52  freecad(+0x6d0c) [0x6222d4611d0c]
#53  /usr/lib/libc.so.6(+0x25e08) [0x7c6752234e08]
#54  /usr/lib/libc.so.6(__libc_start_main+0x8c) [0x7c6752234ecc]
#55  freecad(+0x7105) [0x6222d4612105]
```

furgo16, 2024-10-11:
In my case, just opening the file and hitting Refresh (Ctrl+R) crashes FreeCAD instantly.

```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x42520) [0x71ebb3c79520]
#1  0x71eb875b2a60 in Part::TopoShape::operator=(Part::TopoShape const&) from /snap/freecad/1161/usr/lib/Part.so+0x170
#2  0x71eb87582cbc in Part::TopoShape::TopoShape(Part::TopoShape const&) from /snap/freecad/1161/usr/lib/Part.so+0x9c
#3  0x71eb6c1897fc in Assembly::AssemblyObject::isFaceType(App::DocumentObject*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, GeomAbs_SurfaceType) from /snap/freecad/1161/usr/lib/AssemblyApp.so+0x4c
#4  0x71eb6c18a087 in Assembly::AssemblyObject::getDistanceType(App::DocumentObject*) from /snap/freecad/1161/usr/lib/AssemblyApp.so+0x587
#5  0x71eb6c18a70b in Assembly::AssemblyObject::makeMbdJointDistance(App::DocumentObject*) from /snap/freecad/1161/usr/lib/AssemblyApp.so+0x4b
#6  0x71eb6c18b153 in Assembly::AssemblyObject::makeMbdJointOfType(App::DocumentObject*, Assembly::JointType) from /snap/freecad/1161/usr/lib/AssemblyApp.so+0x433
#7  0x71eb6c18b66e in Assembly::AssemblyObject::makeMbdJoint(App::DocumentObject*) from /snap/freecad/1161/usr/lib/AssemblyApp.so+0x4e
#8  0x71eb6c18c78d in Assembly::AssemblyObject::jointParts(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> >) from /snap/freecad/1161/usr/lib/AssemblyApp.so+0x6d
#9  0x71eb6c18cb93 in Assembly::AssemblyObject::solve(bool, bool) from /snap/freecad/1161/usr/lib/AssemblyApp.so+0x1f3
#10  0x71eb6c18ce37 in Assembly::AssemblyObject::execute() from /snap/freecad/1161/usr/lib/AssemblyApp.so+0x97
#11  0x71ebb606a63a in App::DocumentObject::recompute() from /snap/freecad/1161/usr/lib/libFreeCADApp.so+0xba
#12  0x71ebb600494d in App::Document::_recomputeFeature(App::DocumentObject*) from /snap/freecad/1161/usr/lib/libFreeCADApp.so+0x38d
#13  0x71ebb600c27b in App::Document::recompute(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&, bool, bool*, int) from /snap/freecad/1161/usr/lib/libFreeCADApp.so+0x65b
#14  0x71ebb60db8e8 in App::DocumentPy::recompute(_object*) from /snap/freecad/1161/usr/lib/libFreeCADApp.so+0x2e8
#15  0x71ebb60dba2c in App::DocumentPy::staticCallback_recompute(_object*, _object*) from /snap/freecad/1161/usr/lib/libFreeCADApp.so+0x2c
#16  /snap/freecad/1161/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x128008) [0x71ebb5623008]
#17  /snap/freecad/1161/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyObject_MakeTpCall+0x8c) [0x71ebb55daadc]
#18  /snap/freecad/1161/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x9dbc) [0x71ebb5576a1c]
#19  /snap/freecad/1161/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c23af) [0x71ebb56bd3af]
#20  /snap/freecad/1161/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(PyEval_EvalCode+0xbe) [0x71ebb56b83de]
#21  /snap/freecad/1161/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x20c4cd) [0x71ebb57074cd]
#22  /snap/freecad/1161/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(PyRun_StringFlags+0x79) [0x71ebb57084f9]
#23  0x71ebb5c2118c in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /snap/freecad/1161/usr/lib/libFreeCADBase.so+0x6c
#24  0x71ebb6b6c80c in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /snap/freecad/1161/usr/lib/libFreeCADGui.so+0x8c
#25  0x71ebb6b6c997 in Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...) from /snap/freecad/1161/usr/lib/libFreeCADGui.so+0xe7
#26  0x71ebb6b7a162 in StdCmdRefresh::activated(int) from /snap/freecad/1161/usr/lib/libFreeCADGui.so+0x72
#27  0x71ebb6b732ae in Gui::Command::_invoke(int, bool) from /snap/freecad/1161/usr/lib/libFreeCADGui.so+0x3fe
#28  0x71ebb6b735b6 in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /snap/freecad/1161/usr/lib/libFreeCADGui.so+0x126
#29  /snap/freecad/1161/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5(+0x2f48f4) [0x71ebb43d08f4]
#30  0x71ebb4f2bc26 in QAction::triggered(bool) from /snap/freecad/1161/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x46
#31  0x71ebb4f2e91c in QAction::activate(QAction::ActionEvent) from /snap/freecad/1161/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xac
#32  /snap/freecad/1161/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x26496a) [0x71ebb502a96a]
#33  0x71ebb502aac7 in QAbstractButton::mouseReleaseEvent(QMouseEvent*) from /snap/freecad/1161/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xf7
#34  0x71ebb512974e in QToolButton::mouseReleaseEvent(QMouseEvent*) from /snap/freecad/1161/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xe
#35  0x71ebb4f7594e in QWidget::event(QEvent*) from /snap/freecad/1161/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x20e
#36  0x71ebb4f32763 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /snap/freecad/1161/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x83
#37  0x71ebb4f3a3a4 in QApplication::notify(QObject*, QEvent*) from /snap/freecad/1161/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xa84
#38  0x71ebb6b10d20 in Gui::GUIApplication::notify(QObject*, QEvent*) from /snap/freecad/1161/usr/lib/libFreeCADGui.so+0xd0
#39  0x71ebb4398e4a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /snap/freecad/1161/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x13a
[... cut ...]
#58  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x80) [0x71ebb3c60e40]
#59  /snap/freecad/1161/usr/bin/FreeCAD(+0xab55) [0x6509d9f27b55]
```

```
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38926 (Git) Snap 1161
Build type: Release
Branch: main
Hash: a9bff78974f301100134a13c636e23aa91be5bcf
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * OpenTheme 2024.9.1
  * ezydxf
  * Ondsel-Lens 2024.9.13.02
  * Assembly4 0.50.15
```

PaddleStroke, 2024-10-12:
https://github.com/FreeCAD/FreeCAD/pull/15629 will solve this likely

luzpaz, 2024-10-21:
@rmdug JFYI, you should be using RC2 since it's the latest RC release.

rmdug, 2024-10-21:
@luzpaz 
I just installed the appimage and get the same issue.

Here is the configuration : 

OS: Manjaro Linux (KDE/plasma/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38991 (Git) AppImage
Build type: Release
Branch: main
Hash: ef2156d0f6d116b1642a5531a2d44fc6b7d17042
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: French/Belgium (fr_BE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion

PaddleStroke, 2025-07-18:
So I have just tested and : 

I cannot reproduce the issue in 42585 / win11. Neither with the original file '140x200...' Neither with the 'leg'. I only get `<App> Document.cpp(3039): _40x200cm_bed_frame#Link001 still touched after recompute` messages

So it does not really look related to 20614 afterall.

However I can't reproduce so perhaps this issue is gone now?

Or it is specific to Linux. @maxwxyz I think you use linux no? Can you reproduce those issues in current main?

maxwxyz, 2025-07-18:
I can only test on macOS and Windows, I do not use Linux

furgo16, 2025-07-18:
I'll try this evening before I start my travels, but I cannot guarantee it.