FreeCAD/FreeCAD #17167: [Feature] Allow user to arrange macro in subfolders

FlachyJoe, 2024-10-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When working with macros, they are all stored in one folder. A productive user may have too many to make it easy to use the current macro dialog list.
I propose to list the subfolders in the macro list so that the user can organize his files.

### Full version info

```shell
all
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-10-10:
Is this not the same as https://github.com/FreeCAD/FreeCAD/issues/16003 ?

Syres916, 2024-10-11:
OK, but it would be prudent for a single solution to accommodate both requests then, @FEA-eng would you be in agreement?

FEA-eng, 2024-10-11:
> @FEA-eng would you be in agreement?

My request is only about separating actual macros from the files belonging to some add-on workbenches. But more customizability and user-defined macro subdirectories could be useful as well.