FreeCAD/FreeCAD #17168: FreeCAD crashes disgracefully when a measurement point coordinate is edited

TadPath, 2024-10-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

(See video file attached)
I open my FreeCAD file.
I click on a measurement to get its properties.
I expand the position coordinates information of the two points of the measurement.
I edit the Y coordinate one of the coordinate points (change it from -14 to -15)
When I click away from the edit box, the program freezes for a few seconds then crashes disgracefully.

Variants:
I have tried creating a new measurement from scratch - which works OK - and tried to edit one of its points as above and the same disgraceful crash occurs (i.e. this is not limited to pre-existing measurements that were made with an older version of FreeCAD but also with new measurements made with the current version).

I have never encountered this problem with previous versions (prior to v1.0 RC)

### Full version info

```shell
OS: Arch Linux (LXDE/LXDE)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: makepkg
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.12.6, Qt 6.7.2, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Installed mods: 
  * fasteners 0.4.15
  * OpticsWorkbench 1.0.9
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-10:
Please test if the issue is still present in the dev version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
There were some fixed to measurement already.
@hlorus FYI

goleon69, 2024-10-10:
I do not see the video, am I missing something?

TadPath, 2024-10-10:
Same crash occurs with the weekly build appimage you linked to. No difference.
I did drag-drop the video to my issue but if it is not visible see the same video here on my GoogleDrive:
[FreeCAD_Crash_on_measurement_edit_video](https://drive.google.com/file/d/1xsaXFHow7KRH_Q7V-mIY4Cz8rF-IjTqi/view?usp=sharing)

luzpaz, 2024-10-11:
@tedpath can you add the full [About](https://wiki.freecad.org/About) info of the weekly build you tested ? TIA

TadPath, 2024-10-11:
Here is the About info for that weekly build:

OS: Arch Linux (LXDE/LXDE)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38923 (Git) AppImage
Build type: Release
Branch: main
Hash: d20cb9e6ee198beb2bfd7e72d3dec0a575e3f28c
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * fasteners 0.5.29
  * OpticsWorkbench 1.0.24

goleon69, 2024-10-11:
Tried with the 1.0 RC2, create a couple of sketches and padded, then put some measurements and then try to change multiple time the data in the position. but never crashed
[image]

But I have an old Mac :)
OS: macOS 12.7.6
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * IconThemes.backup1670063478.460454
  * ExtremeProDark.backup1670063426.751586
  * Glass.backup1670063844.392323
  * PieMenu.backup1670064528.67621 1.1.10

TadPath, 2024-10-12:
Yes, @goleon69 , when I make a simple model with the RC1 and make a new measurement, it works fine with me too. However, the problem occurs with measurements made on files created on older versions of FreeCAD, which ought to be compatible with new releases unless the FreeCAD developers want to announce otherwise or provide a project 'updater' function to make pre v1.0RC files compatible with  the latest release?
So that people can test this 'crash' on their own  machines (at their own risk!) I provide the FreeCAD file I used for the video above. Since I am having issues uploading content to this GitHub repo I again provide the link to a copy on my Google Drive:

[FreeCad_File_to_test_crash](https://drive.google.com/file/d/1PX8t_RD7S85UvGcPX-SS1friJfQOVDli/view?usp=sharing)

Perhaps @luzpaz can attach it properly to this issue thread for us?

wwmayer, 2024-10-12:
I can confirm the crash but it has nothing to do with changing the coordinates.

If you recompute the document after loading the file there are printed many warnings of the form:
`<TopoShape> TopoShapeExpansion.cpp(1887): unnamed lower element Edge1`
Then after a few seconds this is printed:
```
SIGSEGV signal raised: 11
terminate called after throwing an instance of 'std::runtime_error'
  what():  throw_signal
```

Running it in a debugger shows that the creation of a fillet fails:
```
0x00007fffc0e7600b in TopLoc_Location::Multiplied(TopLoc_Location const&) const () from /lib/x86_64-linux-gnu/libTKMath.so.7
(gdb) bt 10
#0  0x00007fffc0e7600b in TopLoc_Location::Multiplied(TopLoc_Location const&) const () at /lib/x86_64-linux-gnu/libTKMath.so.7
#1  0x00007fffc03506c3 in BRep_Tool::Surface(TopoDS_Face const&, TopLoc_Location&) () at /lib/x86_64-linux-gnu/libTKBRep.so.7
#2  0x00007fffc03651fb in BRepAdaptor_Surface::Initialize(TopoDS_Face const&, bool) () at /lib/x86_64-linux-gnu/libTKBRep.so.7
#3  0x00007fffc143db2e in ChFi3d_Builder::StartSol(opencascade::handle<ChFiDS_Spine> const&, opencascade::handle<BRepAdaptor_HSurface>&, gp_Pnt2d&, opencascade::handle<BRepAdaptor_HCurve2d>&, double&, opencascade::handle<ChFiDS_SurfData> const&, bool, int, opencascade::handle<BRepAdaptor_HSurface>&, opencascade::handle<BRepAdaptor_HCurve2d>&, bool&, bool&, bool&, bool&, opencascade::handle<BRepAdaptor_HSurface>&, gp_Pnt2d&, bool, TopoDS_Vertex const&) const () at /lib/x86_64-linux-gnu/libTKFillet.so.7
#4  0x00007fffc1442cee in ChFi3d_Builder::PerformSetOfSurfOnElSpine(opencascade::handle<ChFiDS_HElSpine> const&, opencascade::handle<ChFiDS_Stripe>&, opencascade::handle<BRepTopAdaptor_TopolTool>&, opencascade::handle<BRepTopAdaptor_TopolTool>&, bool) () at /lib/x86_64-linux-gnu/libTKFillet.so.7
#5  0x00007fffc144aabc in ChFi3d_Builder::PerformSetOfKGen(opencascade::handle<ChFiDS_Stripe>&, bool) () at /lib/x86_64-linux-gnu/libTKFillet.so.7
#6  0x00007fffc144df29 in ChFi3d_Builder::PerformSetOfSurf(opencascade::handle<ChFiDS_Stripe>&, bool) () at /lib/x86_64-linux-gnu/libTKFillet.so.7
#7  0x00007fffc1415d95 in ChFi3d_Builder::Compute() () at /lib/x86_64-linux-gnu/libTKFillet.so.7
#8  0x00007fffc1573e0b in BRepFilletAPI_MakeFillet::Build() () at /lib/x86_64-linux-gnu/libTKFillet.so.7
#9  0x00007fffc021800d in BRepBuilderAPI_MakeShape::Shape() () at /lib/x86_64-linux-gnu/libTKTopAlgo.so.7
(More stack frames follow...)
(gdb) bt 20
#0  0x00007fffc0e7600b in TopLoc_Location::Multiplied(TopLoc_Location const&) const () at /lib/x86_64-linux-gnu/libTKMath.so.7
#1  0x00007fffc03506c3 in BRep_Tool::Surface(TopoDS_Face const&, TopLoc_Location&) () at /lib/x86_64-linux-gnu/libTKBRep.so.7
#2  0x00007fffc03651fb in BRepAdaptor_Surface::Initialize(TopoDS_Face const&, bool) () at /lib/x86_64-linux-gnu/libTKBRep.so.7
#3  0x00007fffc143db2e in ChFi3d_Builder::StartSol(opencascade::handle<ChFiDS_Spine> const&, opencascade::handle<BRepAdaptor_HSurface>&, gp_Pnt2d&, opencascade::handle<BRepAdaptor_HCurve2d>&, double&, opencascade::handle<ChFiDS_SurfData> const&, bool, int, opencascade::handle<BRepAdaptor_HSurface>&, opencascade::handle<BRepAdaptor_HCurve2d>&, bool&, bool&, bool&, bool&, opencascade::handle<BRepAdaptor_HSurface>&, gp_Pnt2d&, bool, TopoDS_Vertex const&) const () at /lib/x86_64-linux-gnu/libTKFillet.so.7
#4  0x00007fffc1442cee in ChFi3d_Builder::PerformSetOfSurfOnElSpine(opencascade::handle<ChFiDS_HElSpine> const&, opencascade::handle<ChFiDS_Stripe>&, opencascade::handle<BRepTopAdaptor_TopolTool>&, opencascade::handle<BRepTopAdaptor_TopolTool>&, bool) () at /lib/x86_64-linux-gnu/libTKFillet.so.7
#5  0x00007fffc144aabc in ChFi3d_Builder::PerformSetOfKGen(opencascade::handle<ChFiDS_Stripe>&, bool) () at /lib/x86_64-linux-gnu/libTKFillet.so.7
#6  0x00007fffc144df29 in ChFi3d_Builder::PerformSetOfSurf(opencascade::handle<ChFiDS_Stripe>&, bool) () at /lib/x86_64-linux-gnu/libTKFillet.so.7
#7  0x00007fffc1415d95 in ChFi3d_Builder::Compute() () at /lib/x86_64-linux-gnu/libTKFillet.so.7
#8  0x00007fffc1573e0b in BRepFilletAPI_MakeFillet::Build() () at /lib/x86_64-linux-gnu/libTKFillet.so.7
#9  0x00007fffc021800d in BRepBuilderAPI_MakeShape::Shape() () at /lib/x86_64-linux-gnu/libTKTopAlgo.so.7
```

TadPath, 2024-10-12:
Thanks for exploring this @wwmayer .
The file was made and saved with a pre v.1.0RC FreeCAD and has not been manually edited or manipulated in any way outside FreeCAD. So are you suggesting the previous versions of FreeCAD created and saved a corrupt file or that there are deeper incompatibility issues with RC1.0 reading and processing files from older versions?
Any suggestions on how to proceed?

wwmayer, 2024-10-12:
The file is valid. But due to changes in v1.0 RC to solve the TNP older project files lack of some extra information that must be created when loading them.  This is achieved by doing a recompute of the document.

TadPath, 2024-10-12:
I tried to recompute the models in the document but it just keeps running and hangs - for over 1 hour on my desktop i5 Dell machine - so I decided to kill the process because that seems excessive (there should at least be some progress bar or something - the program shouldn't hang when doing long tasks). It didn't crash out in all that time, it was just hanging and hogging a core.
Should this issue be re-classified as a 'document migration fault' issue or something like that?

wwmayer, 2024-10-13:
According to the callstack I posted it looks like a bug in the CAD kernel. About the issue that it takes a long time I cannot confirm.

> Should this issue be re-classified as a 'document migration fault' issue or something like that?

No, I don't think so. I can load the file with v0.21. All objects are marked as "touched", i.e. they must be recomputed. When I do that it crashes and I get the same callstack as with v1.0RC.

goleon69, 2024-10-13:
Did a couple of tests just to confirm, opening the file on RC2 from @TadPath and recompute it crash on my pc. trying to change the value in positions also crash.
Design a new part with 0,21 and put some measurements, then opened with RC2 and the below window appear.
Once recomputed it didn't crash when changing the data on properties for measurement as well I tried not to recompute and change data as well in the same place and also there did't crash.
[image]

maxwxyz, 2025-02-26:
is this still an issue in the latest dev version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

TadPath, 2025-02-26:
> is this still an issue in the latest dev version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

Hello. Yes - If I follow the same steps I get exactly the same crash behaviour with this latest build. The console shows this error:

/tmp/.mount_fcwb.AOfBFam/AppRun: line 31: 10095 Segmentation fault      (core dumped) "${MAIN}" "$@"

About info:
OS: Arch Linux (LXDE/LXDE/xcb)
Architecture: x86_64
Version: 1.1.0dev.40408 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: b07caa732e63064b8851ecdda184292962d95c15
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.7.0, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical/physical DPI: 96.221/96.221
Installed mods: 
  * fasteners 0.5.29
  * OpticsWorkbench 1.0.24

luzpaz, 2026-09-06:
ping @drwho495

drwho495, 2026-09-06:
> ping [@drwho495](https://github.com/drwho495)

I see the crash, but it doesn't appear to be caused by anything related to topological naming.