FreeCAD/FreeCAD #17169: very unstable under new OSX

splaisan, 2024-10-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

New to free-cad
I built a part from a YT video and applied constraints, at the end of the process most of my constraints were declared redundant (not true) and when I pad and start applying fillets, the app crashes or the model disappears.
Is there a known incompatibility between FC and the latest OSX.

[image]

### Full version info

```shell
v0.21.2 mac OSX 15.0.1
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-10:
@splaisan please test the latest dev version (https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds) or 1.0 release candidate (https://github.com/FreeCAD/FreeCAD/releases/tag/1.0rc2)
If there is still an issue we need more info, like an example file and steps how to reproduce.

depthoffocus, 2024-10-10:
Looking at this sketch in the latest weekly build on the Mac (see below), there is definitely something odd or corrupted about this sketch. How was it originally produced?

The problematic edge appears to be the inner bottom edge -- Line1, the first edge in the sketch.

Delete it and redraw it and you can re-constrain it the same way. 

[image]

It then produces one of the weird erroneous "wire is not closed" errors. If you resize one of the edges, then close the sketch, it will extrude.

[image]

```
OS: macOS 14.6.1
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38923 (Git)
Build type: Release
Branch: main
Hash: d20cb9e6ee198beb2bfd7e72d3dec0a575e3f28c
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * ThreadProfile 1.92.0
  * freecad.gears 1.3.0
  * OpenTheme 2024.9.1
  * fasteners 0.5.29
  * Curves 0.6.47
```

splaisan, 2024-10-11:
thanks for your kind help
As you suggested, I had to recreate the model from scratch because some more corruption showed up. 
The current latest rc2 version worked out and I could proceed with finishing and adding filets.
The icons in the toolboxes have somewhat changed and I had to re-figure out my way through this but it finally worked.
thanks for the great tool!!
I attach the final model, not exactly what was created in the YT video but close enough

[gsm_stand.FCStd.zip](https://github.com/user-attachments/files/17340495/gsm_stand.FCStd.zip)

closed 2024-10-11 by splaisan