FreeCAD/FreeCAD #17173: Double entry in the preference menu

goleon69, 2024-10-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

on Mac OS with Freecad 1.0 RC2 when opening the preferences menu, on the general drop down there are two help menu at the bottom. see picture below. I guess one need to be removed.

[image]

### Full version info

```shell
OS: macOS 12.7.6
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * IconThemes.backup1670063478.460454
  * ExtremeProDark.backup1670063426.751586
  * Help.backup1670064522.0207932 1.0.3
  * A2plus.backup1670063488.6481578 0.4.60
  * Glass.backup1670063844.392323
  * PieMenu.backup1670064528.67621 1.1.10
  * EM.backup1670063448.14293
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-10-10:
Do you still have the configuration files from the previous version? Can you try clearing them?

goleon69, 2024-10-10:
I do not know how to do it and yes, I do have both the 0.21 and the RC2 installed. So I guess I have the configuration from the previous version. How do I clear them?

Roy-043, 2024-10-10:
The problem is caused by the old Help addon that you still have in your Mod folder. You need to remove it.
> Help.backup1670064522.0207932 1.0.3

Enter this in the Python console to find your Mod folder:
`App.getUserAppDataDir()`

goleon69, 2024-10-11:
Perfect! thanks! I removed that Help folder in. the directory and now it works properly

[image]

Now the configuration is like this:
OS: macOS 12.7.6
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * IconThemes.backup1670063478.460454
  * ExtremeProDark.backup1670063426.751586
  * A2plus.backup1670063488.6481578 0.4.60
  * Glass.backup1670063844.392323
  * PieMenu.backup1670064528.67621 1.1.10

closed 2024-10-11 by luzpaz