FreeCAD/FreeCAD #17174: Regression: Ellipse by Center Tool in Sketcher behaves weird

bruddha, 2024-10-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. Create a sketch on the XY plane
2. Create an Ellipse by Center starting in the center of the sketch, second point on X-Axis, third point on Y-Axis.
3. Result is either an ellipse with height 0, width 0, an ellipse where the center is not at 0,0, or an ellipse with some solver issues

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Switzerland (de_CH)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-10-11:
@bruddha what version does it work in ? (since you mention it's a regression)

luzpaz, 2024-10-11:
An aside, just wondering if there is any relationship to #13925

bruddha, 2024-10-11:
I always used the LinkStage version before. So I assume it also worked on 0.21 which LinkStage is based on.
There doesn't seem to be a relation to the other bug, as it is constructed to not work by moving the mouse until NaN is shown or by aborting with the right mouse button, whereas this bug is able to be reproduced by normal creation of an ellipse with 3 clicks or also by entering values.

bruddha, 2024-10-11:
I just checked and it actually does work in the latest LinkStage version and also in 0.21.2.

Roy-043, 2024-10-11:
Confirmed in:
```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38794 (Git)
Build type: Release
Branch: main
Hash: a47800d9ccd7d7a7cdb1779d56de03d3c64cd76e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Dutch/Netherlands (nl_NL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods:
```

Fixed in:
```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38923 (Git)
Build type: Release
Branch: main
Hash: d20cb9e6ee198beb2bfd7e72d3dec0a575e3f28c
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Dutch/Netherlands (nl_NL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods:
```

closed 2024-10-11 by Roy-043

bruddha, 2024-10-13:
Which commit fixed the bug and will it also be backported to 1.0? Anyway, thanks for the fix.

Roy-043, 2024-10-14:
I have not been able to find the actual commit. Let's wait for the next RC. This issue can be reopened if the problem has not been fixed in that version.

hogwell, 2024-10-15:
I am having the same problems with trouble drawing a 3-point ellipse at the origin in 1.0RC2.
(e.g. solver issues)

I will look for this to be working better in RC3.