FreeCAD/FreeCAD #17176: BIM: Ambiguous relationship between "BIM" material and "Material WB" material

semhustej, 2024-10-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Priority: low

- When Material workbench material is set for a BIM element (e.g. wall), it does not show up in the 'Material' property of the element.
- When "BIM" material is than assigned to the wall, the wall has two materials assigned - one "BIM" material and "Material WB" material

See the behavior in the video:

https://github.com/user-attachments/assets/bf6b800d-de12-4d01-81dd-36111f5c22f8

Sample file with 2 materials assigned to the wall: 
[material-test-assigned.zip](https://github.com/user-attachments/files/17332163/material-test-assigned.zip)

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38923 (Git)
Build type: Release
Branch: main
Hash: d20cb9e6ee198beb2bfd7e72d3dec0a575e3f28c
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-11:
@yorikvanhavre FYI
Maybe related: https://github.com/FreeCAD/FreeCAD/issues/5657

furgo16, 2024-10-11:
I don't recall there being an issue about the BIM port to Materials system (I might be wrong). There was minimal work done on https://github.com/FreeCAD/FreeCAD/issues/13821.

If there isn't one, we should probably file a new issue to start discussing the steps required for supporting the new Materials system. It might be something that new contributors could perhaps collaborate on, even if it's not included in 1.0.

yorikvanhavre, 2024-10-11:
Let's turn this issue into it then? It's basically about it...

What I like in the current system, that we should preserve:

- quick bulk-assign materials to objects: Select objects, press a key, click a material, and it's done
- the material color controls the object color
- easy to create a material without filling much detail, basically just a name, like "I plan to use that special wood I'll research later"
- the multimaterials system to define objects with multiple materials
- materials are saved within the file

Most of this seems easily achievable with the new Materials module, except the last point.

hoshengwei, 2024-10-25:
> It's strange, I thought we had an open issue for that already (BIM port to Materials system), but I can't find it back...

not sure is https://github.com/FreeCAD/FreeCAD/pull/16418 or not
I have asked "are we going to use the new Material Editor to load the material info or keep stay on the legacy one?" before

I was trying to update the Material Dialog to supprt operate IFC material , and trying to find a way to combine profile as type also.     but ...... kind of too hard of me    lol

semhustej, 2025-08-07:
The issue is still relevant.

furgo16, 2025-10-08:
Still relevant