FreeCAD/FreeCAD #17180: Assembly: crash: Assembly crash on inverting joint direction

AIRCAP, 2024-10-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Assembly crashes when you freshly insert a joint and click here:

[image]

works with any new assembly - i only added a cube and a cylinder
[assemblytest1.zip](https://github.com/user-attachments/files/17334539/assemblytest1.zip)

### Full version info

```shell
OS: Ubuntu 20.04.6 LTS (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38925 +22 (Git)
Build type: Debug
Branch: main
Hash: 31a49e7d1eec3f5d00857c2354b44a24ac12f50d
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

Assembly

### Anything else?

```
FreeCAD 1.1.0, Libs: 1.1.0devR38925 +22 (Git)
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

connect failed: No such file or directory
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x43090) [0x7fc7fa9cf090]
#1  0x7fc7c2b3e935 in Assembly::AssemblyObject::isPartGrounded(App::DocumentObject*) from ~/src/FreeCAD/main/Mod/Assembly/AssemblyApp.so+0xbb
#2  0x7fc7c2b3dbd2 in Assembly::AssemblyObject::isJointConnectingPartToGround(App::DocumentObject*, char const*) from ~/src/FreeCAD/main/Mod/Assembly/AssemblyApp.so+0xa2
#3  0x7fc7c2b9d389 in Assembly::AssemblyObjectPy::isJointConnectingPartToGround(_object*) from ~/src/FreeCAD/main/Mod/Assembly/AssemblyApp.so+0x99
#4  0x7fc7c2b9c2de in Assembly::AssemblyObjectPy::staticCallback_isJointConnectingPartToGround(_object*, _object*) from ~/src/FreeCAD/main/Mod/Assembly/AssemblyApp.so+0xee
#5  /usr/lib/x86_64-linux-gnu/libpython3.8.so.1.0(+0x2a8697) [0x7fc7fc343697]
#6  /usr/lib/x86_64-linux-gnu/libpython3.8.so.1.0(_PyObject_MakeTpCall+0xab) [0x7fc7fc343b2b]
#7  /usr/lib/x86_64-linux-gnu/libpython3.8.so.1.0(+0x74df3) [0x7fc7fc10fdf3]
#8  /usr/lib/x86_64-linux-gnu/libpython3.8.so.1.0(_PyEval_EvalFrameDefault+0x7d86) [0x7fc7fc117ef6]
#9  /usr/lib/x86_64-linux-gnu/libpython3.8.so.1.0(+0x8006b) [0x7fc7fc11b06b]
#10  /usr/lib/x86_64-linux-gnu/libpython3.8.so.1.0(+0x74d6d) [0x7fc7fc10fd6d]
#11  /usr/lib/x86_64-linux-gnu/libpython3.8.so.1.0(_PyEval_EvalFrameDefault+0xea8) [0x7fc7fc111018]
#12  /usr/lib/x86_64-linux-gnu/libpython3.8.so.1.0(+0x8006b) [0x7fc7fc11b06b]
#13  /usr/lib/x86_64-linux-gnu/libpython3.8.so.1.0(+0x2a8e3b) [0x7fc7fc343e3b]
#14  /usr/lib/x86_64-linux-gnu/libpython3.8.so.1.0(PyVectorcall_Call+0x60) [0x7fc7fc343840]
#15  0x7fc7f9aba39c in PySide::SignalManager::callPythonMetaMethod(QMetaMethod const&, void**, _object*, bool) from /usr/lib/x86_64-linux-gnu/libpyside2.cpython-38-x86_64-linux-gnu.so.5.14+0x9c
#16  0x7fc7f9aba722 in PySide::SignalManager::qt_metacall(QObject*, QMetaObject::Call, int, void**) from /usr/lib/x86_64-linux-gnu/libpyside2.cpython-38-x86_64-linux-gnu.so.5.14+0x112
#17  0x7fc7fb17c28c in QMetaObject::activate(QObject*, int, int, void**) from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x88c
#18  0x7fc7fbc5f806 in QAbstractButton::clicked(bool) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x46
#19  /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x25fa2e) [0x7fc7fbc5fa2e]
#20  /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x260e73) [0x7fc7fbc60e73]
#21  0x7fc7fbc61035 in QAbstractButton::mouseReleaseEvent(QMouseEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xe5
#22  0x7fc7fbd545fe in QToolButton::mouseReleaseEvent(QMouseEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xe
#23  0x7fc7fbbad2b6 in QWidget::event(QEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x286
#24  0x7fc7fbd546a8 in QToolButton::event(QEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x38
#25  0x7fc7fbb6aa66 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x86
#26  0x7fc7fbb74343 in QApplication::notify(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x583
#27  0x7fc7ff407a96 in Gui::GUIApplication::notify(QObject*, QEvent*) from ~/src/FreeCAD/main/lib/libFreeCADGui.so+0x14c
#28  0x7fc7fb15080a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x18a
#29  0x7fc7fbb73457 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1b7
#30  /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1c935d) [0x7fc7fbbc935d]
#31  /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1cc1ec) [0x7fc7fbbcc1ec]
#32  0x7fc7fbb6aa66 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x86
#33  0x7fc7fbb740f0 in QApplication::notify(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x330
#34  0x7fc7ff407a96 in Gui::GUIApplication::notify(QObject*, QEvent*) from ~/src/FreeCAD/main/lib/libFreeCADGui.so+0x14c
#35  0x7fc7fb15080a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x18a
#36  0x7fc7fb5397d3 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5+0x6e3
#37  0x7fc7fb53b10b in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5+0x25b
#38  0x7fc7fb51535b in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5+0xbb
#39  /usr/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5(+0x7932e) [0x7fc7f4cda32e]
#40  /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x27d) [0x7fc7f8dfb17d]
#41  /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(+0x52400) [0x7fc7f8dfb400]
#42  /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_iteration+0x33) [0x7fc7f8dfb4a3]
#43  0x7fc7fb1a8435 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x65
#44  0x7fc7fb14f3ab in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x12b
#45  0x7fc7fb157116 in QCoreApplication::exec() from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x96
#46  ~/src/FreeCAD/main/lib/libFreeCADGui.so(+0x134b318) [0x7fc7ff29d318]
#47  ~/src/FreeCAD/main/lib/libFreeCADGui.so(+0x134b647) [0x7fc7ff29d647]
#48  0x7fc7ff29d984 in Gui::Application::runApplication() from ~/src/FreeCAD/main/lib/libFreeCADGui.so+0x22c
#49  ./bin/FreeCAD(+0x2dddc) [0x563f7c276ddc]
#50  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf3) [0x7fc7fa9b0083]
#51  ./bin/FreeCAD(+0x2cfee) [0x563f7c275fee]
Aborted (core dumped)

```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

furgo16, 2024-10-11:
Confirmed.

Steps to reproduce:

1. Download and uncompress test file
2. Open `assemblytest1.FCStd`
3. Double-click on `Assembly` on the document tree to activate the assembly and load the Assembly Workbench
4. Click on the "Create a Fixed Joint" command on the toolbar, or press <kbd>F</kbd>
5. On the Task panel, click on the "Reverse direction of the joint." button at the bottom of the dialog
6. Result: FreeCAD crashes

<details><summary>Stacktrace</summary>

```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x42520) [0x7bef4ee48520]
#1  0x7bef02fc2427 in Assembly::AssemblyObject::isPartGrounded(App::DocumentObject*) from /snap/freecad/1161/usr/lib/AssemblyApp.so+0xa7
#2  0x7bef02fc465c in Assembly::AssemblyObject::isJointConnectingPartToGround(App::DocumentObject*, char const*) from /snap/freecad/1161/usr/lib/AssemblyApp.so+0x8c
#3  0x7bef02fe959b in Assembly::AssemblyObjectPy::isJointConnectingPartToGround(_object*) from /snap/freecad/1161/usr/lib/AssemblyApp.so+0x6b
#4  0x7bef02fe961c in Assembly::AssemblyObjectPy::staticCallback_isJointConnectingPartToGround(_object*, _object*) from /snap/freecad/1161/usr/lib/AssemblyApp.so+0x2c
#5  /snap/freecad/1161/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x128008) [0x7bef507f2008]
#6  /snap/freecad/1161/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyObject_MakeTpCall+0x8c) [0x7bef507a9adc]
#7  /snap/freecad/1161/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x9dbc) [0x7bef50745a1c]
#8  /snap/freecad/1161/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c23af) [0x7bef5088c3af]
#9  /snap/freecad/1161/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x829e) [0x7bef50743efe]
#10  /snap/freecad/1161/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c23af) [0x7bef5088c3af]
#11  /snap/freecad/1161/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0xe246c) [0x7bef507ac46c]
#12  0x7bef4e251ec5 in PySide::SignalManager::callPythonMetaMethod(QMetaMethod const&, void**, _object*, bool) from /snap/freecad/1161/usr/lib/x86_64-linux-gnu/libpyside2.cpython-310-x86_64-linux-gnu.so.5.15+0x95
#13  0x7bef4e256c3d in PySide::SignalManager::qt_metacall(QObject*, QMetaObject::Call, int, void**) from /snap/freecad/1161/usr/lib/x86_64-linux-gnu/libpyside2.cpython-310-x86_64-linux-gnu.so.5.15+0x14d
#14  /snap/freecad/1161/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5(+0x2f4666) [0x7bef4f59f666]
#15  0x7bef501f79f6 in QAbstractButton::clicked(bool) from /snap/freecad/1161/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x46
#16  /snap/freecad/1161/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x262c7e) [0x7bef501f7c7e]
#17  /snap/freecad/1161/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x2648a4) [0x7bef501f98a4]
#18  0x7bef501f9ac7 in QAbstractButton::mouseReleaseEvent(QMouseEvent*) from /snap/freecad/1161/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xf7
#19  0x7bef502f874e in QToolButton::mouseReleaseEvent(QMouseEvent*) from /snap/freecad/1161/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xe
#20  0x7bef5014494e in QWidget::event(QEvent*) from /snap/freecad/1161/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x20e
#21  0x7bef50101763 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /snap/freecad/1161/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x83
#22  0x7bef501093a4 in QApplication::notify(QObject*, QEvent*) from /snap/freecad/1161/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xa84
#23  0x7bef51cdfd20 in Gui::GUIApplication::notify(QObject*, QEvent*) from /snap/freecad/1161/usr/lib/libFreeCADGui.so+0xd0
#24  0x7bef4f567e4a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /snap/freecad/1161/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x13a
#25  0x7bef50107e87 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /snap/freecad/1161/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1b7
#26  /snap/freecad/1161/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1c91cb) [0x7bef5015e1cb]
#27  /snap/freecad/1161/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1cc525) [0x7bef50161525]
#28  0x7bef50101763 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /snap/freecad/1161/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x83
#29  0x7bef51cdfd20 in Gui::GUIApplication::notify(QObject*, QEvent*) from /snap/freecad/1161/usr/lib/libFreeCADGui.so+0xd0
#30  0x7bef4f567e4a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /snap/freecad/1161/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x13a
#31  0x7bef4f9901e7 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /snap/freecad/1161/kf5/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5+0x687
#32  0x7bef4f963dbc in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /snap/freecad/1161/kf5/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5+0xac
#33  /snap/freecad/1161/kf5/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms/../../../libQt5XcbQpa.so.5(+0x76f5e) [0x7bef48fbef5e]
#34  /snap/freecad/1161/kf5/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x26b) [0x7bef4d006d3b]
#35  /snap/freecad/1161/kf5/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(+0xab258) [0x7bef4d05c258]
#36  /snap/freecad/1161/kf5/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_iteration+0x33) [0x7bef4d0043e3]
[... cut ...]
#44  /snap/freecad/1161/usr/bin/FreeCAD(+0xab55) [0x643ade6a2b55]
```

</details>

<details><summary>FreeCAD version</summary>

```
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38926 (Git) Snap 1161
Build type: Release
Branch: main
Hash: a9bff78974f301100134a13c636e23aa91be5bcf
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * OpenTheme 2024.9.1
  * ezydxf
  * Ondsel-Lens 2024.9.13.02
  * Assembly4 0.50.15
```

</details>

maxwxyz, 2024-10-11:
@PaddleStroke FYI

furgo16, 2024-10-11:
https://docs.github.com/en/get-started/writing-on-github/working-with-advanced-formatting/organizing-information-with-collapsed-sections

PaddleStroke, 2024-10-12:
Thanks for the report. Fix by https://github.com/FreeCAD/FreeCAD/pull/17205

closed 2024-10-14 by chennes