FreeCAD/FreeCAD #17181: Arm64 build not available for v1.0

alexfacciorusso, 2024-10-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The Apple M1/M2 builds are available, but the Windows Arm64 builds (for PCs like the new Surface Pro and Surface Laptop with ARM64 CPU) aren't available.

It would be great to have native support and performance for those devices.

### Full version info

```shell
ND-
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

alexfacciorusso, 2024-10-10:
I will be on holidays in the next few weeks but if anybody can easily-ish integrate the builds, I can definitely test them on my "holidays laptop"

hyarion, 2024-10-11:
While you wait for a native build:
It looks like arm64 windows can run x86_64 windows builds of freecad by installing the OpenCL and OpenGL Compatibility Pack from Microsoft Store: https://forum.freecad.org/viewtopic.php?style=5&p=676066#p676066

maxwxyz, 2024-10-11:
@oursland FYI

oursland, 2024-10-11:
Unfortunately, that's not possible at this time as the underlying dependency packages are not available.

See: https://github.com/conda/conda/issues/11472

Hopefully this will change in the future.

alexfacciorusso, 2024-10-11:
Yes it runs by default on the latest windows 11 update for arm64.

Although, the difference between "runs" and "runs natively" is quite huge (the CPUs on those machines are quite beefed up, especially the Snapdragon X Elite version, but when running in x64 emulation mode, the performance isn't incredible, but yes it works).

It seems something's moving now though, and being Conda a big project, I hope we'll see builds for it soonish.
A lot of software are finally getting arm64 builds in the last month or so, possibly helped by Microsoft; in the end, if there's native apps, people will buy the newer copilot+ laptops, so it's a Microsoft priority to get support as much as possible IMHO.

adrianinsaval, 2024-10-12:
getting this done will probably require lots of patching in many projects, I don't think it's doable in time for 1.0 release.

@chennes is there any chance of compiling libpack for arm64?

chennes, 2024-10-13:
It is probably doable, though I did hardcode x86 is a couple of places.

ArminiusTux, 2024-10-25:
> ..but the Windows Arm64 builds (for PCs like the new Surface Pro and Surface Laptop with ARM64 CPU) aren't available.
> It would be great to have native support and performance for those devices too.
> 

Indeed official support is always preferable, but a 3rd party solution can be better than nothing.

The [FreeCAD MSYS2 pre-built WinARM64 package](https://packages.msys2.org/packages/mingw-w64-clang-aarch64-freecad?repo=clangarm64) is partially working - there seems to be continuining [issues with certain Python components](https://github.com/msys2/MINGW-packages/pull/19156).

hmartinez82, 2024-10-25:
The MSYS2 project already has it compiled for Windows on ARM64: https://packages.msys2.org/base/mingw-w64-freecad
Most of the work for official builds will be compiling 3rd party libraries for ARM64 first.

hmartinez82, 2024-10-26:
After the latest PySide fix for 6.8.0 (the packages are built and staged for upload soon). I got this:
[image]

hyarion, 2024-10-26:
Nice!
Would be nice to add arm64 builds as weeklies, but maybe skip it for 1.0, as it needs proper testing