FreeCAD/FreeCAD #17183: GUI Frozen when creating large objects.

AIRCAP, 2024-10-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When creating very large objects, the FreeCAD GUI freezes with 100% CPU load.

CLI is unaffected.

Steps to reproduce:

```
testdoc=App.newDocument("DeathStar")
sphere=testdoc.addObject("Part::Sphere","DysonSphere")
sphere.Radius="10000000 km"
testdoc.recompute()
```
or similar

in CLI:
executes in a fraction of a second-

in GUI
freezes FreeCAD with 100% CPU load.

### Full version info

```shell
OS: Ubuntu 20.04.6 LTS (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38930 +23 (Git)
Build type: Debug
Branch: main
Hash: 8b7cd3cf0563d5f20f9e6d97dbcd0e2e1fc147ad
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

Core

### Anything else?

My suspicion is the default tesselation, which does not take part size into account creates a few gazillion triangles on the surface of this object.

if in FreeCADCmd one types

`sphere.Shape.tessellate(1)`  or  `sphere.Shape.tessellate(0.01)`  - FreeCADCmd also freezes

but if you type  `sphere.Shape.tessellate(1e9)` - which is a reasonable tolerance for a sphere that size, it returns quickly

it looks like the tessellation for shapes in FreeCAD uses constant  tolerance instead of a multiplication with the shapes principal dimension as one would expect.

this in turn probably also means really weird things happen at very small parts, smaller than the default tessellation

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

AIRCAP, 2024-10-11:
After a bit of looking, the tesselation accuracy is actually calculated based on object size -- but then clamped to an upper limit.
for very big objects, this upper limit is catastrophic

https://github.com/FreeCAD/FreeCAD/blob/736f03caf238205630c0940d0071dbaf1ddd75ea/src/Mod/Part/Gui/ViewProviderExt.cpp#L960-L991

AIRCAP, 2024-10-11:
without above line, this is possible:

[image]

FlachyJoe, 2024-10-11:
And where does this `300.0` come from?
https://github.com/FreeCAD/FreeCAD/blob/736f03caf238205630c0940d0071dbaf1ddd75ea/src/Mod/Part/Gui/ViewProviderExt.cpp#L967

AIRCAP, 2024-10-11:
Absolutely, we also need this for Dyson spheres, Halo style rings, star cruisers like "Spaceball One".
If we ever want to become a Type II civilization on the Kardashev scale, we need a CAD program that allows us to make these large scale innovations.

furgo16, 2024-10-11:
Glad to hear FreeCAD can be used for the future common good then. Back to work now :-)

FlachyJoe, 2024-10-11:
@furgo16 It seems FreeCAD display mechanism [can be improved](https://dev.opencascade.org/node/82031#comment-22347) so working on it, even for unnecessary scale, is great because it gives to @AIRCAP the necessary knowledge.

furgo16, 2024-10-11:
To remove any doubt, my comments were made jokingly :-)

PR #17349 (2024-10-21): Part: Disable tesselation limit in ViewProviderGUI to prevent large parts from freezing FreeCAD
fix #17183
Part: Disable tesselation limit in ViewProviderGUI to prevent large parts from freezing FreeCAD

took this from #17214

required for work on #9377

@ Triage see https://github.com/FreeCAD/FreeCAD/pull/17214#issuecomment-2411342073 - assign for 1.0 ?
closed 2024-10-21 by yorikvanhavre