FreeCAD/FreeCAD #17184: Data compatibility issue of Revolution from ShapeBinder in FreeCAD 1.0 RC2

marbocub, 2024-10-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I found a data compatibility issue in FreeCAD 1.0 RC2.

An example is attached.

When opening it in FreeCAD 1.0 RC2 or weekly build 38923, a recalicuration error occurs.
```
Revolution: SketchBased: No face selected
```
Then I deleted Revolution and try to re-create new Revolution by same process before, a different error occurs.
```
Revolution: Axis must not be perpendicular to the sketch plane
Revolution: Could not create face from sketch.
Intersecting sketch entities in a sketch are not allowed.
```

On the other hand, when opening or recalicurating the example in FreeCAD 0.21.2, there are no errors.

It may be related to #11934 or #16700.

### Full version info

```shell
OS: Windows 11 build 26100
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Japanese/Japan (ja_JP)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * Assembly4 0.50.15
-----
OS: Windows 11 build 26100
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38923 (Git)
Build type: Release
Branch: main
Hash: d20cb9e6ee198beb2bfd7e72d3dec0a575e3f28c
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Japanese/Japan (ja_JP)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * Assembly4 0.50.15
```

### Subproject(s) affected?

PartDesign

### Anything else?

[Example.zip](https://github.com/user-attachments/files/17338143/Example.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-10-12:
> On the other hand, when opening or recalicurating the example in FreeCAD 0.21.2, there are no errors.

Therefore this is a regression ?

marbocub, 2024-10-12:
Thank you. The Example.FCStd created by using ver.0.21.2.

I have real data containing this compatibility issue created by 0.21 or below. If it is useful for analyzing this issue I can upload it.

luzpaz, 2024-10-13:
Yes, if you could make those files available that would be ideal.

luzpaz, 2024-10-13:
@FlachyJoe any chance you could evaluate this ticket?

marbocub, 2024-10-13:
A CAD file is attached.
Revolution, Revolution001, Revolution004 and Revolution005 have the issue.
[RealExample.zip](https://github.com/user-attachments/files/17356377/RealExample.zip)

FlachyJoe, 2024-10-13:
Hi !
The problem is with ShapeBinder, they can't be used as Profile for revolution. If we replace them with SubShapeBinder and set the Profile to `(Binder, 'Face1')` it works as expected.
I've not look at the code yet but we certainly loose the ShapeBinder compatibility somewhere and as we don't propose them anymore in the PartDesign toolbar it was not noticed before.

FlachyJoe, 2024-10-23:
In fact the given file has two problems:

- the first one is with the ShapeBinder
- the second one is a TNP, Revolution005 reference switch from
[image]
to
[image]

closed 2024-11-11 by yorikvanhavre