FreeCAD/FreeCAD #17185: Performance issues with large files (hover/preselection/navigation)

maxwxyz, 2024-10-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When working in large files selecting/preselection is super slow which leads to lagging when starting to orbit or move the camera.
I guess that is also the case why moving around in the scene is so hard because it takes about 5 seconds where nothing happens, then I can rotate without performance issues

It seems to be related that the app checks what is under the cursor. Just scrolling for zooming in and out takes 2 seconds until it starts. when it's started I can zoom in and out without issues when holding the mouse at the same position. Moving the mouse freezes again for 2 seconds then it works normally.

The app behaves normally until the cursor is over geometry.
When the mouse is moved without clicks over the geometry the preselection highlight of faces or edges lags behind significantly, it also takes a few seconds when clicking that the element gets selected.
When the mouse cursor is not over geometry (just over the background) the navigation and orbiting starts instantly, so no issues.

@Rexbas @kadet1090 FYI
The file itself is around 190 mb and has a lot of nested bodies in parts in other part containers. I cannot provide the file.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38923 (Git)
Build type: Release
Branch: main
Hash: d20cb9e6ee198beb2bfd7e72d3dec0a575e3f28c
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Curves 0.6.50
  * fasteners 0.5.29
  * freecad.gears 1.2.0
  * Manipulator 1.5.7
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

AIRCAP, 2024-10-11:
Do you have an example file? how "large" a file are we talking? 10 shapes? 100? 1000? 10k? any specific shape types that cause that?

AIRCAP, 2024-10-11:
we really need 'a' file to reproduce this. how many shapes does it need ? 10 ? 100 ? 1k ? 10k?  do the type of shapes influence this problem or is it just the raw number? is the nesting the issue?  really needs an example for reproduction, even if its a toy-file

maxwxyz, 2024-10-11:
@AIRCAP here a demo file. If it keeps working just duplicate all a few more times :)
https://1drv.ms/u/s!Aj7DqOi0Q65dkssr-_5w_RlENsfGWg?e=cgijGg

AIRCAP, 2024-10-11:
I can't reproduce it on linux - might be a windows specific issue.

Rexbas, 2024-10-11:
I can reproduce it on my linux system. I will do some more testing tomorrow.

maxwxyz, 2024-10-11:
@Rexbas great thanks!

maxwxyz, 2024-10-12:
Seems to be related or the same as https://github.com/FreeCAD/FreeCAD/issues/7214 with findings from https://github.com/FreeCAD/FreeCAD/issues/13230

Rexbas, 2024-10-12:
Yes that is related, the (pre)selection uses a SoRayPickAction and when you start rotating the camera it also uses a SoRayPickAction (depending on the rotation mode). This action takes more time when there is more geometry under the mouse. In my testing it appeared to increase linearly with the number of objects under the mouse. So when you have a big file it takes more time and could cause performance problems.

jffmichi, 2024-10-12:
Might this be related to #16560? Maybe checking if Qt5/Qt6 or Wayland/X11 has an influence might help tracking the issue down?

maxwxyz, 2024-10-13:
And what about the idea that soraypickaction is only called when the cursor is still for 1/2 ms?

Rexbas, 2024-10-13:
> Might this be related to https://github.com/FreeCAD/FreeCAD/issues/16560?

I don't think this is related as this does not seem to be a Qt6 or Wayland problem but rather a coin3d problem.

> And what about the idea that soraypickaction is only called when the cursor is still for 1/2 ms?

@maxwxyz, maybe that helps. I disabled preselection by adding `mymode = HighlightModes::OFF;` below
https://github.com/FreeCAD/FreeCAD/blob/e478192aa5393de9b7df190341a043f79ec6ac49/src/Gui/SoFCUnifiedSelection.cpp#L725

However, there is still a delay when starting to rotate when there is a lot of geometry under the mouse.

Besides that I also experience a long loading time (>45 seconds), long saving time (>20 seconds) and low fps when using "As is" or "Shaded" drawing style (10 - 20 fps), is this the same for you?

maxwxyz, 2024-10-13:
@Rexbas yes, can confirm this!

maxwxyz, 2024-10-16:
@Rexbas any other idea what might cause this?   When the cursor is not above geometry the performance is perfect, as soon as I move the cursor above geometry, working in FreeCAD with larger files and assemblies is painful.

Rexbas, 2024-10-16:
I looked a bit at the docs and code of the ray action and I believe what happens is this: when you start a ray action, all objects are checked. First a bounding box intersection is checked which is probably not computationally expansive. If there is an intersection then the computationally expansive code is executed to find the exact intersection point on the geometry.  When all objects have been checked the exact intersection points are sorted and the first/closest point is returned. So, when there are no bounding box intersections then the long computationally expansive code never runs thus there is normal performance.

Maybe it is possible to refactor it a bit smarter to prevent unnecessary heavy calculations. Like maybe analyzing the bounding boxes and running the heavy code first on the closest boxes. If there is a hitpoint on the closest boxes and we are sure it is impossible that there are hitpoint on bounding boxes behind it then there is no point in doing the heavy calculations there so we can return early (its probably more complicated than that..... xD).

Or maybe the computationally expansive function can be optimized...

I am willing to take a better look at the code, which is in coin3d, and attempt to improve the performance. However, its going to be a post 1.0 thing because it is complicated and time consuming. I first want to look at some other smaller bugs.

maxwxyz, 2024-10-16:
@Rexbas great, sounds good! I'll assign you. Maybe there are other things to complement to the idea, like only checking for objects when the cursor is not moving 0,0x seconds.

zamonary1, 2024-10-26:
I have faced simillar issues when working with imported kicad PCBs, can confirm - on latest weekly build the problen is gone. After the RC3 or V1 release the issue can be closed.

maxwxyz, 2024-10-27:
Issue is still present in the latest weekly.
```
OS: Windows 11 build 26100
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39057 (Git)
Build type: Release
Branch: main
Hash: 3be681ef62bf82d32eb1f1935d99e83c48e48da3
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CfdOF 1.27.12
  * Curves 0.6.50
  * dodo 1.0.1
  * fasteners 0.5.29
  * fasteners.backup1721936159.7542229 0.5.24 (Disabled)
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.25
  * Rocket 3.3.0
  * sheetmetal 0.5.3
  * Silk 0.1.5
  * Silk.backup1727166062.7227929 0.1.5 (Disabled)

```

maxwxyz, 2024-11-05:
I've noticed that the performance especially decreases when the geometry is nested inside nested part containers which are common for step imports of assemblies too.
Part0
-Part1
--Part2
---Part3
----Part4
-----Body1
-----Body2
----Part5
-----Body3

If I pull the bodies to the document root or only in one parent part container, there are signficantly less performance issues:

-Part4
--Body1
--Body2
-Part5
--Body3

maxwxyz, 2024-12-21:
@oursland is that the issue you have just mentioned you also experienced?

oursland, 2024-12-21:
Oh my, yes.

I do robotics work, so I have several electronics boards that have been exported as STEP files with very large, very deep hierarchies.  If I get my mouse cursor near these boards in the window, FreeCAD becomes unresponsive for a while.  My temporary workaround is to make these components invisible to remove them from the possible selection list.  Even after making the electronics boards invisible, the problem can become substantial on assemblies with enough parts.

My discussion on the call about the system profiling would help highlight these issues and provide a [Flame Graph](https://brendangregg.com/flamegraphs.html) visualization to identify the portions of code.  @bgbsww and I had a conversation about this at the North American Meetup, but existing approaches are ad-hoc.  Ideally, we could have a regular Workflow launch FreeCAD and perform a variety of common operations and generate a web-accessible flame graph visualization for developers to view.

maxwxyz, 2024-12-21:
@oursland disabling preselection was mentioned as a workaround here: https://forum.freecad.org/viewtopic.php?p=798720#p798720
But for me it is not ideal as I have no idea what I am about to pick when tasks need elements as inputs. Maybe we can only check for preselection/what is under the curser, if it is not moving for a few milliseconds.

lukash, 2025-02-01:
I'm affected by this. I'd like to point out this was not an issue in 0.21, it's a regression. (and, hence, could potentially be found by bisection)

Disabling preselection as mentioned above fixes the issue (but is not a viable workround due to not being able to see edges being picked etc.). It would seem preselection evaluation is not needed when moving and rotating the camera.

Pendelf, 2025-04-07:
https://youtu.be/TQSri9w2GnU

It slows down quite a bit

OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.41155 (Git) Conda
Build date: 2025/04/06 04:00:07
Build type: Release
Branch: main
Hash: e537278f1296211edb90bf1c938dfaebd2626f0d
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * AirPlaneDesign 0.4.1
  * btl 0.9.9
  * CurvedShapes 1.0.13
  * Curves 0.6.60
  * fasteners 0.5.38
  * PieMenu 1.11.2
  * Road 2025.2.22
  * sheetmetal 0.7.22
  * Telemetry 1.0.0

oursland, 2025-04-08:
@Pendelf You want to make that even worse?  Enable transparency on your models!

The issue is a lack of optimization in the Coin3D library which was not designed for scenes of this size and complexity.  @hyarion and @tritao have submitted a proposal to fix the issue that has recently been approved.

jhaand, 2025-06-20:
@tritao Do you have any HOWTO on how to make these graphs?

I'm trying to find out why my model in Part Design is getting so slow. 
Every time when I close the sketcher or start an operation, it takes at least 4 seconds to affect the changes. 

If I look at the depency graph, it all looks fine. OTOTH, the design includes a couple of PCB's from KiCAD and consists of around 30 sketches and operations. 

You can find it here. 
https://github.com/Nicolai-Electronics/tanmatsu-mechanical

leoheck, 2025-08-14:
@maxwxyz, selecting edges and faces works way better on `1.0.2`. On my current `1.1.0 42885 (AppImage)`, it is still a slow operation.

leoheck, 2025-12-02:
@ickby what is your full version?

Rexbas, 2025-12-02:
Wayland has more performance problems: #25203. But that is not the root cause of the original report.