FreeCAD/FreeCAD #17186: Fit all - Fit All incorrectly scales Measure labels

mowczar100, 2024-10-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Fit All does not scale the Measure label in proportion to the solid, which causes the model to scale to a very small size at certain label positions
[image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * FreeCAD-themes 2024.7.24
  * freecad.gears 1.3.0
  * OpenTheme 2024.9.1
  * ProDarkThemePreferencePack 1.0.0
  * sheetmetal 0.4.26
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-10-11:
cc @hlorus

wwmayer, 2024-10-14:
> Fit All does not scale the Measure label

Which is correct behaviour because the size of a label is independent of the zoom level.

If the size depended on the zoom level the text would become very tiny and thus unreadable when heavily zooming out. And it will become huge when heavily zooming in. In the one extreme case a label will completely disappear or in the other extreme case it will cover the whole screen.

FEA-eng, 2025-08-20:
Still relevant

FEA-eng, 2026-09-07:
Still relevant

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 26.3.0dev.20260902 (Git shallow)
Build date: 2026/09/02 02:30:22
Build type: Release
Branch: (HEAD detached at c68c45227)
Hash: c68c452275ec188f35c02f6b30aba880b3783cf1
Python 3.11.14, Qt 6.8.3, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL) [ OS: Polish/Poland (pl_PL) ]
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/113.268/1.25
OpenGL version/vendor/renderer: 4.6.0 Compatibility Profile Context 26.5.2.260413/ATI Technologies Inc./AMD Radeon(TM) Graphics
```