FreeCAD/FreeCAD #17187: Text background color of data window gets black when toggle overlay.

Panda631, 2024-10-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

This is the familiar view of the model tree and the data window.

[image]

Now click "toggle overlay" 2 times
[image]

You get this aprearance. No way to get the normal view back
[image]

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38923 (Git)
Build type: Release
Branch: main
Hash: d20cb9e6ee198beb2bfd7e72d3dec0a575e3f28c
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * EM 2.1.1
  * fasteners 0.5.27
  * Help 1.0.3
  * kicadStepUpMod 10.17.1
  * Pyramids-and-Polyhedrons
  * Render 2024.8.4
  * sheetmetal 0.2.61
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-10-11:
Did you check the other open tickets https://github.com/FreeCAD/FreeCAD/labels/Topic%3A%20Stylesheets ?

Panda631, 2024-10-11:
> Did you check the other open tickets [ Topic: Stylesheets](https://github.com/FreeCAD/FreeCAD/labels/Topic%3A%20Stylesheets) Anything regarding stylesheets/themes ?

I regulary looking into it. But as a not native english speaker I'm not allways sure to find the right key words.
I'm sorry if I missed something.

luzpaz, 2025-06-05:
@Panda631 can you still trigger in latest dev ?

closed 2025-06-05 by luzpaz