FreeCAD/FreeCAD #17189: TD: Dimensions do not take into account changes in the model.

kaiwas, 2024-10-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If the model changes (for example, when moving a window), the dimensions do not take these changes into account.

However, this happened in version 0.21
I am attaching a test file. Remove the zip extension.
[Test_techDraw_2 v 0.21.FCStd.zip](https://github.com/user-attachments/files/17342516/Test_techDraw_2.v.0.21.FCStd.zip)
[Test_techDraw_2 v 1.0.FCStd.zip](https://github.com/user-attachments/files/17342520/Test_techDraw_2.v.1.0.FCStd.zip)

Video demonstration (First 0.21, Then version 1.0)

https://github.com/user-attachments/assets/605793ea-2369-4a37-934c-4f0b4023a080

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38923 (Git) AppImage
Build type: Release
Branch: main
Hash: d20cb9e6ee198beb2bfd7e72d3dec0a575e3f28c
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-10-11:
Confirmed. Even changing the dimensions of a Part_Box shows the problem. IMO this a blocker.

WandererFan, 2024-10-11:
The 3 TD dimensions near the window all have errors.  What are the messages in the report view?

[image]

Syres916, 2024-10-11:
@WandererFan error only in tree and status bar (not shown in Report View) is:

`Autocorrect failed to fix broken references, Internal name: Dimension`

Last good working build I have is:

```
OS: Linux Mint 22 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38794 (Git) AppImage
Build type: Release
Branch: main
Hash: a47800d9ccd7d7a7cdb1779d56de03d3c64cd76e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: Behave-dark.qss/Behave-dark/Qt default
```

WandererFan, 2024-10-12:
Over zealous auto correct - after moving the window, the references have the same names, but point to different geometry.  

The quick work around is to turn off "Auto correct dimension refs" on the advanced tab.

I'll dial the auto-correct back a bit.

WandererFan, 2024-10-12:
My local build does not like the sample file.  Moving the window causes a crash in BIM. 

If somebody is able to test PR 17189 with the sample, that would be much appreciated.

kaiwas, 2024-10-13:
> My local build does not like the sample file. Moving the window causes a crash in BIM.
> 
> If somebody is able to test PR 17189 with the sample, that would be much appreciated.

I don't know how to check the changes.
But you don't necessarily have to use BIM. The problem appears even on the cube from the workbench part.

https://github.com/user-attachments/assets/26740307-4657-45b0-8316-6bf49e9311fe

The file is attached (remove the zip extension)
[TD_problem.FCStd.zip](https://github.com/user-attachments/files/17354967/TD_problem.FCStd.zip)
And yes - if you disable autocorrection, the problem does not appear.

closed 2024-10-14 by yorikvanhavre

kaiwas, 2024-10-15:
> Moving the window causes a crash in BIM.

Just for your information. I checked BIM it on today's build - everything works great! Thanks!