FreeCAD/FreeCAD #17190: Bug Constrain Horizontal and Vertical tool

marksteven, 2024-10-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Easily replicate-able  on my Mac - M2 - Ventura 13.6 - FreeCAD - 0.21.33771
Steps below that causes a crash every time, at least on my system

### Full version info

```shell
Mac M2 Ventura 13.6 
Open program FreeCAD 0.21.2-33771
Goto Part Design
New 
create body
create sketch x-y plane
(centered Rectangle) around 0,0
constrain vertical to 50mm (I key)
constrain horizontal to 50mm  (L key)
Fully constrained - green - close
Pad to 50mm
click on face - sketch - X-Y
create external geometry on 2 sides (G  key)
create rectangle connected to one corner roughly 42mm x 42mm

click on line to alter exact size to 42mm using L or I key
crash out completely

Then tried this after some forums discussion on Open GL so 

Preferences-Display-Rendering-nothing ticked
crashed
then OpenGL ticked
crashed
Software OpenGL
crashed
Both ticked
crashed

Deleted program and re installed number of times same problem

Can anyone else replicate this an a Mac M2 or know what is going on?
Thanks for any help.
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-10-11:
@marksteven can you test on RC2 ? (https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/1.0rc2)

Also please include your full [About](https://wiki.freecad.org/About) info for all versions you've tested on. TIA

marksteven, 2024-10-11:
THankyou. Here is the about information - didn't know that so again thank you. [code]
OS: macOS 13.6
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * fasteners 0.5.29
  * 3D_Printing_Tools
  * Assembly4 0.50.15
[/code]

marksteven, 2024-10-12:
Yes that worked. Have tired the same sequence a few times and no crashing so this version works well. Thankyou for the heads up

maxwxyz, 2024-10-12:
If it works on the RC I'm closing this

closed 2024-10-14 by luzpaz