FreeCAD/FreeCAD #17191: [Problem] No way to select some geometries of a particular part to attach an Assembly JCS

MiqCG, 2024-10-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

As suggested in ["Hasher mismatch warning when recomputing" issue](https://github.com/FreeCAD/FreeCAD/issues/17033) by @furgo16 , I open this new issue because the impossibility to select some geometries of a part to attach a JCS of an Assembly. Only tested with Fixed joint.

On the Report view, lots of " TopoShapeExpansion.cpp(970): hasher mismatch" and "<class 'IndexError'>" when trying to select some features of the part.

The conflicting part is named #110-006 (attached) and has been made in Part Design with revision 38898 (v.1.1.0). In the attached folder there are several tests:
- with parts linked (with inserted component tool)
- with copied parts inside of the assembly
- and with the assembly made with RC2 revision (and parts inserted/linked).

Tests made with clean .config .cache. share folders.

### Full version info

```shell
OS: Devuan GNU/Linux 5 (daedalus) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38898 (Git) AppImage
Build type: Release
Branch: main
Hash: 32e09d9
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Catalan/Spain (ca_ES)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

Assembly

### Anything else?

[Ass_test_1.zip](https://github.com/user-attachments/files/17343838/Ass_test_1.zip)

https://github.com/user-attachments/assets/5c9cc7e1-e315-4f32-83a4-90b9f6fcae02

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-10-11:
I have not done any testing with the files, but the body in the #110-006.FCStd file has its Tip set to a different feature than the visible one. This needs to be corrected.

See also the forum topic:
https://forum.freecad.org/viewtopic.php?t=91121

MiqCG, 2024-10-11:
@Roy-043 , you were totally correct, the problem was the location of the tip. 
Thank you very much and sorry, I feel a bit ashamed.
I wasn't aware of the presence of the green arrow neither knew its meaning. Definitely a good example for issue https://github.com/FreeCAD/FreeCAD/issues/13016

I don't know if this issue could be the initiator of a feature request for creating a more comprehensive message/warning to notice that the tip is not in its last position when trying to use a part in an assembly. Even with a more visible tip sign, this could be not obvious to many people. Do you (everybody) think this a useful or reasonable request?

closed 2024-10-11 by MiqCG