FreeCAD/FreeCAD #17192: Error occurs when creating a new Working Plane Proxy in the BIM Workbench

JohnnyArts, 2024-10-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I selected the top face of one of the walls as described in the tutorial. Then I hit the “Create new Working Plane Proxy” and the below error message appears. The result is that I can’t use the Working Plane Proxy.
_Error message from the report:_
15:57:07 Writing camera position
15:57:07 Writing objects shown/hidden state
15:57:07 Traceback (most recent call last):
15:57:07 File "C:\Users\cjohn\AppData\Roaming\FreeCAD\Mod\BIM\.\BimViews.py", line 324, in addProxy
15:57:07 self.update(False)
15:57:07 File "C:\Users\cjohn\AppData\Roaming\FreeCAD\Mod\BIM\.\BimViews.py", line 264, in update
15:57:07 obj.getParent()
15:57:07 AttributeError: 'FeaturePython' object has no attribute 'getParent'
15:57:08 Traceback (most recent call last):
15:57:08 File "C:\Users\cjohn\AppData\Roaming\FreeCAD\Mod\BIM\.\BimViews.py", line 264, in update
15:57:08 obj.getParent()
15:57:08 AttributeError: 'FeaturePython' object has no attribute 'getParent'
[Working_Plane_Proxy_Error.docx](https://github.com/user-attachments/files/17343956/Working_Plane_Proxy_Error.docx)
[Working_Plane_Proxy_Error.docx](https://github.com/user-attachments/files/17343964/Working_Plane_Proxy_Error.docx)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods: 
  * A2plus 0.4.67
  * BIM 2021.12.0
[/code]
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-10-11:
@JohnnyArts you're on an obsolete version of FreeCAD. Please try recreating the bug in RC2 https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/1.0rc2

yorikvanhavre, 2024-10-15:
The error in this report is not reproducible anymore in latest dev version. 

However, the working plane proxy is not positioned on the face, as the user might expect, but at the working plane position. So in order to achieve the desired result, the user has to first position the working plane on the face, then create a working plane proxy. I don't know if this should be changed or not...

JohnnyArts, 2024-10-16:
Hi Luzpaz,
I downloaded the 7zip file froms the given site, but I have no clue on how to get this installed.
Do I need to overwrite the existing folders in the "FreeCAD 0.20" folder?

As I am an mechanical engineer and not a software engineer, I am not familiar with this way of installing without proper instructions.

Thank you in advance for your help.
Best regards,

luzpaz, 2024-10-16:
@JohnnyArts download 7zip (https://www.7-zip.org/) if you don't already have it installed. Unzip the .7z archive. Then click on the executable. Should be good to go.

JohnnyArts, 2024-10-17:
Hi,

after I unzipped the downloaded file FreeCAD_1.0.0RC2-conda-Windows-x86_64-py311.7z, I have 6 folders as you can see in the added screenshot.
So whare can I find the executable?

Best regards,

JohnnyArts, 2024-10-17:
Screenshot of unzipped folder.

luzpaz, 2024-10-17:
@JohnnyArts when you respond from email we don't see any of the assets you've attached. You need to:  
1. Click on https://github.com/FreeCAD/FreeCAD/issues/17192  
2. Drag/Drop the images into the github textbox (this will auto-attach your asset(s) to this ticket.

JohnnyArts, 2024-10-18:
This is the screenshot.
[image]

luzpaz, 2024-10-18:
@JohnnyArts `bin` directory

adrianinsaval, 2024-10-18:
extract and run bin/freecad.exe

luzpaz, 2024-10-18:
Can you see the freecad.exe in bin/ ?

luzpaz, 2024-10-18:
@adrianinsaval my bad... i didn't see it in there because of the overwhelming amount of files in `bin/`. Is there a way we can make it easier for folks to find freecad.exe (where it is in the top level directory, easy to find by itself ?

adrianinsaval, 2024-10-18:
We could add a cmd script I guess, but people usually feel uncomfortable running scripts. Or a simple wrapper app as we do for macos: https://github.com/FreeCAD/FreeCAD-Bundle/blob/main/conda/osx/launcher/FreeCAD.cpp

luzpaz, 2024-10-19:
> Or a simple wrapper app

That makes sense to me.

Roy-043, 2024-11-26:
Since the original problem has been resolved this issue can be closed.

closed 2024-11-26 by Roy-043