FreeCAD/FreeCAD #17193: BIM: Error when switching default BIMExample to strict IFC mode

semhustej, 2024-10-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When trying to switch the default BIMExample.FCstd to strict IFC mode, all elements disappear and I get this error message:

```
12:04:10  Traceback (most recent call last):
12:04:10    File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-38923-conda-Windows-x86_64-py311\Mod\BIM\nativeifc\ifc_status.py", line 70, in on_toggle_lock
12:04:10      lock_document()
12:04:10    File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-38923-conda-Windows-x86_64-py311\Mod\BIM\nativeifc\ifc_status.py", line 252, in lock_document
12:04:10      doc.removeObject(n)
12:04:10  ValueError: No document object found with name 'Template'
```

Steps to reproduce:

1. Open FreeCAD
2. Open sample file BIMExample.FCstd, which is part of the installation
3. Switch to strict IFC mode
4. Error is the result and all elements disappear

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38923 (Git)
Build type: Release
Branch: main
Hash: d20cb9e6ee198beb2bfd7e72d3dec0a575e3f28c
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

BIM

### Anything else?

The issue was confirmed by one other user (Linux): https://forum.freecad.org/viewtopic.php?p=786448#p786448

Users might try new FC 1.0 functionality on sample files, so this might be good to investigate.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-12:
@yorikvanhavre fyi

paullee0, 2024-10-12:
See one confirmation at https://forum.freecad.org/viewtopic.php?p=786448#p786448

yorikvanhavre, 2024-10-15:
Confirmed. Working on a fix...

commit 8fa80ee4 (2024-10-21): BIM: Better objects filtering when IFC locking/unlocking - fixes #17193 (#17285)
commit a338d8f5 (2024-10-21): BIM: Better objects filtering when IFC locking/unlocking - fixes #17193 (#17285)
closed 2024-10-21 by yorikvanhavre