FreeCAD/FreeCAD #17194: Assembly: exploded view causes "Solve failed" errors

furgo16, 2024-10-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:

1. Download and extract the test file: [f4-window.zip](https://github.com/furgo16/f4-window/archive/refs/heads/all-in-one.zip)
2. Open the extracted file with FreeCAD
3. Double-click on `Window assembly` in the document tree to activate the assembly and to load the workbench
5. Click on the "Create exploded view" command
4. Select all elements of the right window sash
5. When the transform gizmo appears, simply move the selected elements a bit in the Y direction
6. Press OK in the Task panel dialog
7. Notice:
    - Unexpected: The exploded view does not seem to work: only the nested assemblies have been relocated
    - Unexpected: Exiting the exploded view keeps the nested assemblies relocated
    - Unexpected: "Solve failed" error on the Report view panel
    - Unexpected: Error does not go away when recomputing
    - Unexpected: Deleting the exploded view does not solve the issues above, only the Undo command does

https://github.com/user-attachments/assets/a48749c8-3bcf-495c-a44f-8d7e8b3a948d

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38926 (Git) Snap 1161
Build type: Release
Branch: main
Hash: a9bff78974f301100134a13c636e23aa91be5bcf
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * OpenTheme 2024.9.1
  * ezydxf
  * Ondsel-Lens 2024.9.13.02
  * Assembly4 0.50.15
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-10-15:
> 6. When the transform gizmo appears, simply move the selected elements a bit in the Y direction

I can confirm it regardless of the axis. Pretty nasty bug as it breaks the assembly.

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38977 (Git)
Build type: Release
Branch: main
Hash: facd8ca6e0303cf96297a7a7398185770c1a1e0b
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

furgo16, 2024-11-04:
@luzpaz or @maxwxyz, given the last message, would you mind changing the label to confirmed? Thanks!

maxwxyz, 2024-11-05:
Cab you try on the latest weekly as there have been some chances for subassemblies. Please play with the rigid parameter.

closed 2024-11-09 by chennes