FreeCAD/FreeCAD #17195: PartDesign: SubShapeBinder of App::Plane fails

PaddleStroke, 2024-10-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

[image]

Reproduction steps : 
- Create new body. Show its origin
- Select XZ plane
- Click 'SubShapeBinder'

Expected : plane on the XZ plane
Result: plane on the XY plane

So I am currently running additional test since I'm working on https://github.com/FreeCAD/FreeCAD/pull/16675 in which this bugs becomes more problematic since with this PR people will be using App::Plane much more.

### Full version info

```shell
0.22
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-10-11:
My current investigation says that the problem is that the transformation matrix of a App::Plane is somehow applied twice

So for example make a App::Plane with a rotation of 20deg yaw. The subshape binder will appear as a plane with 40deg yaw.
[image]

What is weird is that if you do 
`binder.Shape.Placement` it will tell you identity.
`binder.Shape.Faces[0].Placement` will tell you 20 deg yaw
So that looks correct. But yet the 3d view shows that it's not OK.

PaddleStroke, 2024-10-11:
If I try to make some changes to `SubShapeBinder::update` by moving Shape.setValue(xxxx) then returning after: 
1 - Placed right after `makeElementCompound` and the result is same as in above screenshot. So NOK.
```
        result.makeElementCompound(shapes);
        Shape.setValue(result);
        return;
```
2 - Now if instead I put right before like so : 
```
        Shape.setValue(shapes[0]);
        return;
        result.makeElementCompound(shapes);
```
Then the result is OK.

So something fishy happens in makeElementCompound

PaddleStroke, 2024-10-11:
Additional information : The same happens with PartDesign::Plane

luzpaz, 2024-10-11:
@PaddleStroke just curious why your About info shows `0.22` ? Shouldn't it be `1.1-dev` ?

PaddleStroke, 2024-10-11:
Indeed I just had the habit of writing 0.22. I pasted the correct details.

FEA-eng, 2026-09-15:
[This](https://github.com/FreeCAD/FreeCAD/issues/17195#issuecomment-4016053538) is still relevant

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 26.3.0dev.20260902 (Git shallow)
Build date: 2026/09/02 02:30:22
Build type: Release
Branch: (HEAD detached at c68c45227)
Hash: c68c452275ec188f35c02f6b30aba880b3783cf1
Python 3.11.14, Qt 6.8.3, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL) [ OS: Polish/Poland (pl_PL) ]
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 474.82/NVIDIA Corporation/Quadro K4000/PCIe/SSE2
```