FreeCAD/FreeCAD #17196: Assembly: exploded view moves along the Y axis are ignored

furgo16, 2024-10-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:

1. Download and extract the test file: [test-exploded-view.zip](https://github.com/user-attachments/files/17344869/test-exploded-view.zip)
2. Open extracted file with FreeCAD
3. Double click on `Assembly` on the document tree to activate it and to load the workbench
4. Click on the Create exploded view command
5. Select all components on the assembly
6. Move the components along the Y axis
7. Press OK to save the exploded view
8. Expand the Exploded_Views group on the document tree
9. Click on your newly created exploded view
10. Expected: the new exploded view shows the move along the Y axis
11. Actual: the new exploded view does show the move

https://github.com/user-attachments/assets/874b18ef-0adc-4bf8-9b44-ae61c4a23ce1

The test file already contains an exploded view. Clicking on it shows that the first move (also along the Y axis) has no effect.

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38926 (Git) Snap 1161
Build type: Release
Branch: main
Hash: a9bff78974f301100134a13c636e23aa91be5bcf
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * OpenTheme 2024.9.1
  * ezydxf
  * Ondsel-Lens 2024.9.13.02
  * Assembly4 0.50.15
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-10-15:
I can confirm it but step 6 should also mention moving the two components in the X axis. Otherwise, with just the Y axis move, there's no issue.

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38977 (Git)
Build type: Release
Branch: main
Hash: facd8ca6e0303cf96297a7a7398185770c1a1e0b
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

furgo16, 2024-10-25:
@luzpaz or @maxwxyz , given the last comment, could you please add the https://github.com/FreeCAD/FreeCAD/labels/Status%3A%20Confirmed label? Thanks!

PaddleStroke, 2024-11-07:
So when cllicking OK something weird happen. The moves objects get messed up. Some get deleted and new one created.

This was introduced by https://github.com/FreeCAD/FreeCAD/pull/16671 I'd guess.

closed 2024-11-09 by chennes