FreeCAD/FreeCAD #17198: Assembly: overflow of `getMovingPart() in UtilsAssembly.py the object name is too short` errors

furgo16, 2024-10-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:

1. Download test file and extract it: [test-exploded-view-sub.zip](https://github.com/user-attachments/files/17345426/test-exploded-view-sub.zip)
2. Open test file with FreeCAD
3. Double click on the `Assembly001` nested assembly to activate it
4. Click on the "Create a Fixed Joint" command
5. Click on a face from the `Cube008` component to select the first connector of the joint
6. Result: `getMovingPart() in UtilsAssembly.py the object name is too short, at minimum it should be something like ['Box','edge16']. It shouldn't be shorter` on the report view. Subsequently, whenever the mouse cursor is moved, there is an overflow of such messages in the report view

https://github.com/user-attachments/assets/f6d9877a-cdfd-4f02-90b5-87e3e7909742

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38926 (Git) Snap 1161
Build type: Release
Branch: main
Hash: a9bff78974f301100134a13c636e23aa91be5bcf
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * OpenTheme 2024.9.1
  * ezydxf
  * Ondsel-Lens 2024.9.13.02
  * Assembly4 0.50.15
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

furgo16, 2024-10-11:
Applying the small change to the code below returns more specific output (i.e. the name of the object at fault):

```
getMovingPart() in UtilsAssembly.py the object name ['Face4'] is too short, at minimum it should be something like ['Box','edge16']. It shouldn't be shorter
```

It turns out that the `getMovingPart()` method is getting `Face4` as the object name, where it would expect, I believe `Cube008.Face4`. You can also see that when the error is triggered, a wrong name appears on the task panel as well: `Assembly001.Face4`.

So it seems that when a nested assembly (not a link) is created inside an assembly, there's something wrong with picking up the names for the joint connectors inside the nested assembly.

```diff
diff --git a/src/Mod/Assembly/UtilsAssembly.py b/src/Mod/Assembly/UtilsAssembly.py
index 98dc4d18aa..5c50ac59fe 100644
--- a/src/Mod/Assembly/UtilsAssembly.py
+++ b/src/Mod/Assembly/UtilsAssembly.py
@@ -1053,7 +1053,7 @@ def getMovingPart(assembly, ref):
 
     if len(names) < 2:
         App.Console.PrintError(
-            "getMovingPart() in UtilsAssembly.py the object name is too short, at minimum it should be something like ['Box','edge16']. It shouldn't be shorter"
+            f"getMovingPart() in UtilsAssembly.py the object name {names} is too short, at minimum it should be something like ['Box','edge16']. It shouldn't be shorter"
         )
         return None
```

furgo16, 2024-10-12:
After investigating a bit, there seem to be a few issues:

1. The error logging code (see code snippet below) does not include the name of the object it's complaining about
2. The error logging code does not add a newline after its message
3. The function containing the error logging code ([`getMovingPart()`](https://github.com/FreeCAD/FreeCAD/blob/e478192aa5393de9b7df190341a043f79ec6ac49/src/Mod/Assembly/UtilsAssembly.py#L1027C5-L1027C18)) is being called from a method executed on mouse movement ([`moveMouse()`](https://github.com/FreeCAD/FreeCAD/blob/e478192aa5393de9b7df190341a043f79ec6ac49/src/Mod/Assembly/JointObject.py#L1755-L1761)), hence the overflow of messages whenever the mouse is being moved
4. If an assembly is nested inside a parent assembly and the nested assembly is being edited to add a joint, the subelement name is being too aggressively shortened, becoming invalid (see add selection snippet below). The workaround is to drag the nested assembly outside of the parent.
5. If the document has an assembly, new assemblies can only be added if that initial assembly is activated. Thus only nested assemblies can be created. This leads to the error in point 4.

Error logging code:
https://github.com/FreeCAD/FreeCAD/blob/e478192aa5393de9b7df190341a043f79ec6ac49/src/Mod/Assembly/UtilsAssembly.py#L1054-L1057

Add selection code: when this issue is triggered `Assembly001.Cube008.Face4` will be too eagerly shortened to `Assembly001.Face4` in the `target_link` variable:
https://github.com/FreeCAD/FreeCAD/blob/e478192aa5393de9b7df190341a043f79ec6ac49/src/Mod/Assembly/JointObject.py#L1829-L1838

This is what I believe leads to this issue. I might be wrong though, and I'll be happily corrected.

I'm unsure about submitting a PR, as point 5 goes beyond the scope of this issue but it's central to it happening. And on point 4, I feel blindly removing the second last element of the list is not very robust, but I wouldn't know how to detect whether it's a feature (and discard it) or an element (and keep it).

luzpaz, 2024-10-14:
cc @PaddleStroke

PaddleStroke, 2024-11-04:
I think this is fixed by https://github.com/FreeCAD/FreeCAD/pull/17267
Can you please confirm @furgo16 ?

furgo16, 2024-11-04:
@PaddleStroke thanks for the heads up. I can no longer trigger the issue with the original steps to reproduce.

For points 1, 2 mentioned on https://github.com/FreeCAD/FreeCAD/issues/17198#issuecomment-2408418974, I've submitted https://github.com/FreeCAD/FreeCAD/pull/17674

furgo16, 2024-11-16:
Closing as per the last comment.

closed 2024-11-16 by furgo16