FreeCAD/FreeCAD #17201: STEP dimension not accurate.

GAZ082, 2024-10-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

step: https://github.com/chirpy2605/voron/blob/main/V0/Dragon_Burner/CAD/Dragon_Burner_v0.2_v8.step (24 mb)
Cowl > Ducts > Duct Left/Right > Standard
Check the vid when comparing to the same file in Onshape. Got the same thing checked in Fusion, same diameter 4.7mm. Whereas in FC is 5.5mm (2.75*2)

Tested this in 1.1dev and in RC2

https://github.com/user-attachments/assets/2fdbfe70-b118-45b5-90c9-81be847446d5

### Full version info

```shell
OS: Arch Linux (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * OpenTheme 2024.9.1
  * QuickMeasure 2022.10.28
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-10-11:
The radii are accurate. The bigger radius is the outer edge of the chamfer. So I suspect that for some reason that edge gets measured. But I can't reproduce that effect, even after switching from CAD to Gesture navigation. Note that there is some overlapping geometry that I have made invisible first.

GAZ082, 2024-10-11:
@Roy-043 its the inner radius. Please send a screenshot. Thanks!

GAZ082, 2024-10-12:
If i do a Draft > Downgrade the measurements are ok.
If used next to a known size part, the scale is ok. So the issue is on what the measure tool is picking.

GAZ082, 2024-10-12:
@Roy-043 can you test with 0.21? I cannot run the appimage nor old versions of the package in my Arch system.

I worked with this file like 6 months ago and did not have  this issue if I'm not mistaken. But cant check myself :(

Roy-043, 2024-10-12:
Quick measure has been introduced in the current dev version, it does not exist in V.21.

GAZ082, 2024-10-13:
@GAZ082 you can use the quick dimensions wb. i don't think this is a quick measure issue but how the step is imported in 1.0, that is why I want someone to test it in 0.21 and measure with any tool.

hyarion, 2024-10-13:
I cannot reproduce the issue:
[image]

```
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38947 +6 (Git)
Build type: Release
Branch: safe-mode
Hash: a2ee0ce76d9c3a9b9198a6dea98e39ae84d5ead2
Python 3.11.8, Qt 5.15.8, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

I do notice that the selection view is reporting Edge 2 when you select the inner circle, which for me is the outer circle. The inner circle is Edge 3.

Could you try to install latest weekly (and uninstall the QuickMeasure addon) and see if it still is a problem?

GAZ082, 2024-10-13:
Thanks for checking and the screenshot @hyarion 

Ditched my config files, removed Quick Measure, same thing. Interesting the edge selection, there is something going on. Would be nice to have another Linux tester.

Interesting the edge thing,
OS: Arch Linux (GNOME/gnome-xorg)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38946 (Git) AppImage
Build type: Release
Branch: main
Hash: e478192aa5393de9b7df190341a043f79ec6ac49
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * OpenTheme 2024.9.1

hyarion, 2024-10-13:
@GAZ082 what happens if you run the following command in the python console:

`Gui.Selection.clearSelection();Gui.Selection.addSelection('Unnamed','Dragon_Burner_v0_2_v8_v409','Cowl.Ducts.Duct_Left.Part__Feature.Edge3')`

(The document must be called Unnamed for it to work)

GAZ082, 2024-10-13:
@hyarion 
[image]

Well, it selects Edge3, and I do see the 2.35 "radius" but its a straight line, not a circle :D

hyarion, 2024-10-14:
What in the...

Can you move your config files to another directory and see how it works then (would have been nice with safe mod).

If that fixes it, then it's either open theme or something strange in the configs. Either way, can you send me the configs if that was the issue? (Either upload here or on discord)

GAZ082, 2024-10-14:
@hyarion I already moved the config files to another location when tested and sent you the screenshot. I'll remove OpenTheme and make sure i'm testing a completely vanilla 1.1.dev or RC2.

GAZ082, 2024-10-19:
With 1.1.0dev, moved config files, the same @hyarion 

[image]
[image]

OS: Arch Linux (GNOME/gnome/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38991 (Git) AppImage
Build type: Release
Branch: main
Hash: ef2156d0f6d116b1642a5531a2d44fc6b7d17042
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * OpenTheme 2024.9.1 (Disabled)

maxwxyz, 2024-10-19:
@hlorus any idea? @obelisk79 you're on Linux an using open theme, can you reproduce this?

hlorus, 2024-10-19:
Apparently this isn't related to the measure command, tagging @PaddleStroke

PaddleStroke, 2024-10-20:
@hlorus same issue with the measure command ;)
[image]

Simplified file below. Same bug occurs
[WrongRadius.zip](https://github.com/user-attachments/files/17449157/WrongRadius.zip)

PaddleStroke, 2024-10-20:
Besides this is not a measurement issue at all. Here's a sketch with a 2.75mm radius circle on it : 
[image]

So this bug is that the step is somehow scaled down on import. If I have to take a guess I'd say it has something to do with units?

hyarion, 2024-10-20:
> So this bug is that the step is somehow scaled down on import.

I'm doubt it is (at least with the original file). See comment https://github.com/FreeCAD/FreeCAD/issues/17201#issuecomment-2409167283 and the discussion afterwards.

The problem seems to be that selected and highlighted edge in the ui isn't the edge that freecad selects under the hood. The reported edge name is wrong.

PaddleStroke, 2024-10-21:
My mistake ! Let me check again

PaddleStroke, 2024-10-21:
I cannot reproduce this. I see correct radius for the inner circle for both quickmeasure and measure tool (@hlorus false alert)

GAZ082, 2024-10-21:
@PaddleStroke using latest dev appimage? Can you paste your fc info? please?

maxwxyz, 2024-10-21:
So still not confirmed, right?

PaddleStroke, 2024-10-21:
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38981 (Git)
Build type: Release
Branch: core_selection_bug
Hash: d2637ec88117dc1c13d7d21d1860ba67ea6e39e6
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk , OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/FreeCAD Classic/Qt default
Installed mods: 
  * A2plus 0.4.60k
  * Assembly4 0.50.2
  * BillOfMaterials-WB-main 0.0.8.2
  * Curves 0.6.8
  * ExtremeProLight 1.5.0
  * fasteners 0.5.0
  * FreeCAD - Shortcut.lnk
  * freecad.gears 1.2.0
  * FreeCAD_FastenersWB-master 0.4.56
  * kicadStepUpMod 10.23.0
  * Ondsel 2024.9.13.02
  * OpenDark 2023.12.17
  * ProDarkThemePreferencePack 1.0.0
  * sheetmetal 0.2.63

GAZ082, 2024-10-21:
@PaddleStroke @maxwxyz @hyarion 

We need to test this with the same libraries the latest (as Saturday) Linux Appimage was built:

Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2

I have a gut feeling this is a library issue...

obelisk79, 2024-10-21:
I get 2.75 in FreeCAD. Also the measure tool is super inconsistent in behavior. Very quirky and imo not ready for primetime.

maxwxyz, 2024-10-21:
are you using quickmeasure in the status bar or the measure tool? What units are you using and did you switch units in the open session?

obelisk79, 2024-10-21:
Measure tool, units are mm, I didn't switch

GAZ082, 2024-10-22:
A user in Discord, lewis, is having a similar issue in windows, same libs as mine

OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38991 (Git)
Build type: Release
Branch: main
Hash: ef2156d0f6d116b1642a5531a2d44fc6b7d17042
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/ProDark/Qt default
Installed mods: 
  * A2plus 0.4.56a
  * Beltrami 1.1.0
  * Curves 0.4.4
  * Defeaturing
  * ExplodedAssembly
  * fasteners 0.3.44
  * fcgear 1.0.0
  * IconThemes
  * kicadStepUpMod 10.13.0
  * Manipulator 1.4.5
  * QuickMeasure 2022.10.28
  * scripts
  * sheetmetal 0.4.24
  * VendorParts

GAZ082, 2024-10-23:
Tested with the new and awesome Restart in Safe Mode and the issue persists.

OS: Arch Linux (GNOME/gnome/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39024 (Git) AppImage
Build type: Release
Branch: main
Hash: cf8ad6637351819dfa07d49cfd65d511fe14a084
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/unset/Qt default

GAZ082, 2024-11-10:
Tested with the buggy Appimage that use other libraries, more up to date, and the issue persists. So it is not a library issue.

OS: Arch Linux (GNOME/gnome/xcb)
Architecture: x86_64
Version: 1.1.0dev.39141 (Git) AppImage
Build type: Release
Branch: main
Hash: 4bd80bdcbb97d6038bc7030aed9ecebf57f4e428
**Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1**
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Qt default
Installed mods: 
  * fasteners 0.5.29
  * ThreadProfile 1.92.0
  * SearchBar 1.0.2
  * OpenTheme 2024.9.1

GAZ082, 2024-11-23:
This seems **NOT** to be happening now in 1.0 and Dev.

GAZ082, 2025-01-26:
@maxwxyz hey, this needs to be reopened. This is happening in 1.0. All what was mentioned in this ticket stays.

maxwxyz, 2025-01-26:
can you produce it in any file? I cannot reproduce it.

GAZ082, 2025-01-26:
Yes. Just import the file i posted in the original post and check the video.

In addition to the wrong radius, you can even see how selecting another circle, displays "length" instead of radius.

And this happens in both quick and normal measure tools.

https://github.com/user-attachments/assets/6a960ee5-1674-4334-8832-0aab271e1948

maxwxyz, 2025-01-27:
Does it only happen with imported step files or also with a cube created in FreeCAD?

GAZ082, 2025-01-27:
Had this to happen just with this imported step file. Native FC designs haven't found to have this issue.

hyarion, 2025-01-27:
[From debugging this with GAZ082](https://github.com/FreeCAD/FreeCAD/issues/17201#issuecomment-2409167283), it seems like the internal name for the selected edge gets mixed up somehow.
When he clicks on an edge, he gets another name than I do. And when I manually checks the value for his reported name, I get the same value on my side.

hyarion, 2025-01-27:
@GAZ082 Could you check if it happens in this file too? (if not, then it's probably the step importer)

[dragonburner.FCStd.zip](https://github.com/user-attachments/files/18557973/dragonburner.FCStd.zip)

GAZ082, 2025-01-29:
@hyarion with your file the dimensions are correct!

hyarion, 2025-01-29:
> @hyarion with your file the dimensions are correct!

That was unexpected! 😅

Can you import the step file, verify the issue, save as a freecad file and upload it here?

maxwxyz, 2025-01-29:
Seems to be an issue with the STEP file, the example file had also issues with aligning the new transform dragger.

GAZ082, 2025-01-29:
Before saving (check the selection and the output in the quick measure tool):

[image]

[image]

...

And after saving, the result is the same. I even tested opening in Safe Mode.
Are we all working with the same libs?

OS: Arch Linux (GNOME/gnome/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/unset/Qt default

File:
[Output.zip](https://github.com/user-attachments/files/18595209/Output.zip)

hyarion, 2025-01-29:
@GAZ082 I've tried 1.0 and latest dev build and get the same behavior as you with your file. So I can finally reproduce this inspection issue.

I don't fully understand why this happens but something strange is going on.

<details><summary>Version info</summary>
<p>

OS: macOS 14.5
Architecture: arm64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion

OS: macOS 14.5
Architecture: arm64
Version: 1.1.0dev.40006 (Git) Conda
Build type: Release
Branch: main
Hash: 11125be89d7ed063b706f633d36552849a22ef7e
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion

OS: macOS Sonoma (14.5)
Architecture: arm64
Version: 1.1.0dev.40038 +7 (Git) Conda
Build type: Release
Branch: ci-test/addobject
Hash: 15aba6ac79948a80c162d43ff26e445700fd7dc8
Python 3.13.0, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/Sweden (en_SE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical/physical DPI: 72/127.5

</p>
</details> 

There's a difference in OCC version

GAZ082, 2025-02-03:
@maxwxyz hey max, now that @hyarion managed to also reproduce this, can you reopen the issue? Perhaps needs to be renamed to something more descriptive?

Gebert81, 2026-08-30:
[image]

After the patch on #32297 **it works :)**  shows 4.7 mm to me, I hope I understood the problem correctly.

GAZ082, 2026-08-31:
As soon as I'm able  to run latest nightly i'll test it!

Gebert81, 2026-08-31:
> As soon as I'm able to run latest nightly i'll test it!

maybe include also the PR #32302. Its independent but another fix in that realm (at least for me)

GAZ082, 2026-08-31:
I see it has yet to be merged. I'll have to wait that until I can test in latest dev.

drwho495, 2026-09-08:
@maxwxyz https://codeberg.org/wwmayer/FreeCAD11/commit/f9c720ca5a0f0d3bb83915c9cfe2a70a471cc7e1

closed 2026-09-08 by maxwxyz

GAZ082, 2026-09-09:
I have yet to test this. Managed to get the latest dev working, will wait till the upcoming weekly release is pushed.

maxwxyz, 2026-09-09:
@GAZ082 it is in todays weekly

GAZ082, 2026-09-10:
Interesting, the issue is gone but only if i work in Safe Mode. Why may this be?

This is my non-safe info:

```
OS: Arch Linux (GNOME/gnome/wayland)
Architecture: x86_64
Version: 26.3.0dev.20260902 (Git shallow)
Build date: 2026/09/02 02:30:22
Build type: Release
Branch: (HEAD detached at c68c45227)
Hash: c68c452275ec188f35c02f6b30aba880b3783cf1
Python 3.11.14, Qt 6.8.3, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
QPA/File dialog/Color dialog: xcb/via Qt/via Qt
Navigation Style/Orbit Style/Rotation Mode: Gesture/Trackball/Object center
Logical DPI/Physical DPI/Pixel Ratio: 96/101.6/1
OpenGL version/vendor/renderer: 4.6 (Compatibility Profile) Mesa 26.2.2-arch1.1/AMD/AMD Radeon RX 550 / 550 Series (radeonsi, polaris12, ACO, DRM 3.64, 7.2.3-arch1-2)
Installed mods:
  * Fasteners Workbench 0.5.39
  * FusedFilamentDesign 0.25.250
  * Lattice2 Workbench 1.0.0
  * OpenTheme 2025.5.20
```