FreeCAD/FreeCAD #17202: RC2 Master sketch regression?

pike2k, 2024-10-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

See the forum link. Sometimes a master sketch created in RC2 will not create correct external geometry I think?

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/Sweden (en_SE)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Qt default
Installed mods: 
  * ExtremeProDark 2.6.4
  * fasteners 0.5.29
  * OpenTheme 2024.9.1
  * Render 2024.8.4
  * ThreadProfile 1.92.0
```

### Subproject(s) affected?

Sketcher

### Anything else?

https://forum.freecad.org/viewtopic.php?t=91147

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-12:
Is this the same as https://github.com/FreeCAD/FreeCAD/issues/16956?

pike2k, 2024-10-12:
Issue have been reproduced on forum. 
It is not clear yet what causes it. If you create the example in older versions of FC, it works, but not if you create this in RC2.
I am am not certain if it's the same issue as #16956 

Here's the simplest example file from forum thread. Try and pad 'PAD_ME'.
[wire_not_closed3.zip](https://github.com/user-attachments/files/17349396/wire_not_closed3.zip)

pike2k, 2024-10-20:
I have been testing the latest weekly to see if the issue was fixed. It is difficult because when I try to re-create the 3rd example "wire_not_closed3.zip" I am unable to do so.

What happens is that when I try to create the Arc in the PAD__ME sketch from the external geometry , I am not able to constrain it any longer. I always end up with 1 DOF. Thus I can not pad it.

OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38991 (Git)
Build type: Release
Branch: main
Hash: ef2156d0f6d116b1642a5531a2d44fc6b7d17042
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/Sweden (en_SE)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Qt default
Installed mods: 
  * ExtremeProDark 2.6.4
  * fasteners 0.5.29
  * OpenTheme 2024.9.1
  * Render 2024.8.4
  * ThreadProfile 1.92.0

pike2k, 2024-11-15:
I just tested if this issue was fixed in RC4. 
Using the above example there's no longer an issue!

closed 2024-11-15 by pike2k