FreeCAD/FreeCAD #17207: Sketcher: Dimension values get displayed upside down if Y is not pointing "upwards"

ElPatoNoEsTu, 2024-10-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

While editing a sketch I'm used to rotate the displayed sketch into the position which is most convenient to me to.   So it may happen, that the Sketch ist upside down while editing.   Imho the text/values displayed of horizontal alligned dimensions should always reflect the screen orientation.  

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38923 (Git)
Build type: Release
Branch: main
Hash: d20cb9e6ee198beb2bfd7e72d3dec0a575e3f28c
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
Installed mods: 
  * DynamicData 2.62.0
  * freecad.gears 1.2.0
  * LCInterlocking
  * Manipulator 1.5.7
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-12:
Please provide a demo file. There is a similar issue when the sketch is in a mirrored object. 
@FlachyJoe can you confirm?

luzpaz, 2024-10-12:
Related to #13852 ?

maxwxyz, 2024-12-03:
Closing for no feedback and duplicate

closed 2024-12-03 by maxwxyz