FreeCAD/FreeCAD #17208: Assembly: exploded view does not work correctly with nested assemblies

furgo16, 2024-10-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

https://github.com/user-attachments/assets/5ae9af43-dbd2-4c32-98a1-dda19cc81bd1

test file: [test-exploded-view-sub-new.zip](https://github.com/user-attachments/files/17349438/test-exploded-view-sub-new.zip)

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38926 (Git) Snap 1161
Build type: Release
Branch: main
Hash: a9bff78974f301100134a13c636e23aa91be5bcf
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * OpenTheme 2024.9.1
  * ezydxf
  * Ondsel-Lens 2024.9.13.02
  * Assembly4 0.50.15
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-10-15:
This issue will be affected by the upcoming https://github.com/FreeCAD/FreeCAD/pull/15629

furgo16, 2024-10-15:
Thanks. Affected or fixed?

furgo16, 2024-11-03:
Unfortunately, even after having merged https://github.com/FreeCAD/FreeCAD/pull/15629, this issue exhibits the same behavior.

```
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.39100 (Git) Snap 1179
Build type: Release
Branch: main
Hash: 8865450a3e14220925e0e449c0f1f79056b4fb89
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * OpenTheme 2024.9.1
  * ezydxf
  * Ondsel-Lens 2024.9.13.02
  * Assembly4 0.50.15
```

closed 2024-11-09 by chennes