FreeCAD/FreeCAD #17209: Reinforcement add-on crashes BIM

fhajji, 2024-10-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

After installing the "Reinforcement" Add-on (via Tools/Addon Manager) as suggested by the BIM Setup Dialog, switching to BIM results in this error message, and BIM workbench not loading:

`Workbench failure: partially initialized module 'bimcommands' has no attribute 'BimArchUtils' (most likely due to a circular import).
`

After disabling the add-on and restarting FreeCAD, BIM Workbench works again.

Reinforcement add-on version: 0.3.0

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * dodo 1.0.1
  * fasteners 0.5.29
  * parts_library
  * Reinforcement
  * Reporting
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-10-12:
This should be reported to https://github.com/amrit3701/FreeCAD-Reinforcement  

CC @amrit3701

Closing.

closed 2024-10-12 by luzpaz

yorikvanhavre, 2024-10-14:
https://github.com/amrit3701/FreeCAD-Reinforcement/issues/220