FreeCAD/FreeCAD #17210: Pocket operation using wrong length after revolution in exported STL

jnorre, 2024-10-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

After making a simple sketch and using the revolution tool, making a pocket on the end of the revolution will cause the exported STL (or 3MF) to use the wrong length. In the attached example, the figure has a 10 mm cylinder on top of a wider cylinder (this makes the problem easy to see later) after the revolution, but making a 10 mm pocket into the narrow cylinder and exporting to a STL file, the pocket will only be 5 mm deep when the STL is opened in a slicer. Setting the pocket to two dimensions and 2nd length to 10 mm will make the correct pocket depth.

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38926 (Git) Snap 1161
Build type: Release
Branch: main
Hash: a9bff78974f301100134a13c636e23aa91be5bcf
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

None

### Anything else?

[error.zip](https://github.com/user-attachments/files/17350002/error.zip)

Works correctly in 0.21.2 (AppImage).
Output STL tried in Cura and Bambu Studio.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-10-12:
The problem is caused by the Shape property of the Body which is wrong for some reason. You can check this by creating a [Part_SimpleCopy](https://wiki.freecad.org/Part_SimpleCopy) from the Body. Recomputing the whole document fixes the issue.

maxwxyz, 2024-10-13:
@Roy-043 is it working in the stable 0.21.2 version?

FlachyJoe, 2024-10-14:
@maxwxyz yes, I've a suspicion. I'm checking.

chennes, 2024-10-16:
Temporarily reopening the auto-closed issue -- it's fixed in main, but needs a manually-created PR in the release branch.

FlachyJoe, 2024-10-19:
@chennes this issue can be closed following https://github.com/FreeCAD/FreeCAD/pull/17248#issuecomment-2417372913
No branches have the issue anymore.

PR #17248 (2024-10-16): PartDesign: Fix body shape not updated
Fix #17210
closed 2024-10-19 by maxwxyz