FreeCAD/FreeCAD #17213: Draft_Offset wrong does not take units setup into account

semhustej, 2024-10-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Draft_Offset always uses mm as an input number for the offset distance, even though it is showing units, which are set in the project.

Steps to reproduce:

1. Open new file
2. Set different units than mm (in for instance)
3. Create a line
4. Draft_Offset the line
5. Select the direction by moving mouse
6. Input 5 vie keyboard and hit enter
7. The new offset line is not offset by 5 inches, but by 5mm

See behavior in video:

https://github.com/user-attachments/assets/24665b35-e70a-4552-ae8d-4d8b3648a3e2

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38923 (Git)
Build type: Release
Branch: main
Hash: d20cb9e6ee198beb2bfd7e72d3dec0a575e3f28c
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

Draft

### Anything else?

I was able to reproduce the behavior in FC 0.21.2

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2024-10-16:
Isn't that the standard behaviour of all Quantity-based input fields? That is, if you don't specify any unit, it uses mm?

semhustej, 2024-10-16:
> Isn't that the standard behaviour of all Quantity-based input fields? That is, if you don't specify any unit, it uses mm?

Oh I see. I got confused by using line and polyline, where you can input point location or length and selected units (cm or in) are added automatically. Apparently tools as offset (distance setting), circle, polygon or fillet (radius setting) require from the user to input units, otherwise FC defaults to mm.

This is very confusing in my opinion. When I set cm in **Draft WB**, there are several different ways the numbers I input are taken:

**- XYZ location and Length (Line, Polyline, Circle, Rectangle Ellipse, Wall, etc.)**
user can input number, FC automatically sets cm which is set-up
if user deletes the unit from the input field, FC defaults to mm

**- offset, radius (Circle, Fillet, Offset, etc.)**
even though by default FC shows cm, when user types in number, cm text is deleted and FC defaults to mm. So user has to always write the unit.

**- Properties panel (Length, Radius, etc)**
units are shown in cm.
when user inputs number, cm are automatically added to the input
when user inputs number and deletes 'cm', the input is taken as cm

**- Sketcher (BIM WB)**
when user inputs number, FC automatically adds cm
if user forcefully deletes 'cm', sketcher defaults to cm. So this is consistent with the properties panel.

These are three different ways, how units behave in Draft and BIM workbenches. It is inconsistent.

I checked few tools from **Part WB**. All the input fields automatically add cm as a unit and even if user forcefully deletes this text, FC defaults to cm. This is consistent and the behavior Draft/BIM tools should mimick.

See example Part_CompOffset:

https://github.com/user-attachments/assets/535ce1b4-ab6f-4129-927e-96763b961120

Can this 'bug' be changed to feature request across Draft/BIM? Or close this one and create a more global feature?

- I understand that it would be large and pretty tedious task and not a priority for near future, but at least it would here on GitHub
- I think consistency in behavior in regards to units across mechanical and BIM/Draft workbenches should be a goal for FreeCAD
- in my opinion the Part workbench behavior of always defaulting to selected units is the desired one
- in my opinion current status, where user has to be aware of units in a different way for every tool is leading to mistakes and user fatigue. I think it must be confusing for architects/engineers from countries using imperial unit system or systems defaulting to cm or m

yorikvanhavre, 2024-10-17:
> - XYZ location and Length (Line, Polyline, Circle, Rectangle Ellipse, Wall, etc.) user can input number, FC automatically sets cm which is set-up if user deletes the unit from the input field, FC [...]

This is not the case AFAICS, if you use "200" (without any unit) in the coordinate fields, it will interpret as 200mm, whatever your current unit scheme is.

If other WBs are doing this differently, then indeed we should adapt Draft. But we need to discover how they did :sweat_smile:

semhustej, 2024-10-17:
> This is not the case AFAICS, if you use "200" (without any unit) in the coordinate fields, it will interpret as 200mm, whatever your current unit scheme is.

Yes, it occurs to me Draft/BIM WB work like that in most places.

> 
> If other WBs are doing this differently, then indeed we should adapt Draft. But we need to discover how they did 😅

That would be great. Part_CompOffset might be a good candidate to look at. Although there is no interactive option to click/point direction in the view with this tool as in Draft_Offset, I do not know how much that complicates things.

Roy-043, 2025-01-14:
There was an issue with the focus behavior of the offset field. This was fixed with #18142.

semhustej, 2025-01-16:
> There was an issue with the focus behavior of the offset field. This was fixed with #18142.

I tested the tool with the latest build:

- the original post issue is solved

I think this issue can be closed.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39841 (Git) Conda
Build type: Release
Branch: main
Hash: 7a5a3d1ffcd6a5a01c90f6d17f5dd7eb8f6fa7aa
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion

```

closed 2025-01-17 by Roy-043