FreeCAD/FreeCAD #17216: Mesh WB: Mesh_FromPartShape using Netgen mesher crashes every time

johnny-bit, 2024-10-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Using MeshWB Mesh_FromPartShape with netgen mesher crashes every time.

Here's easy reproduction steps:

1. Open new empty document
2. Go to Part WB and create primitive solid, any will do
3. Go to Mesh WB, select solid created in previous step
4. Use "Create mesh from shape..." action
5. Select "netgen" tab and click OK
6. [here crash happens]

The used netgen package version is: 6.2.2302

I've also asked about this in:

- FreeCAD Forums: https://forum.freecad.org/viewtopic.php?p=785286#p785286
- Gentoo Bugzilla: https://bugs.gentoo.org/940623

### Full version info

```shell
OS: Gentoo Linux (GNOME/gnome-xorg)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.Unknown
Build type: Release
Python 3.11.10, Qt 6.7.2, Coin 4.0.2, Vtk 9.2.5, OCC 7.7.0
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * Assembly4 0.50.15
  * Gridfinity 0.6.0
  * Render 2024.9.25
  * ExplodedAssembly
  * Pyramids-and-Polyhedrons
  * Manipulator 1.5.7
  * A2plus 0.4.68
  * fasteners 0.5.29
  * Curves 0.6.48
  * Estimate 0.1.2
  * sheetmetal 0.4.26
```

### Subproject(s) affected?

Mesh

### Anything else?

Log from crash:

```
paź 12 20:41:55 localhost org.freecad.FreeCAD.desktop[263291]: Program received signal SIGSEGV, Segmentation fault.
paź 12 20:41:55 localhost org.freecad.FreeCAD.desktop[263291]: #0  /usr/lib64/libc.so.6(+0x3c3b0) [0x7f5a13b743b0]
paź 12 20:41:55 localhost org.freecad.FreeCAD.desktop[263291]: #1  0x7f59a44b9277 in netgen::OCCGeometry::GetEdge(TopoDS_Shape const&) const from /usr/lib64/libnglib.so+0x27
paź 12 20:41:55 localhost org.freecad.FreeCAD.desktop[263291]: #2  0x7f59a44b42a2 in netgen::OCCSetLocalMeshSize(netgen::OCCGeometry const&, netgen::Mesh&, netgen::MeshingParameters const&, netgen::OCCParameters const&) from /usr/lib64/libnglib.so+0xca2
paź 12 20:41:55 localhost org.freecad.FreeCAD.desktop[263291]: #3  0x7f59a43c291f in netgen::NetgenGeometry::GenerateMesh(std::shared_ptr<netgen::Mesh>&, netgen::MeshingParameters&) from /usr/lib64/libnglib.so+0x6ff
paź 12 20:41:55 localhost org.freecad.FreeCAD.desktop[263291]: #4  0x7f59afbe6fcb in netgen::OCCGenerateMesh(netgen::OCCGeometry&, std::shared_ptr<netgen::Mesh>&, netgen::MeshingParameters&) from /usr/lib64/freecad/lib64/libNETGENPlugin.so+0x2b
paź 12 20:41:55 localhost org.freecad.FreeCAD.desktop[263291]: #5  0x7f59afbdbda3 in NETGENPlugin_Mesher::Compute() from /usr/lib64/freecad/lib64/libNETGENPlugin.so+0x343
paź 12 20:41:55 localhost org.freecad.FreeCAD.desktop[263291]: #6  0x7f59afbe56e6 in NETGENPlugin_NETGEN_2D::Compute(SMESH_Mesh&, TopoDS_Shape const&) from /usr/lib64/freecad/lib64/libNETGENPlugin.so+0xb6
paź 12 20:41:55 localhost org.freecad.FreeCAD.desktop[263291]: #7  0x7f59a529b4eb in SMESH_subMesh::ComputeStateEngine(int) from /usr/lib64/freecad/lib64/libSMESH.so+0xa7b
paź 12 20:41:55 localhost org.freecad.FreeCAD.desktop[263291]: #8  0x7f59a51dd80e in SMESH_Gen::Compute(SMESH_Mesh&, TopoDS_Shape const&, bool, bool, MeshDimension, std::set<int, std::less<int>, std::allocator<int> >*) from /usr/lib64/freecad/lib64/libSMESH.so+0xf3e
paź 12 20:41:55 localhost org.freecad.FreeCAD.desktop[263291]: #9  0x7f59f06dc5be in MeshPart::Mesher::createMesh() const from /usr/lib64/freecad/lib64/MeshPart.so+0x13e
paź 12 20:41:55 localhost org.freecad.FreeCAD.desktop[263291]: #10  /usr/lib64/freecad/lib64/MeshPart.so(+0x32c5a) [0x7f59f06b8c5a]
paź 12 20:41:55 localhost org.freecad.FreeCAD.desktop[263291]: #11  0x7f59f06c16ca in MeshPart::Module::meshFromShape(Py::Tuple const&, Py::Dict const&) from /usr/lib64/freecad/lib64/MeshPart.so+0xefa
[... 27 lines cut ...]
paź 12 20:41:55 localhost org.freecad.FreeCAD.desktop[263291]: #39  0x7f5a151ae823 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /usr/lib64/libQt6Widgets.so.6+0x1e3
paź 12 20:41:55 localhost org.freecad.FreeCAD.desktop[263291]: #40  /usr/lib64/libQt6Widgets.so.6(+0x50b4ee) [0x7f5a151b44ee]
paź 12 20:41:55 localhost org.freecad.FreeCAD.desktop[263291]: #41  /usr/lib64/libQt6Widgets.so.6(+0x50f010) [0x7f5a151b8010]
paź 12 20:41:55 localhost org.freecad.FreeCAD.desktop[263291]: #42  0x7f5a151a7025 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib64/libQt6Widgets.so.6+0x85
paź 12 20:41:55 localhost org.freecad.FreeCAD.desktop[263291]: #43  0x7f5a16a23818 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib64/freecad/lib64/libFreeCADGui.so+0x98
paź 12 20:41:55 localhost org.freecad.FreeCAD.desktop[263291]: #44  0x7f5a14281d18 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib64/libQt6Core.so.6+0x168
paź 12 20:41:55 localhost org.freecad.FreeCAD.desktop[263291]: #45  0x7f5a146b7a1d in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /usr/lib64/libQt6Gui.so.6+0x94d
paź 12 20:41:55 localhost org.freecad.FreeCAD.desktop[263291]: #46  0x7f5a1470e54c in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib64/libQt6Gui.so.6+0xac
paź 12 20:41:55 localhost org.freecad.FreeCAD.desktop[263291]: #47  /usr/lib64/qt6/plugins/platforms/../../../libQt6XcbQpa.so.6(+0x55d0e) [0x7f5a0de5ad0e]
paź 12 20:41:55 localhost org.freecad.FreeCAD.desktop[263291]: #48  /usr/lib64/libglib-2.0.so.0(+0x58449) [0x7f5a124c9449]
paź 12 20:41:55 localhost org.freecad.FreeCAD.desktop[263291]: #49  /usr/lib64/libglib-2.0.so.0(+0x5a4c7) [0x7f5a124cb4c7]
paź 12 20:41:55 localhost org.freecad.FreeCAD.desktop[263291]: #50  /usr/lib64/libglib-2.0.so.0(g_main_context_iteration+0x30) [0x7f5a124cbae0]
paź 12 20:41:55 localhost org.freecad.FreeCAD.desktop[263291]: #51  0x7f5a14061510 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib64/libQt6Core.so.6+0x70
paź 12 20:41:55 localhost org.freecad.FreeCAD.desktop[263291]: #52  0x7f5a142ab57a in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib64/libQt6Core.so.6+0x14a
paź 12 20:41:55 localhost org.freecad.FreeCAD.desktop[263291]: #53  0x7f5a142ab72a in QCoreApplication::exec() from /usr/lib64/libQt6Core.so.6+0x9a
paź 12 20:41:55 localhost org.freecad.FreeCAD.desktop[263291]: #54  0x7f5a1694cb66 in Gui::Application::runApplication() from /usr/lib64/freecad/lib64/libFreeCADGui.so+0xd16
paź 12 20:41:55 localhost org.freecad.FreeCAD.desktop[263291]: #55  /usr/lib64/freecad/bin/FreeCAD(+0x842e) [0x55743ed4142e]
paź 12 20:41:55 localhost org.freecad.FreeCAD.desktop[263291]: #56  /usr/lib64/libc.so.6(+0x260d0) [0x7f5a13b5e0d0]
paź 12 20:41:55 localhost org.freecad.FreeCAD.desktop[263291]: #57  /usr/lib64/libc.so.6(__libc_start_main+0x89) [0x7f5a13b5e189]
paź 12 20:41:55 localhost org.freecad.FreeCAD.desktop[263291]: #58  /usr/lib64/freecad/bin/FreeCAD(+0x8775) [0x55743ed41775]
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

mwganson, 2024-10-14:
I do not know anything about gentoo.  I assume this is a self-compiled version since it doesn't say AppImage in the about information details.  When I build FreeCAD in ubuntu using cmake the option to build netgen is turned off by default.  You might need to enable that option in gentoo in whatever build tool the distribution provides.  Alternatively, you might need to build netgen separately, especially if you suspect there is a version issue.  I never try building netgen because I don't need it on the build system (which actually resides in a virtualbox vm under a windows host).  I only build for testing PR's, not for actually using FreeCAD, which I do on the windows host using the provided builds.

luzpaz, 2024-10-14:
@johnny-bit can you reproduce on the weekly AppImage? (https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds)

luzpaz, 2024-10-14:
I can't reproduce crash or any errors
```
OS: Manjaro Linux (KDE/plasma/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38977 (Git) AppImage
Build type: Release
Branch: main
Hash: facd8ca6e0303cf96297a7a7398185770c1a1e0b
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

FEA-eng, 2024-10-15:
@marioalexis84 Can it be related to the new Netgen implementation in FEM?

johnny-bit, 2024-10-15:
@luzpaz - the crash doesn't happen on AppImage build, so that points me to gentoo packaging of netgen being a problem. What's the netgen version bundled in AppImage?

marioalexis84, 2024-10-15:
The bug comes from the Netgen plugin built into the SMESH version provided by FreeCAD, which has not worked for several years and is the main reason why a new interface to interact with Netgen via python bindings was implemented in the FEM module.
The Windows and AppImage versions work because they use another external plugin.

luzpaz, 2024-10-16:
@marioalexis84 can you propose a possible way forward?

lukh, 2025-01-05:
Hello, Having the same crash here on arch, with netgen installed by AUR https://aur.archlinux.org/packages/ngsolve-git

marioalexis84, 2025-01-06:
As stated, salomesmesh internal Netgen plugin is broken.
To use the new implementation, you need the Netgen Python bindings and uncheck `Legacy Netgen` checkbox in Netgen preferences.

FEA-eng, 2026-11-06:
Still has open PR linked.