FreeCAD/FreeCAD #17219: Errors on intital startup, previous versions worked fine.

Hueristic, 2024-10-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

No logfile created in  %APPDATA%\FreeCAD.

Error list from display on initial startup:

> 19:30:41  During initialization the error "DLL load failed while importing Measure: The specified procedure could not be found." occurred in E:\FreeCAD_1.0.0RC2-conda-Windows-x86_64-py311\Mod\Measure\InitGui.py
19:30:41  Please look into the log file for further information
19:30:41  Migrating Start Workbench to Start command... done.
19:30:41  BIM addon path has been renamed to BIM021 to avoid conflicts with the builtin BIM workbench. Please restart FreeCAD
19:30:41  Wizard shaft module cannot be loaded
19:30:41  DLL load failed while importing PartDesignGui: The specified procedure could not be found.
19:30:41  Wizard shaft module cannot be loaded
19:30:41  DLL load failed while importing PartDesignGui: The specified procedure could not be found.
19:30:41  DirectWrite: CreateFontFaceFromHDC() failed (Unknown error 0x88985000.) for QFontDef(Family="8514oem", pointsize=8.25, pixelsize=20, styleHint=5, weight=50, stretch=100, hintingPreference=0) LOGFONT("8514oem", lfWidth=0, lfHeight=-20) dpi=192
19:30:41  DirectWrite: CreateFontFaceFromHDC() failed (Unknown error 0x88985000.) for QFontDef(Family="Fixedsys", pointsize=8.25, pixelsize=15, styleHint=5, weight=50, stretch=100, hintingPreference=0) LOGFONT("Fixedsys", lfWidth=0, lfHeight=-15) dpi=192
19:30:41  DirectWrite: CreateFontFaceFromHDC() failed (Unknown error 0x88985000.) for QFontDef(Family="Modern", pointsize=8.25, pixelsize=73, styleHint=5, weight=50, stretch=100, hintingPreference=0) LOGFONT("Modern", lfWidth=0, lfHeight=-73) dpi=192
19:30:41  DirectWrite: CreateFontFaceFromHDC() failed (Unknown error 0x88985000.) for QFontDef(Family="MS Sans Serif", pointsize=8.25, pixelsize=13, styleHint=5, weight=50, stretch=100, hintingPreference=0) LOGFONT("MS Sans Serif", lfWidth=0, lfHeight=-13) dpi=192
19:30:41  DirectWrite: CreateFontFaceFromHDC() failed (Unknown error 0x88985000.) for QFontDef(Family="MS Serif", pointsize=8.25, pixelsize=11, styleHint=5, weight=50, stretch=100, hintingPreference=0) LOGFONT("MS Serif", lfWidth=0, lfHeight=-11) dpi=192
19:30:41  DirectWrite: CreateFontFaceFromHDC() failed (Unknown error 0x88985000.) for QFontDef(Family="Roman", pointsize=8.25, pixelsize=73, styleHint=5, weight=50, stretch=100, hintingPreference=0) LOGFONT("Roman", lfWidth=0, lfHeight=-73) dpi=192
19:30:41  DirectWrite: CreateFontFaceFromHDC() failed (Unknown error 0x88985000.) for QFontDef(Family="Script", pointsize=8.25, pixelsize=72, styleHint=5, weight=50, stretch=100, hintingPreference=0) LOGFONT("Script", lfWidth=0, lfHeight=-72) dpi=192
19:30:41  DirectWrite: CreateFontFaceFromHDC() failed (Unknown error 0x88985000.) for QFontDef(Family="Small Fonts", pointsize=8.25, pixelsize=11, styleHint=5, weight=50, stretch=100, hintingPreference=0) LOGFONT("Small Fonts", lfWidth=0, lfHeight=-11) dpi=192
19:30:41  DirectWrite: CreateFontFaceFromHDC() failed (Unknown error 0x88985000.) for QFontDef(Family="System", pointsize=8.25, pixelsize=16, styleHint=5, weight=50, stretch=100, hintingPreference=0) LOGFONT("System", lfWidth=0, lfHeight=-16) dpi=192
19:30:41  DirectWrite: CreateFontFaceFromHDC() failed (Unknown error 0x88985000.) for QFontDef(Family="Terminal", pointsize=8.25, pixelsize=12, styleHint=5, weight=50, stretch=100, hintingPreference=0) LOGFONT("Terminal", lfWidth=0, lfHeight=-12) dpi=192
19:30:41  The `.FCStd#` backup format is deprecated and may be removed in future versions.
To update, check the 'Preferences->General->Document->Use date and FCBak extension' option.

### Full version info

```shell
OS: Windows 7 build 7601
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
Installed mods: 
  * BIM021 2021.12.0
  * fasteners 0.4.6
  * fcgear 1.0.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-10-13:
Notes:  
* `DirectWrite: CreateFontFaceFromHDC() failed (Unknown error 0x88985000.)` font related issue: https://forum.freecad.org/viewtopic.php?t=78007

luzpaz, 2024-10-16:
Can anyone diagnose this?

luzpaz, 2024-10-19:
@chennes any idea?

Hueristic, 2024-10-26:
System was heavily modified and after investigation found to have fonts missing, thanks for the insight and reply's.

closed 2024-10-26 by Hueristic