FreeCAD/FreeCAD #17221: remember window arrangement

herrdeh, 2024-10-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When using spreadsheets, you need to rearrange windows arrangement after every Freecad relaunch. As that's quite a lot of clicks, I'd appreciate to include the arrangements of windows such as spreadsheets, as already done with models, tasks etc. 

### Full version info

```shell
OS: Ubuntu 22.04.5 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38898 (Git)
Build type: Release
Branch: main
Hash: 32e09d9554db26414697ed19e0189ccab4ea9029
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/Classic/Qt default
Installed mods: 
  * 3D_Printing_Tools
  * Assembly4 0.50.12
  * dodo 1.0.1
  * ExtremeProLight 1.6.0
  * Curves 0.6.29
  * BOLTSFC
  * nurbs
  * animation
  * IconThemes
  * lattice2 1.0.0
  * ProDarkThemePreferencePack 1.0.0 (Disabled)
```

### Subproject(s) affected?

None

### Anything else?

[This forum entry](https://forum.freecad.org/viewtopic.php?t=90187) indicates that there is no solution for this.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

hyarion, 2024-10-13:
I don't understand what you mean, can you take a screenshots of:
1. how it looks after save, before closing and
2. how it looks after restart and loading file?

maxwxyz, 2024-10-14:
Looks like a duplicate of https://github.com/FreeCAD/FreeCAD/issues/11979

herrdeh, 2024-10-16:
No, it's not like #11979 .
Here are two screens. 
As I arranged my windows:
[image]

Pressed Shift-F afterwards, then closed.
After relaunch:

[image]

But I'd like to have view #1 again.

Cheers, Wolf

hyarion, 2024-10-16:
I also this differs from 11979.
That issue was about remembering window placement on multiple screens, even if that screen isn't available anymore - making it impossible to reach some windows.

This bug is, to my understanding, about windows _inside_ the main window. And that the setting for how they are tiled aren't saved between sessions.

1. open freecad
2. load a file with a spreadsheet
3. tile windows (which will show spreadsheet on one side and 3d view on the other)
4. save file
5. close freecad
6. open freecad again with the same file

now the windows aren't tiled anymore.

@herrdeh correct me if I'm wrong :)

herrdeh, 2024-10-16:
> @herrdeh correct me if I'm wrong :)
Perfect diagnosis!
(:

herrdeh, 2026-09-30:
> [Verfolgungsbild][https://github.com/notifications/beacon/ABWSJT5EOPUZDXJINDQCHBD5RRJLVBFCNFSM6AAAAADAZVLTPCWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTYAAAAACX6GPP3KM4TFMFZW63VHNVSW45DJN5XA.gif]

PLLEEAAASE.
Kerp it open. Still present, still annoying.