FreeCAD/FreeCAD #17222: extrusion on reference plane generates unwanted shapes

Roberto8303, 2024-10-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

[image]
[image]
[image]
[image]
why are the two rectangles generated on a curved sketch? and also it is not closed but an open profile?

### Full version info

```shell
OS: Windows 10 build 19042
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38923 (Git)
Build type: Release
Branch: main
Hash: d20cb9e6ee198beb2bfd7e72d3dec0a575e3f28c
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Italian/Italy (it_IT)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * ExplodedAssembly
  * fasteners 0.5.29
  * PieMenu 1.8.0
  * QuickMeasure 2022.10.28
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-14:
Seems like an OCC issue.
Does it work in the stable 0.21.2 version?
Please share a demo file and steps how to reproduce.

Roberto8303, 2024-10-14:
Now it seems to work, I did the job again

luzpaz, 2024-10-14:
I'm confused. What changed?

Roberto8303, 2024-10-14:
I tried on another PC where I work with the same version 1.1 0 dev and when opening the same file the two rectangular lines do not appear, later I try again on my personal PC where the error occurred and I will update you

Roberto8303, 2024-10-14:
[image]
here is the problem in this pc I had too low settings in the figure display
by raising the values ​​they no longer appear

luzpaz, 2024-10-15:
>  here is the problem in this pc I had too low settings in the figure display
by raising the values ​​they no longer appear

Should we add a warning to this preference page, in that case?

maxwxyz, 2024-10-15:
Closing this when fixed.